openbmc/openbmc #1045: IPMI reboot from skiboot fails due to timing

GHF, 2017-01-27:
We found on Zaius that if we reboot from skiboot, the flow seems to work but mboxd-reload doesn't complete in time before the following boot starts.

That service needs a corresponding wait for the mboxd process to come back. That wait also needs to block the subsequent boot.

williamspatrick, 2017-01-27:
I assume by mboxd-reload you are referring to `mboxd-reload@.service`, which simply sends a SIGHUP at the currently running mboxd process.

There was an observation originally that this SIGHUP hung the BMC for about 20 seconds while the PNOR chip was re-read.  @cyrilbur-ibm and I had a conversation about this before I accepted the patch that made the SIGHUP non-blocking, by doing reads at a smaller size instead of reading the whole blob in one syscall.  There are two things that happen by the SIGHUP:
  1. The mailbox and LPC registers are reset.
  2. The PNOR is re-read into the reserved memory blob.

@cyrilbur-ibm decided that 1 was near-instant and not a problem, given that it takes longer for the SBE to start than this takes, and that 2 can be done in parallel.  He implemented the skiboot side of this as well and there are messages that go across the mbox before skiboot reads the memory region.  If the read has not completed, the message will not be responded to, which effectively blocks skiboot from using the not-yet-populated memory region.

Are you observing a real problem here?  Or are you just noticing that the SIGHUP is no longer blocking and want to make sure there is no race?

GHF, 2017-01-27:
Yes, I do mean `mbox-reload@.service`.

We did observe that the SIGHUP it sends hangs the BMC and usually drops its SSH sessions, but I didn't know about the fix and will test it today.

The problem we observed is that IPMI reboot did not result in the host coming back up.

In fact, IPMI reboot *did not* cause the BMC hang, which we had come to expect from the SIGHUP being sent. Given that SBE also didn't successfully kick off when this happened, I assumed that there was a race during reboot that caused the SIGHUP to not be sent (e.g. if starting `obmc-chassis-start@0.target` killed `mboxd-reload@0.service`, not necessarily that sending the SIGHUP causes a race).

That said, I don't know enough about the mailbox/SBE/skiboot flow to determine what is or isn't a race.

williamspatrick, 2017-01-27:
@geissonator - I suspect this is because the IPMI initiated reset is not going through our systemd targets correctly right now?  Any ideas?  We might need to hack something in to reset the mailboxd in there until you get it all moved over to systemd properly.

geissonator, 2017-01-27:
Host IPMID is still using the chassis control object which just does a power off then on.  Something going on in that off/on path, don't see anything obvious in meta-phosphor/common/recipes-phosphor/mboxd/mboxd/mboxd-reload@.service though.  Matt and I are taking a look.

geissonator, 2017-01-27:
Oh, I see what's you're saying Patrick.  We put the mboxd-reload@.service in the  obmc-chassis-stop@0.target but the current code doesn't look at target states in the reboot path, it just looks at pgood.  So as soon as pgood goes to off, and our system state changes to HOST_POWERED_OFF, we power it back on (before the mboxd-reload@.service runs).

GHF, 2017-01-27:
I got the log for a failed reboot and I see that obmc-power-stop@0.target is never actually reached.

See "systemd[1]: Stopped target Power0 Off."

```
Jan 27 21:51:52 zaius chassis_control.py[792]: Rebooting
Jan 27 21:51:52 zaius chassis_control.py[792]: Turn off power
Jan 27 21:51:52 zaius systemd[1]: Stopped target Start Host0.
Jan 27 21:51:52 zaius systemd[1]: Stopped target Chassis0 (Start).
Jan 27 21:51:52 zaius systemd[1]: Stopped Start host0 SBE (pdbg).
Jan 27 21:51:52 zaius systemd[1]: Stopped target Start Host0 (Pre).
Jan 27 21:51:52 zaius systemd[1]: Stopped Enable AVSBus on VRMs.
Jan 27 21:51:52 zaius systemd[1]: Stopped Inventory Upload.
Jan 27 21:51:52 zaius systemd[1]: Starting Turn VCS rails off...
Jan 27 21:51:52 zaius systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
Jan 27 21:51:52 zaius systemd[1]: Starting Wait for Power0 to turn off...
Jan 27 21:51:52 zaius systemd[1]: Stopped Turn VCS rails on.
Jan 27 21:51:52 zaius systemd[1]: Stopped Run VCS workaround on host0.
Jan 27 21:51:52 zaius systemd[1]: Stopped Fix up Zaius host refclock.
Jan 27 21:51:52 zaius systemd[1]: Stopped target Power0 (On).
Jan 27 21:51:52 zaius systemd[1]: Stopped target Power0 On.
Jan 27 21:51:52 zaius zaius_vcs.sh[1383]: Turning off VCS
Jan 27 21:51:52 zaius zaius_vcs.sh[1383]:         Unbinding UCD driver 0-0064
Jan 27 21:51:52 zaius zaius_vcs.sh[1383]:         Setting UCD GPIO 5 to 0x3
Jan 27 21:51:52 zaius zaius_vcs.sh[1383]:         Setting UCD GPIO 6 to 0x3
Jan 27 21:51:52 zaius systemd[1]: Stopped Wait for Power0 to turn on.
Jan 27 21:51:52 zaius systemd[1]: Stopped Start Power0.
Jan 27 21:51:52 zaius systemd[1]: Starting Disable AVSBus on VRMs...
Jan 27 21:51:52 zaius zaius_vcs.sh[1383]:         Binding UCD driver 0-0064
Jan 27 21:51:52 zaius zaius_avsbus.sh[1385]: Disabling AVSBus on bus 7 VRM @0x60 rail 0x01...
Jan 27 21:51:52 zaius kernel: ucd9000 0-0064: Device ID UCD90160|2.3.4.0000|110603
Jan 27 21:51:52 zaius kernel[791]: ucd9000 0-0064: Device ID UCD90160|2.3.4.0000|110603
Jan 27 21:51:52 zaius zaius_avsbus.sh[1385]: Disabling AVSBus on bus 7 VRM @0x64 rail 0x01...
Jan 27 21:51:52 zaius zaius_avsbus.sh[1385]: Disabling AVSBus on bus 7 VRM @0x64 rail 0x00...
Jan 27 21:51:52 zaius zaius_avsbus.sh[1385]: Disabling AVSBus on bus 8 VRM @0x60 rail 0x01...
Jan 27 21:51:52 zaius zaius_avsbus.sh[1385]: Disabling AVSBus on bus 8 VRM @0x64 rail 0x01...
Jan 27 21:51:53 zaius zaius_avsbus.sh[1385]: Disabling AVSBus on bus 8 VRM @0x64 rail 0x00...
Jan 27 21:51:53 zaius systemd[1]: Started Turn VCS rails off.
Jan 27 21:51:53 zaius systemd[1]: Started Disable AVSBus on VRMs.
Jan 27 21:51:53 zaius systemd[1]: Reached target Power0 Off (Pre).
Jan 27 21:51:53 zaius systemd[1]: Started Stop Power0.
Jan 27 21:51:53 zaius power_control.exe[880]: PowerControl: setting power up SOFTWARE_PGOOD to 0
Jan 27 21:51:53 zaius power_control.exe[880]: PowerControl: setting power up BMC_POWER_UP to 0
Jan 27 21:51:53 zaius system_manager.py[843]: Running System State: HOST_POWERING_OFF
Jan 27 21:51:53 zaius timemanager[878]:  PGOOD has changed..
Jan 27 21:51:54 zaius phosphor-chassis-state-manager[881]: Pgood has turned off
Jan 27 21:51:54 zaius phosphor-chassis-state-manager[881]: Change to Chassis Power State
Jan 27 21:51:54 zaius phosphor-host-state-manager[884]: Change to Host State
Jan 27 21:51:54 zaius system_manager.py[843]: Running System State: HOST_POWERED_OFF
Jan 27 21:51:54 zaius phosphor-host-state-manager[884]: Reached final state
Jan 27 21:51:55 zaius timemanager[878]:  User Settings have changed..
Jan 27 21:51:55 zaius chassis_control.py[792]: Turn on power and boot
Jan 27 21:51:55 zaius timemanager[878]:  PGOOD has changed..
Jan 27 21:51:55 zaius timemanager[878]: Changing time settings allowed now
Jan 27 21:51:55 zaius zaius_avsbus.sh[1433]: Enabling AVSBus on bus 7 VRM @0x60 rail 0x01...
Jan 27 21:51:55 zaius systemd[1]: Starting Enable AVSBus on VRMs...
Jan 27 21:51:55 zaius systemd[1]: Starting Wait for Power0 to turn on...
Jan 27 21:51:55 zaius systemd[1]: Started Start Power0.
Jan 27 21:51:55 zaius systemd[1]: Stopped target Power0 Off.
Jan 27 21:51:55 zaius power_control.exe[880]: PowerControl: setting power up SOFTWARE_PGOOD to 1
Jan 27 21:51:55 zaius power_control.exe[880]: PowerControl: setting power up BMC_POWER_UP to 1
Jan 27 21:51:55 zaius system_manager.py[843]: Running System State: HOST_POWERING_ON
Jan 27 21:51:55 zaius timemanager[878]:  PGOOD has changed..
Jan 27 21:51:55 zaius systemd[1]: Starting Inventory Upload...
Jan 27 21:51:55 zaius systemd[1]: Stopped target Chassis0 (Stop).
Jan 27 21:51:55 zaius systemd[1]: Stopped Reload mboxd after power off.
Jan 27 21:51:55 zaius systemd[1]: Stopped Stop Power0.
Jan 27 21:51:55 zaius systemd[1]: Stopped target Power0 Off (Pre).
Jan 27 21:51:55 zaius systemd[1]: Stopped Disable AVSBus on VRMs.
Jan 27 21:51:55 zaius systemd[1]: Stopped Turn VCS rails off.
[... 37 lines cut ...]
Jan 27 21:51:59 zaius systemd[1]: Starting Turn VCS rails on...
Jan 27 21:51:59 zaius zaius_vcs.sh[1498]: Turning on VCS
Jan 27 21:51:59 zaius zaius_vcs.sh[1498]:         Unbinding UCD driver 0-0064
Jan 27 21:51:59 zaius zaius_vcs.sh[1498]:         Setting UCD GPIO 5 to 0x7
Jan 27 21:51:59 zaius zaius_vcs.sh[1498]:         Setting UCD GPIO 6 to 0x7
Jan 27 21:51:59 zaius zaius_vcs.sh[1498]:         Binding UCD driver 0-0064
Jan 27 21:51:59 zaius kernel: ucd9000 0-0064: Device ID UCD90160|2.3.4.0000|110603
[... cut ...]
Jan 27 21:52:00 zaius systemd[1]: Reached target Start Host0 (Pre).
Jan 27 21:52:00 zaius systemd[1]: Starting Start host0 SBE (pdbg)...
Jan 27 21:52:01 zaius start_host.sh[1515]: p0:0x2918: 0x00000000eeeef30c
Jan 27 21:52:01 zaius start_host.sh[1515]: p0:0x283f: 0x0000000020000000
Jan 27 21:52:01 zaius start_host.sh[1515]: p0:0x2801: 0x0000000084c00002
Jan 27 21:52:01 zaius systemd[1]: Started Start host0 SBE (pdbg).
Jan 27 21:52:01 zaius systemd[1]: Reached target Chassis0 (Start).
Jan 27 21:52:01 zaius systemd[1]: Reached target Start Host0.
```

geissonator, 2017-01-27:
It's not pretty, but it works - https://gerrit.openbmc-project.xyz/#/c/2111/

You can directly apply to the file system if you want to just copy my change directly over /usr/sbin/chassis_control.py and reboot or run "systemctl restart org.openbmc.control.Chassis@0.service" at BMC_READY.

We also had to disable the hostboot eSEL processing, that was causing ipmid to crash during power on:
rm /usr/lib/host-ipmid/libstrgfnhandler.*
rm /usr/lib/host-ipmid/liboemhandler.*

geissonator, 2017-02-01:
@spinler is going to take my patch and to a .bbappend for witherspoon and zaius as a temporary workaround.  

We'll use this issue to track the final fix.  

@cyrilbur-ibm I have some new code that ensures we correctly run the "Reload mboxd after power off" but when I ran the reboot, I still saw an issue where the host does not start on the reboot (all services to start the host run fine, but some sort of error must be hit).  I was using the master code that had the performance improvements, but the power off and on happens extremely quickly.

You can see in some of the trace blurbs that the start and stop of the service happens very quickly, and is near the start of the inventory upload service.

```
Jan 31 19:50:42 witherspoon systemd[1]: Started Reload mboxd after power off.
Jan 31 19:50:43 witherspoon systemd[1]: Reached target Chassis0 (Power Off).
...
Jan 31 19:50:43 witherspoon systemd[1]: Starting Wait for Power0 to turn on...
Jan 31 19:50:43 witherspoon systemd[1]: Starting Disable UCD VCS rails...
Jan 31 19:50:43 witherspoon systemd[1]: Starting Inventory Upload...
Jan 31 19:50:43 witherspoon systemd[1]: Stopped Reload mboxd after power off.
...
Jan 31 19:50:58 witherspoon systemd[1]: Starting Start host0 SBE (pdbg)...

```

One thing to try may be to remove this line from the service file:
RemainAfterExit=yes

That may be conflicting with the oneshot option, we need to make sure this service completes before we let anything else continue.

geissonator, 2017-02-02:
https://gerrit.openbmc-project.xyz/#/c/2211/ is what we need to remove once we get this fixed.

GHF, 2017-02-06:
Thank you for debugging Andrew!

As for the blip, does it have anything to do with mboxd-reload@.service having
```
Conflicts=obmc-power-chassis-on@%i.target
```
?

I could imagine the reboot interrupting the kill used to send the command.

geissonator, 2017-02-07:
Definitely the issue with the current code but with my new code, that guarantees we run the target, the service has "Type=oneshot" which means the corresponding target wont end until the command returns (the killall) so the signal should have at least been sent.

geissonator, 2017-02-28:
@cyrilbur-ibm any updates on this issue?  With recent code changes to use host state, we've now lost the hack that put the sleep in place (https://gerrit.openbmc-project.xyz/#/c/2211/).

closed 2017-04-19 by williamspatrick