openbmc/openbmc #1050: [Witherspoon v1.99.1-51-g947e967]  org.openbmc.control.led.service: Main process exited, code=exited, status=234/MAKE_STARTER

gkeishin, 2017-01-31:
```Jan 30 02:48:01 witherspoon systemd[1]: org.openbmc.control.led.service: Main process exited, code=exited, status=234/MAKE_STARTER```
Jan 30 02:48:01 witherspoon systemd[1]: Failed to start Phosphor LED Management Daemon.
Jan 30 02:48:01 witherspoon systemd[1]: org.openbmc.control.led.service: Unit entered failed state.
Jan 30 02:48:01 witherspoon systemd[1]: org.openbmc.control.led.service: Failed with result 'exit-code'.
Jan 30 02:48:01 witherspoon systemd[1]: org.openbmc.control.led.service: Service hold-off time over, scheduling restart.
Jan 30 02:48:01 witherspoon systemd[1]: Stopped Phosphor LED Management Daemon.
Jan 30 02:48:01 witherspoon systemd[1]: org.openbmc.control.led.service: Start request repeated too quickly.
```Jan 30 02:48:01 witherspoon systemd[1]: Failed to start Phosphor LED Management Daemon```

- [x]  Journal log  
[journald_v1.99.1-51-g947e967_Leds.txt](https://github.com/openbmc/openbmc/files/741120/journald_v1.99.1-51-g947e967_Leds.txt)

gkeishin, 2017-01-31:
after multiple reboot to standby also, we couldn't see LED manager coming online..

gkeishin, 2017-02-02:
This is pervasive in all newer drivers.. 
SYSTEM     : w37
BUILD_ID   : https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/502/
VERSION_ID : v1.99.1-81-g42a851a
BUILD TIME : 20170202033520

systemctl status org.openbmc.control.led.service

● org.openbmc.control.led.service - Phosphor LED Management Daemon
   Loaded: loaded (/lib/systemd/system/org.openbmc.control.led.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Mon 2017-01-30 02:47:23 UTC; 23s ago
  Process: 801 ExecStart=/usr/sbin/led_controller.exe [[0;1;31m(code=exited, status=234/MAKE_STARTER)[[0m
 Main PID: 801 (code=exited, status=234/MAKE_STARTER)

Jan 30 02:47:23 witherspoon systemd[1]: [[0;1;39morg.openbmc.control.led.service: Main process exited, code=exited, status=234/MAKE_STARTER[[0m
Jan 30 02:47:23 witherspoon systemd[1]: [[0;1;31mFailed to start Phosphor LED Management Daemon.[[0m
Jan 30 02:47:23 witherspoon systemd[1]: [[0;1;39morg.openbmc.control.led.service: Unit entered failed state.[[0m
Jan 30 02:47:23 witherspoon systemd[1]: [[0;1;39morg.openbmc.control.led.service: Failed with result 'exit-code'.[[0m
Jan 30 02:47:23 witherspoon systemd[1]: org.openbmc.control.led.service: Service hold-off time over, scheduling restart.
Jan 30 02:47:23 witherspoon systemd[1]: Stopped Phosphor LED Management Daemon.
Jan 30 02:47:23 witherspoon systemd[1]: [[0;1;39morg.openbmc.control.led.service: Start request repeated too quickly.[[0m
Jan 30 02:47:23 witherspoon systemd[1]: [[0;1;31mFailed to start Phosphor LED Management Daemon.[[0m

vishwabmc, 2017-02-02:
I got on the system in question and saw that 3 LEDs were actually populated so this is not same as openbmc/openbmc#705.

Traces indicated that it failed to add objects to dbus however, I see this trace is duplicated between 

add_object_manager _and_ add_object_vtable so I can not really tell where it really failed.

That said, this daemon would go away since the LED management has been re-factored now and this should go away with openbmc/openbmc#802

So can this be closed now ?

gkeishin, 2017-02-02:
Sounds good to me @vishwabmc  .. however lets keep this as a tracker, will close once we verified the changes are in. Thanks

vishwabmc, 2017-03-23:
@gkeishin https://gerrit.openbmc-project.xyz/#/c/3223/ should remove this old one

gkeishin, 2017-03-24:
@vishwabmc Thats great.. Thanks for update.. looking forward to it..

vishwabmc, 2017-03-27:
This can definitely be closed now with https://gerrit.openbmc-project.xyz/#/c/3223/ already merged

geissonator, 2017-03-27:
@gkeishin can you confirm testing was good with this change so we can close out?

gkeishin, 2017-03-27:
@geissonator I didn't check this one.. let me check on the system real quick..

gkeishin, 2017-03-27:
Not see in latest build.. closing this one out

closed 2017-03-27 by gkeishin