openbmc/openbmc #1054: Document the openbmc target design and flow within systemd

geissonator, 2017-01-31:
It's difficult to visualize and understand the flow of targets on the openbmc using systemd.  There's multiple ways out there to document this, we need to pick one and implement it across all of our targets.

Here's a comment from Brad on it:
https://gerrit.openbmc-project.xyz/#/c/2115/1/meta-phosphor/common/recipes-core/systemd/obmc-targets/obmc-quiesce-system%2540.target

Also, see the mailing list discussion where @amboar generated a systemd-analyze pic.
https://lists.ozlabs.org/pipermail/openbmc/2017-January/006309.html

geissonator, 2018-01-03:
https://github.com/openbmc/docs/blob/master/openbmc-systemd.md is where the base doc is now.

rfrandse, 2018-08-20:
moved to https://github.com/ibm-openbmc/dev/issues/105

closed 2018-08-20 by rfrandse