openbmc/openbmc #1057: runqemu support for openbmc machine types

kc8apf, 2017-02-01:
To run qemu for a palmetto or witherspoon today, qemu must be started directly with flags to set the machine types, etc.  Upstream Yocto's runqemu support setting the default flags for new machine types through configuration files.  Once #1056 is resolved, add palmetto and witherspoon runqemu config files so runqemu can be used to start a VM with an image.

amboar, 2017-02-02:
Also had patches for this prior to upstream changing to their configuration-based qemu invocation. The `runqemu` scripts were pretty hellish and the change was invasive so we never merged it. Happy for someone to pick up the batton and run with getting it done under the new configuration-based scheme!

amboar, 2019-04-02:
Still needs to be done. It would be good to schedule this, and then rework the docs repository accordingly. It should clean up a lot of the qemu noise in howtos.

closed 2020-11-05 by stale[bot]