openbmc/openbmc #1062: [Witherspoon w37 v1.99.1-82-gbfd622e ] Journal filling up with HMON warnings

gkeishin, 2017-02-02:
```Jan 30 03:12:25 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Jan 30 03:12:25 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon1/
Jan 30 03:12:25 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon2/
Jan 30 03:12:25 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon3/
Jan 30 03:12:25 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon4/
Jan 30 03:12:55 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Jan 30 03:12:55 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon1/
Jan 30 03:12:55 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon2/
Jan 30 03:12:55 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon3/
Jan 30 03:12:55 witherspoon hwmon.py[825]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon4/
```

gkeishin, 2017-02-02:
This is just one snapshot Journal Log To Standby - [x]  
[journald_v1.99.1-82-gbfd622e_WARNING.txt](https://github.com/openbmc/openbmc/files/747098/journald_v1.99.1-82-gbfd622e_WARNING.txt)

When the System poweron's its more ...

spinler, 2017-02-02:
@bradbishop's new hwmon code must not be enabled yet, as hwmon.py will be removed.

bradbishop, 2017-02-28:
I got rid of this warning here: https://gerrit.openbmc-project.xyz/#/c/2590/

gkeishin, 2017-02-28:
@bradbishop Thats awesome.. I ll verify this in next build and close it up.. Thanks

gkeishin, 2017-03-01:
Verified..

closed 2017-03-01 by gkeishin