openbmc/openbmc #1064: [palmetto VERSION_ID="v1.99.1-82-gbfd622e-dirty"] BMC reached to unpingable state after running basic poweron test

SrideviRamesh, 2017-02-02:
Issue:

BMC reached to unpingable state after running basic poweron test

Log Info from serial port console:

```
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: NCSI interface down
net_ratelimit: 4 callbacks suppressed
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: [ISR] = 0xf: NO_RXBUF RPKT_LOST
ftgmac100 1e660000.ethernet eth0: NCSI interface down

```

System Info:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.1-82"
VERSION_ID="v1.99.1-82-gbfd622e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.1-82"
BUILD_ID="v1.99.1"

please find attached FFDC and serial port console logs
[20170202011023638885_powerontest.zip](https://github.com/openbmc/openbmc/files/747333/20170202011023638885_powerontest.zip)
[palmetto_serial_port_logs.txt](https://github.com/openbmc/openbmc/files/747336/palmetto_serial_port_logs.txt)

gwshan, 2017-02-03:
@SrideviRamesh, thanks for your testing and reporting. The RxRing in the MAC has been eaten up, but no more RxBD is put back to the ring. It's why the ingress frames (including NCSI packets) are dropped. The speculation is race condition in the driver. If the BMC is in failing state, I can poke for more. Please let me know, thanks!

If it's not in failing state any more, I will try to recreate it on palmetto in local lab.

gwshan, 2017-02-03:
@SrideviRamesh I will try to recreate this issue in local lab. Please confirm how many ethernet cables connected to this box? I want to know if both (two) NCSI channels have ethernet cable connected to L2 switch.

SrideviRamesh, 2017-02-08:
same issue is observed with latest palmetto version "v1.99.1-100"
please find attached serial port logs 
[serial_port_logs.docx](https://github.com/openbmc/openbmc/files/760186/serial_port_logs.docx)

SrideviRamesh, 2017-02-08:
@gwshan i have checked using 'ifconfig -a' command on BMC. i observe IP is connected to eth0 only.. I did not find any other network adapter

gwshan, 2017-02-09:
@SrideviRamesh, Well, I wasn't asking the number of network interfaces that have been brought up. I was asking how many ethernet cables between the box and its ethernet switch. I've started the attached script as a systemd service on a local palmetto machine to see if I can recreate the issue successfully.

[ncsi.txt](https://github.com/openbmc/openbmc/files/762600/ncsi.txt)

gtmills, 2018-05-31:
@SrideviRamesh @gwshan This issue hasn't been updated in over a year. Still, a problem or can this be closed?

SrideviRamesh, 2018-07-06:
This issue is no longer observed.. hence closing the issue

closed 2018-07-06 by gtmills