openbmc/openbmc #1065: [Barreleye v1.99.1-current] REST "org/openbmc/sensors/" view listing gone ?

yadron, 2017-02-02:
If using current "master" build, then REST API for sensors returns nothing:

```
curl -b cjar -k https://172.17.27.61/org/openbmc/sensors/
{
  "data": [], 
  "message": "200 OK", 
  "status": "ok"
}
```

However I still have access to sub-tree of org/openbmc/sensors/ _(but it is hard to guess that "sensors/" exists, so first I thought that sensors doesn't work at all in REST API)_:

```
curl -b cjar -k https://172.17.27.61/org/openbmc/senors/speed/
{
  "data": [
    "/org/openbmc/sensors/speed/fan1", 
    "/org/openbmc/sensors/speed/fan0", 
    "/org/openbmc/sensors/speed/fan3", 
    "/org/openbmc/sensors/speed/fan2", 
    "/org/openbmc/sensors/speed/fan5", 
    "/org/openbmc/sensors/speed/fan4"
  ], 
  "message": "200 OK", 
  "status": "ok"
}
```

If using "stable" (v1.0-stable) build, then sensors returns expected list:

```
{
  "data": [
    "/org/openbmc/sensors/HSCB", 
    "/org/openbmc/sensors/HSCC", 
    "/org/openbmc/sensors/HSCA", 
    "/org/openbmc/sensors/temperature", 
    "/org/openbmc/sensors/host", 
    "/org/openbmc/sensors/voltage", 
    "/org/openbmc/sensors/speed"
  ], 
  "message": "200 OK", 
  "status": "ok"
}

```

gtmills, 2017-02-06:
From Brad, guessing there is a process not running because it died or something
```ps |grep sensor
ps|grep hwmon
```
you should see:
```python /usr/sbin/hwmon.py
python /usr/sbin/sensor_manager2.py```

yadron, 2017-02-08:
No, please check again. Both programs exists:

```
root@barreleye:~# ps|grep sensor
  701 root     251:37 python /usr/sbin/sensor_manager2.py
 1216 root       0:00 grep sensor

root@barreleye:~# ps|grep hwmon
  773 root     466:39 python /usr/sbin/hwmon.py
 1218 root       0:00 grep hwmon

```
Try these commands at yours configuration (for checking if this issue could be reproduced not only at Barreleye) one after other for connecting to BMC and retrieving sensors list:
```
curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "0penBmc" ] }' https://172.17.27.61/login
curl -b cjar -k https://172.17.27.61/org/openbmc/sensors/
```

- if it returns empty answer that is
```
{
  "data": [], 
  "message": "200 OK", 
  "status": "ok"
}

```
like I have, then something wrong inside these programs.

The strange thing that third command for getting specific sensor works Ok (see below) even the list of sensors directory empty (see above "data []"):

```
curl -b cjar -k https://172.17.27.61/org/openbmc/sensors/speed/
{
  "data": [
    "/org/openbmc/sensors/speed/fan1", 
    "/org/openbmc/sensors/speed/fan0", 
    "/org/openbmc/sensors/speed/fan3", 
    "/org/openbmc/sensors/speed/fan2", 
    "/org/openbmc/sensors/speed/fan5", 
    "/org/openbmc/sensors/speed/fan4"
  ], 
  "message": "200 OK", 
  "status": "ok"

```

yadron, 2017-02-08:
For testing and both understanding better all the things, I've modified original REST API program (taken from stable, but works the same at master) so it reads list of values from DBUS and output to the screen without any 443 port listening.

You may try at BMC this program, attached ( like "root@barreleye:~# python obmc-rest2.txt" ), see  [obmc-rest2.txt](https://github.com/openbmc/openbmc/files/761056/obmc-rest2.txt)

(if line n 147 that is "def test(self, path=.." is == '/org/openbmc/', then output
"```
VALUE  /org/openbmc/watchdog
VALUE  /org/openbmc/HostServices
VALUE  /org/openbmc/inventory
VALUE  /org/openbmc/TimeManager
VALUE  /org/openbmc/buttons
VALUE  /org/openbmc/control
VALUE  /org/openbmc/sensors
```"
, and if == '/org/openbmc/sensors/', then output empty
`""`
- this a problem that DBUS currently doesn't answer request "list directory /org/openbmc/sensors/" correctly).

The program that I modified taken from stable and in master already named differently
https://github.com/openbmc/phosphor-rest-server/blob/master/module/obmc/wsgi/apps/rest_dbus.py

bradbishop, 2017-06-23:
I think this fix will help some here: https://gerrit.openbmc-project.xyz/4932

yadron, 2017-06-28:
Looks like suggested patch (https://gerrit.openbmc-project.xyz/4932) doesn't help (I've checked that manually).

Debugging the code:

```
if calling
/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py(64)get_subtree_paths()
, it calls
self.iface.GetSubTreePaths with params p path
'/org/openbmc/sensors/'
(Pdb) p depth
1
(Pdb) p interfaces
[]
,
and then 
/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py(38)retry()

-- so looks like something wrong getting dbus method GetSubTreePaths of "xyz.openbmc_project.ObjectMapper" or maybe it is being called correctly (then returns nothing for sensors)

<dbus.lowlevel.MethodCallMessage path: /xyz/openbmc_project/object_mapper, iface: xyz.openbmc_project.ObjectMapper, member: GetSubTreePaths dest: :1.20>
```

Ok, now I know how to call it directly:
```
root# busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetSubTreePaths sias /org/openbmc/ 1 0   

as 5 "/org/openbmc/sensors" "/org/openbmc/control" "/org/openbmc/inventory" "/org/openbmc/buttons" "/org/openbmc/TimeManager"
```

, but this call returns nothing (that is the problem):

```
# busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetSubTreePaths sias /org/openbmc/sensors/ 1 0
as 0
```

- please fix that.

yadron, 2017-06-28:
Sorry, I was wrong in my previous comment. After restarting affected service
```
(
kill phosphor-mapper
)
```
it starts working (so just required manual reload of my manually fixed code
`"/usr/lib/python2.7/site-packages/obmc/mapper/server.py"
`
):

```
root:~# busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetSubTreePaths sias /org/openbmc/sensors/ 1 0

as 2 "/org/openbmc/sensors/host" "/org/openbmc/sensors/temperature"
```

- so yes, this patch (https://gerrit.openbmc-project.xyz/4932) fixes the problem and can be commited.

gtmills, 2018-05-23:
Closing this since https://gerrit.openbmc-project.xyz/#/c/4932/ fixed the problem.

closed 2018-05-23 by gtmills