openbmc/openbmc #1066: [Witherspoon w37 v1.99.1-82-gbfd622e ]  Host Sensor Property BootProgress shows Off

gkeishin, 2017-02-02:
The Host OS has booted however the BootProgress show its Off..

```
$curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET  https://xx.xx.xx.xx/org/openbmc/sensors/host/BootProgress
{
  "data": {
    "error": 0, 
    "units": "", 
    "value": "Off"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```
- [x] Journal Log  
[journald_v1.99.1-82-gbfd622e_BootProgress.txt](https://github.com/openbmc/openbmc/files/747524/journald_v1.99.1-82-gbfd622e_BootProgress.txt)

**Couple of questions here:**
1. BootProgress "Off" is this expected now ?  we normally see's some string like this "FW Progress, Starting OS"

2. Now the Power on/ off interface is available in xyz as well.. are there going to be equivalent for BootProgress for Host in xyz ?

geissonator, 2017-02-02:
I believe we need to get the HOST IPMI interface enabled to get this sensor populated.  That bringup work is imminent.  Will have to talk with @bradbishop about a new xyz. sensor value for it.

spinler, 2017-02-16:
Just need to update the sensors list in Witherspoon.py to match what's in the MRW/hostboot.

spinler, 2017-02-20:
With my change:

```    "/org/openbmc/sensors/host/BootCount": {
      "error": 0, 
      "units": "", 
      "value": 2
    }, 
    "/org/openbmc/sensors/host/BootProgress": {
      "error": 0, 
      "units": "", 
      "value": "FW Progress, Starting OS"
    }, 
```

spinler, 2017-02-20:
https://gerrit.openbmc-project.xyz/#/c/2550/

gkeishin, 2017-03-10:
Closing as this change working is verified on driver.. Thanks

closed 2017-03-10 by gkeishin