openbmc/openbmc #1067: [Witherspoon w37 v1.99.1-82-gbfd622e ]  Dead Services ?

gkeishin, 2017-02-02:
What are these dead non functional services doing here on the system.. anyone ?

```root@witherspoon:~# systemctl --no-page -t service -a | grep dead
● dropbear.service                                          not-found inactive dead    dropbear.service
  dropbear@0.service                                        loaded    inactive dead    SSH Per-Connection Server
  dropbearkey.service                                       loaded    inactive dead    SSH Key Generation
  emergency.service                                         loaded    inactive dead    Emergency Shell
  getty@tty1.service                                        loaded    inactive dead    Getty on tty1
  kmod-static-nodes.service                                 loaded    inactive dead    Create list of required static device nodes for the current kernel
  mapper-wait@-xyz-openbmc_project-state-bmc.service        loaded    inactive dead    Wait for /xyz/openbmc_project/state/bmc
  mapper-wait@-xyz-openbmc_project-state-chassis.service    loaded    inactive dead    Wait for /xyz/openbmc_project/state/chassis
  mapper-wait@-xyz-openbmc_project-state-host.service       loaded    inactive dead    Wait for /xyz/openbmc_project/state/host
  obmc-console-ssh@0.service                                loaded    inactive dead    Phosphor Host Console SSH Per-Connection
  obmc-flash-init.service                                   loaded    inactive dead    Initialize BIOS Flash
  obmc-mgr-state.service                                    loaded    inactive dead    Restore Last State
  org.openbmc.control.led.service                           loaded    inactive dead    Phosphor LED Management Daemon
  phosphor-gevent.service                                   loaded    inactive dead    Phosphor Webserver
● plymouth-quit-wait.service                                not-found inactive dead    plymouth-quit-wait.service
● plymouth-start.service                                    not-found inactive dead    plymouth-start.service
  rc-local.service                                          loaded    inactive dead    /etc/rc.local Compatibility
  rescue.service                                            loaded    inactive dead    Rescue Shell
  rest-dbus.service                                         loaded    inactive dead    Phosphor DBUS Browser
  run-postinsts.service                                     loaded    inactive dead    Run pending postinsts
  systemd-ask-password-console.service                      loaded    inactive dead    Dispatch Password Requests to Console
  systemd-ask-password-wall.service                         loaded    inactive dead    Forward Password Requests to Wall
  systemd-firstboot.service                                 loaded    inactive dead    First Boot Wizard
  systemd-fsck-root.service                                 loaded    inactive dead    File System Check on Root Device
  systemd-hostnamed.service                                 loaded    inactive dead    Hostname Service
● systemd-hwdb-update.service                               masked    inactive dead    systemd-hwdb-update.service
  systemd-initctl.service                                   loaded    inactive dead    /dev/initctl Compatibility Daemon
  systemd-journal-catalog-update.service                    loaded    inactive dead    Rebuild Journal Catalog
  systemd-machine-id-commit.service                         loaded    inactive dead    Commit a transient machine-id on disk
  systemd-modules-load.service                              loaded    inactive dead    Load Kernel Modules
  systemd-sysusers.service                                  loaded    inactive dead    Create System Users
  systemd-timedated.service                                 loaded    inactive dead    Time & Date Service
  systemd-tmpfiles-clean.service                            loaded    inactive dead    Cleanup of Temporary Directories
  systemd-update-done.service                               loaded    inactive dead    Update is Completed
  systemd-vconsole-setup.service                            loaded    inactive dead    Setup Virtual Console
  var-volatile-lib.service                                  loaded    inactive dead    Bind mount volatile /var/lib
```

gtmills, 2017-02-06:
Most of these will go away when we get rid of https://github.com/openbmc/skeleton

gkeishin, 2017-02-06:
@gtmills Awesome.. thanks

williamspatrick, 2017-03-30:
Many of these will always be "dead" as a default state. This is not a problem.

closed 2017-03-30 by williamspatrick