openbmc/openbmc #1069: [Witherspoon w37 v1.99.1-83-g82472ba] Bad chip id Error

gkeishin, 2017-02-02:
During multiple reboot to standby test.. I observed this... 
```
Jan 30 02:47:01 witherspoon kernel: lm75 9-004a: hwmon0: sensor 'tmp275'
Jan 30 02:47:01 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jan 30 02:47:01 witherspoon kernel: bmp280 3-0077: bad chip id.  expected 55 got 58
Jan 30 02:47:01 witherspoon kernel: bmp280: probe of 3-0077 failed with error -22
```

- [x]  Journal  
[journald_v1.99.1-83-g82472ba_chip_error.txt](https://github.com/openbmc/openbmc/files/748295/journald_v1.99.1-83-g82472ba_chip_error.txt)

spinler, 2017-02-02:
0x58 is the actual chip ID register of the BMP280 temp sensor.  0x55 is the chip ID of the BMP180, which is unfortunately present on some Witherspoons in the lab that apparently got a FIR card from Firestone instead of a Teak card.  I think that led someone to put a bmp180 in the device tree:
```
	bmp180@77 {
		compatible = "bosch,bmp180";
		reg = <0x77>;
	};
```

I would have hoped the same driver could support both.  Apparently that isn't the case

spinler, 2017-02-02:
Unfortunately, if I'm looking at bmp280.c right, the driver doesn't support hwmon anyway, which is a bummer.  On the upside, this chip isn't shipping anyway and will be replaced by one with a later EOL.

spinler, 2017-02-09:
When the commits to generate the device tree from the MRW get merged,  the BMP280 will show up in the device tree instead of the 180.

As for my other comment - this sensor is in the IIO subsystem instead of hwmon, and there appears to be a iio-hwmon driver we can load to bridge it over to hwmon.  We'll worry about that when the fan code gets closer to needing to use that temp in its algorithms.

gkeishin, 2017-03-31:
@spinler We still see this.. anything on this one..

spinler, 2017-03-31:
@bradbishop has a fix coming

gkeishin, 2017-06-22:
Don't see this on latest build closing it.

closed 2017-06-22 by gkeishin