openbmc/openbmc #1070: [Witherspoon w37 v1.99.1-83-g82472ba]  REST reboot hangs BMC

gkeishin, 2017-02-02:
The new REST interface xyz reboot seems buggy.

/xyz/openbmc_project/state/BMC0//attr/RequestedBMCTransition, method:Put, args:{u'data': {u'data': 'xyz.openbmc_project.State.BMC.Transition.Reboot'}

BMC reboots and hangs in boots.. 

1.  Serial console freezes
2.  SSH to BMC connection refused
3.  But system ping is alive

gkeishin, 2017-02-02:
This is also seen on Barreleye too.

sivassrr, 2017-02-06:
@JoshDKing Could you check whether https://github.com/openbmc/openbmc/issues/1084 is something related to this.

JoshDKing, 2017-02-23:
@gkeishin Are we still seeing this?  There has been a few updates surrounding this that might have taken care of it.  I don't seem to have any problems with it freezing after a reboot.

gkeishin, 2017-02-23:
@JoshDKing I m not seeing it this anymore.. Verified working.. Thanks

closed 2017-02-23 by gkeishin