openbmc/openbmc #1072: [Witherspoon w37 v1.99.1-94-g7d19e6c]  Failed to start Phosphor BMC State Manager

gkeishin, 2017-02-03:
```Jan 31 18:56:26 witherspoon phosphor-host-state-manager[838]: Initial Host State will be Off
Jan 31 18:56:26 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Main process exited, code=dumped, status=6/ABRT
Jan 31 18:56:26 witherspoon phosphor-bmc-state-manager[864]: Assertion '!bus->current_slot' failed at ../git/src/libsystemd/sd-bus/sd-bus.c:2760, function bus_process_internal(). Aborting.
Jan 31 18:56:26 witherspoon systemd[1]: Failed to start Phosphor BMC State Manager.
Jan 31 18:56:26 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Unit entered failed state.
Jan 31 18:56:26 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Failed with result 'core-dump'.
```

- [x] Journal log  
[journald_v1.99.1-94-g7d19e6c_coredump_asset.txt](https://github.com/openbmc/openbmc/files/749553/journald_v1.99.1-94-g7d19e6c_coredump_asset.txt)

gkeishin, 2017-02-03:
This failure is pervasive in latest builds tested
https://openpower.xyz/job/openbmc-build/508/
https://openpower.xyz/job/openbmc-build/509/

gkeishin, 2017-02-03:
Can be seen on latest as well

Latest build : https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/510/
   VERSION_ID : v1.99.1-96-g2a46570
   BUILD TIME : 20170203124939

rahulmah, 2017-02-07:
I believe due to this issue we are seeing below REST request failing for BMC state in recent builds(v1.99.1-99-g160919d/20170206225937). I am not logging the issue for this. If separate issue needs to be logged, let me know. 

```curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/BMC0                                                              {
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/state/BMC0"
  },
  "message": "404 Not Found",
  "status": "error"}```

geissonator, 2017-02-08:
Finally got the latest SDK building the latest code and did a few quick tests.  The fail is happening when we call the generated base class
```
sdbusplus::server::object::object<
      sdbusplus::xyz::openbmc_project::State::server::BMC>(bus, objPath)
```

If we pass the optional "true" parameter then we don't hit the error, but if in the constructor we call this->emit_object_added() then we do hit the assert.  I took a quick look at the generated header and don't see anything obvious.  The Host and Chassis code, which does very similar code patterns does not hit this issue.

Will probably need some @williamspatrick help on this one.
```
[barreleye]# /tmp/phosphor-bmc-state-manager 
Assertion '!bus->current_slot' failed at ../git/src/libsystemd/sd-bus/sd-bus.c:2760, function bus_process_internal(). Aborting.
Aborted (core dumped)
```

JoshDKing, 2017-02-09:
So it looks like that the version the system needs to be on is:
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-12"
VERSION_ID="v1.99.2-12-g3d4dc92-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-12"
BUILD_ID="v1.99.2"

I tried it on a few different systems after I built the latest versions onto them and it looks like I am not having any issues.
Please let me know if I'm missing something, but after updating the systems to the latest versions, it appears to work.

rahulmah, 2017-02-10:
I tested with latest release build - v1.99.2-13-gf598488/20170210013109/#520 build and I still see the problem.

```
rahulmah@rackspace:~$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/                                                                   {
  "data": [
    "/xyz/openbmc_project/state/host0",
    "/xyz/openbmc_project/state/chassis0"
  ],
  "message": "200 OK",
  "status": "ok"
}rahulmah@rackspace:~$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/BMC0                                                               {
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/state/BMC0"
  },
  "message": "404 Not Found",
  "status": "error"
}rahulmah@rackspace:~$```

geissonator, 2017-02-13:
Yeah, for some reason this is back now.
```
[barreleye]# phosphor-bmc-state-manager 
Assertion '!bus->current_slot' failed at ../git/src/libsystemd/sd-bus/sd-bus.c:2760, function bus_process_internal(). Aborting.
Aborted (core dumped)
[barreleye]# 
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-30"
VERSION_ID="v1.99.2-30-gb7d3a49-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-30"
BUILD_ID="v1.99.2"
```

sivassrr, 2017-02-14:
yup, I could see it in today's version [535](https://openpower.xyz/job/openbmc-build/535/)

geissonator, 2017-02-14:
of course 2 builds later and it's working again...
```
[barreleye]# ps -ef | grep phosphor-bmc  
  790 root       0:00 /usr/sbin/phosphor-bmc-state-manager
  880 root       0:00 grep phosphor-bmc
[barreleye]# 
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-32"
VERSION_ID="v1.99.2-32-g71526c2-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-32"
BUILD_ID="v1.99.2"
```

gkeishin, 2017-02-14:
I have not seen this working in the builds in tested.. for instance the latest one
https://openpower.xyz/job/openbmc-build/537/

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-35"
VERSION_ID="v1.99.2-35-ga0491c2-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-35"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170214120856
root@witherspoon:~# systemctl status xyz.openbmc_project.State.BMC.service | cat
● xyz.openbmc_project.State.BMC.service - Phosphor BMC State Manager
   Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.State.BMC.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: core-dump) since Tue 2017-01-31 18:56:49 UTC; 3min 32s ago
  Process: 902 ExecStart=/usr/sbin/phosphor-bmc-state-manager (code=dumped, signal=ABRT)
 Main PID: 902 (code=dumped, signal=ABRT)

Jan 31 18:56:49 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Main process exited, code=dumped, status=6/ABRT
Jan 31 18:56:49 witherspoon systemd[1]: Failed to start Phosphor BMC State Manager.
Jan 31 18:56:49 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Unit entered failed state.
Jan 31 18:56:49 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Failed with result 'core-dump'.
Jan 31 18:56:49 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Service hold-off time over, scheduling restart.
Jan 31 18:56:49 witherspoon systemd[1]: Stopped Phosphor BMC State Manager.
Jan 31 18:56:49 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Start request repeated too quickly.
Jan 31 18:56:49 witherspoon systemd[1]: Failed to start Phosphor BMC State Manager.
root@witherspoon:~#

```

gkeishin, 2017-02-16:
This is not seen in last few builds... @JoshDKing @geissonator  any fix went in for this ?  let me know if we are good to close this one out..

closed 2017-02-20 by williamspatrick