openbmc/openbmc #1073: [Witherspoon w37 - v1.99.1-81-g42a851a] Host OS fails to Boot

gkeishin, 2017-02-03:
1. Poweron to Host OS boot
2. Login to OS and issue sudo reboot
3. OS send down reboot signal to BMC via OPAL but never came back Online

Jan 31 19:05:14 witherspoon chassis_control.py[752]: Rebooting
Jan 31 19:05:14 witherspoon chassis_control.py[752]: Turn off power
Jan 31 19:05:14 witherspoon systemd[1]: Stopped target Chassis0 (Start).
Jan 31 19:05:14 witherspoon phosphor-host-state-manager[823]: Recieved signal that host is running
Jan 31 19:05:14 witherspoon phosphor-host-state-manager[823]: Change to Host State

Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .[  299.305089] reboot: Restarting system
Ubuntu 17.04.  .  .  .[  343.087531840,5] OPAL: Reboot request...

-bash-4.1$ ssh -p 220 root@w37
ssh: connect to host w37 port 220: Connection refused

- [x] Journal  
[journald_v1.99.1-94-g7d19e6c_OS_fails.txt](https://github.com/openbmc/openbmc/files/749567/journald_v1.99.1-94-g7d19e6c_OS_fails.txt)

gkeishin, 2017-02-03:
I noticed that after sometime the BMC stops responding as well..

amboar, 2017-02-03:
Known issue. Should be worked-around by https://gerrit.openbmc-project.xyz/#/c/2211/

gkeishin, 2017-02-03:
Ok.. I ll keep watch ahead in newer build.

geissonator, 2017-02-03:
The fix went in with https://github.com/openbmc/openbmc/commit/bfd622e4a8405950dacb87eb45dc1bef04c97ab9 so this driver has that fix.  Was there anything in the console indicating why the BMC became unresponsive?

gkeishin, 2017-02-03:
@geissonator  Not sure.. I just took the Journald and it hung bit latter.. so i don't know what really happened there.. Not network issue though since I was able to access other systems at the time..

gkeishin, 2017-02-03:
Current:
    SYSTEM : Witherspoon (w37)
    Latest build : https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/510/
    VERSION_ID : v1.99.1-96-g2a46570
    BUILD TIME : 20170203124939

On this build I no longer see the issue.. Thanks

gkeishin, 2017-02-06:
We seems to see this as well while validating

https://gerrit.openbmc-project.xyz/#/c/2260/

Build https://openpower.xyz/job/openbmc-build-gerrit-trigger-multi/1880/

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.1-98"
VERSION_ID="v1.99.1-98-g7ae3839-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.1-98"
BUILD_ID="v1.99.1"
root@witherspoon:~# cat /etc/version
20170206024836
```

gkeishin, 2017-02-06:
More data continuation to above build..

@amboar Took a look at the system and per Andrew

In theory as soon as we see:
```
Jan 31 19:15:34 witherspoon mboxd[1475]: [ 1485890134.717490595] Pointing HOST LPC bus at the actual flash
Jan 31 19:15:34 witherspoon mboxd[1475]: [ 1485890134.717625342] Assuming 64MB of flash: HOST LPC 0x0c000000
```
we should be good to power the host
```
# pdbg -a -b fsi -d witherspoon getcfam 0x1007
p0:0x1007: 0x00000000a02c80f4
```
indicates we've check stopped the host.

- [x]  Journal 
[journald_v1.99.1-98-g7ae3839_build1880_20170206024836.txt](https://github.com/openbmc/openbmc/files/753764/journald_v1.99.1-98-g7ae3839_build1880_20170206024836.txt)

sivassrr, 2017-02-14:
Just Test - Some reason conversion of github issue to CSV throw error because of this issue.

gkeishin, 2017-03-31:
I was able to validate this working on, issue reboot on OS and the BMC Powered on back and OS is booted back.

```
- [x] SYSTEM     : W50 ( Witherspoon - 2 Sockets - Pass 2 / 16 DIMMs)
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/673/
- [x] VERSION_ID : v1.99.3-189-gb33f84c
- [x] Kernel     : Linux witherspoon 4.7.10-6786cab8778ba96d84d89c7aec34374b5f4f1968 #1 Thu Mar 30 21:21:56 UTC 2017 armv6l GNU/Linux

- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.14
        op-build-v1.15-75-g986c911-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-6a3a09c
        hostboot-a58f695
        linux-4.10.6-openpower1-pef4dc3d
        petitboot-v1.4.2-pe78e1ef
        witherspoon-xml-b208e4f
        occ-4141b5f
        hostboot-binaries-82b82cd
        capp-ucode-1bb7503
```

closed 2017-03-31 by gkeishin