openbmc/openbmc #1074: [Witherspoon w37 v1.99.1-95-g807fdd2] obmc-flash-init.service: Failed

gkeishin, 2017-02-03:
```
Jan 31 18:56:24 witherspoon systemd[1]: obmc-flash-init.service: Main process exited, code=exited, status=1/FAILURE
Jan 31 18:56:24 witherspoon systemd[1]: obmc-flash-init.service: Unit entered failed state.
Jan 31 18:56:24 witherspoon systemd[1]: obmc-flash-init.service: Failed with result 'exit-code
```

root@witherspoon:~# systemctl status obmc-flash-init.service | cat
● obmc-flash-init.service - Initialize BIOS Flash
   Loaded: loaded (/lib/systemd/system/obmc-flash-init.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2017-01-31 18:56:24 UTC; 9min ago
  Process: 815 ExecStart=/usr/sbin/mapper call /org/openbmc/control/flash/bios org.openbmc.Flash init (code=exited, status=1/FAILURE)
 Main PID: 815 (code=exited, status=1/FAILURE)

Jan 31 18:56:14 witherspoon systemd[1]: Started Initialize BIOS Flash.
Jan 31 18:56:24 witherspoon mapper[815]: Error finding '/org/openbmc/control/flash/bios' service: Device or resource busy
Jan 31 18:56:24 witherspoon systemd[1]: obmc-flash-init.service: Main process exited, code=exited, status=1/FAILURE
Jan 31 18:56:24 witherspoon systemd[1]: obmc-flash-init.service: Unit entered failed state.
Jan 31 18:56:24 witherspoon systemd[1]: obmc-flash-init.service: Failed with result 'exit-code'

- [x]  Journal log  
[journald_v1.99.1-95-g807fdd2_flashinit.txt](https://github.com/openbmc/openbmc/files/749981/journald_v1.99.1-95-g807fdd2_flashinit.txt)

amboar, 2017-02-06:
@gkeishin Please use the triple  back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

gkeishin, 2017-02-17:
@vishwabmc We see this in latest build as well

**Driver info**
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-66"
VERSION_ID="v1.99.2-66-g1488f75-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-66"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170217003621
root@witherspoon:~#
```

```
● obmc-flash-init.service - Initialize BIOS Flash
   Loaded: loaded (/lib/systemd/system/obmc-flash-init.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2017-01-31 18:56:45 UTC; 2min 44s ago
  Process: 836 ExecStart=/usr/sbin/mapper call /org/openbmc/control/flash/bios org.openbmc.Flash init (code=exited, status=1/FAILURE)
 Main PID: 836 (code=exited, status=1/FAILURE)

Jan 31 18:56:37 witherspoon systemd[1]: Started Initialize BIOS Flash.
Jan 31 18:56:45 witherspoon mapper[836]: Error finding '/org/openbmc/control/flash/bios' service: Device or resource busy
Jan 31 18:56:45 witherspoon systemd[1]: obmc-flash-init.service: Main process exited, code=exited, status=1/FAILURE
Jan 31 18:56:45 witherspoon systemd[1]: obmc-flash-init.service: Unit entered failed state.
Jan 31 18:56:45 witherspoon systemd[1]: obmc-flash-init.service: Failed with result 'exit-code'.
root@witherspoon:~# 
```

- [x] journald log  
[journald_v1.99.2-66-g1488f75_20170217003621_obmc-flash-init.service.txt](https://github.com/openbmc/openbmc/files/782384/journald_v1.99.2-66-g1488f75_20170217003621_obmc-flash-init.service.txt)

gkeishin, 2017-03-31:
No longer observed since many builds.. closing one out

closed 2017-03-31 by gkeishin