openbmc/openbmc #1078: [Witherspoon w37 _v1.99.1-96-g2a46570] Power Restore policy and Host OS starting Issue

gkeishin, 2017-02-03:
This is more of a Host starting issue if we look at the bellow sequence of data.

Power Policy applied ALWAYS_POWER_ON, Host powering on but host OS not booted post AC cycle BMC reset. 

1. Set Power Policy to ALWAYS_POWER_ON

https://9.3.185.173/org/openbmc/settings/host0/attr/power_policy, method:Get, args:{}
Response code:200, Content:{
  "data": "ALWAYS_POWER_ON", 
  "message": "200 OK", 
  "status": "ok"
}

2. Power Cycle issued

3. BMC boots, Power policy applied 

Jan 31 18:56:42 witherspoon discover_system_state.py[878]: Last System State:1Power Policy:ALWAYS_POWER_ON

4. Host Poweron in progress

root@witherspoon:~# obmcutil state
 = HOST_POWERED_ON
root@witherspoon:~#

5. Host recieved signal but not booted OS 

Jan 31 18:57:05 witherspoon phosphor-host-state-manager[839]: Recieved signal that host is running
Jan 31 18:57:05 witherspoon phosphor-host-state-manager[839]: Change to Host State
Jan 31 18:57:05 witherspoon systemd[1]: Startup finished in 6.683s (kernel) + 1min 38.067s (userspace) = 1min 44.751s.
Jan 31 18:57:27 witherspoon mboxd[1091]: [ 1485889047.213927738] Setting all MBOX regs to 0xff individually...
Jan 31 18:57:27 witherspoon mboxd[1091]: [ 1485889047.218231212] Entering polling loop

spinler, 2017-02-03:
If I look at the cfam status register:
```
# pdbg -a -b fsi -d p9w getcfam 0x1007
p0:0x1007: 0x00000000a02c80f4
```
That is a TI (terminate immediate) triggered by host code
also, not sure why pdbg can't see both chips

spinler, 2017-02-03:
We also saw:
```Jan 31 18:56:59 witherspoon systemd[1]: dev-aspeed-lpc-ctrl.device: Job dev-aspeed-lpc-ctrl.device/start timed out.
Jan 31 18:56:59 witherspoon systemd[1]: Timed out waiting for device dev-aspeed-lpc-ctrl.device.
Jan 31 18:56:59 witherspoon systemd[1]: dev-aspeed-lpc-ctrl.device: Job dev-aspeed-lpc-ctrl.device/start failed with result 'timeout'.
Jan 31 18:56:59 witherspoon systemd[1]: dev-aspeed-mbox.device: Job dev-aspeed-mbox.device/start timed out.
Jan 31 18:56:59 witherspoon systemd[1]: Timed out waiting for device dev-aspeed-mbox.device.
Jan 31 18:56:59 witherspoon systemd[1]: dev-aspeed-mbox.device: Job dev-aspeed-mbox.device/start failed with result 'timeout'.
Jan 31 18:56:59 witherspoon systemd[1]: Started Phosphor MBOX Daemon.
Jan 31 18:56:59 witherspoon systemd[1]: Reached target Phosphor System - Standby.```

amboar, 2017-02-05:
@spinler I'm looking into the mboxd device issue. It seems to be the gift that keeps on giving.

amboar, 2017-03-20:
As a note, I expect this is a race between mboxd and the host for access to the flash controller. Suraj from our team is working to resolve it.

gkeishin, 2017-03-31:
@amboar Anything on this or we can close this one out ?

gkeishin, 2017-03-31:
This looks good on

```
- [x] SYSTEM     : W50 ( Witherspoon - 2 Sockets - Pass 2 / 16 DIMMs)
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/673/
- [x] VERSION_ID : v1.99.3-189-gb33f84c
- [x] Kernel     : Linux witherspoon 4.7.10-6786cab8778ba96d84d89c7aec34374b5f4f1968 #1 Thu Mar 30 21:21:56 UTC 2017 armv6l GNU/Linux

- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.14
        op-build-v1.15-75-g986c911-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-6a3a09c
        hostboot-a58f695
        linux-4.10.6-openpower1-pef4dc3d
        petitboot-v1.4.2-pe78e1ef
        witherspoon-xml-b208e4f
        occ-4141b5f
        hostboot-binaries-82b82cd
        capp-ucode-1bb7503
```

closed 2017-03-31 by gkeishin