openbmc/openbmc #1083: FSP I2C driver broken

shenki, 2017-02-06:
When doing a probe for the second time:

```
PROBE    eng[00f5112c]Adding ports[0x0000000000000fff] to eng[0x00f5112c]
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1200 at /var/lib/jenkins/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
sysfs: cannot create duplicate filename '/devices/virtual/iic-fsi/i2cfsi00'
CPU: 0 PID: 1200 Comm: pdbg Not tainted 4.7.10-517d997328ab0d56b1fccb9e58cd5cd71090a8f3 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cc0>] (__warn+0xc4/0xf0)
[<80110cc0>] (__warn) from [<80110d30>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d30>] (warn_slowpath_fmt) from [<801f88dc>] (sysfs_warn_dup+0x50/0x70)
[<801f88dc>] (sysfs_warn_dup) from [<801f8970>] (sysfs_create_dir_ns+0x74/0x88)
[<801f8970>] (sysfs_create_dir_ns) from [<8026f3e0>] (kobject_add_internal+0x108/0x30c)
[<8026f3e0>] (kobject_add_internal) from [<8026f660>] (kobject_add+0x7c/0xb4)
[<8026f660>] (kobject_add) from [<802b2a50>] (device_add+0x118/0x53c)
[<802b2a50>] (device_add) from [<802b2f88>] (device_create_groups_vargs+0x7c/0xc8)
[<802b2f88>] (device_create_groups_vargs) from [<802b2ff4>] (device_create_vargs+0x20/0x28)
[<802b2ff4>] (device_create_vargs) from [<802b3024>] (device_create+0x28/0x48)
[<802b3024>] (device_create) from [<8030af58>] (iic_create_bus+0x124/0x18c)
[<8030af58>] (iic_create_bus) from [<8030841c>] (iic_add_ports+0x98/0x15c)
[<8030841c>] (iic_add_ports) from [<80308988>] (iic_fsi_probe+0xd8/0x170)
[<80308988>] (iic_fsi_probe) from [<802b5324>] (driver_probe_device+0x1b0/0x3d8)
[<802b5324>] (driver_probe_device) from [<802b3bd4>] (bus_for_each_drv+0x90/0xb8)
[<802b3bd4>] (bus_for_each_drv) from [<802b4fd0>] (__device_attach+0x7c/0x110)
[<802b4fd0>] (__device_attach) from [<802b3da0>] (bus_probe_device+0x28/0x80)
[<802b3da0>] (bus_probe_device) from [<802b2c04>] (device_add+0x2cc/0x53c)
[<802b2c04>] (device_add) from [<80306d24>] (fsi_master_register+0x4b0/0x5c4)
[<80306d24>] (fsi_master_register) from [<803072b0>] (store_scan+0x7c/0x90)
[<803072b0>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 19897464222f8555 ]---
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1200 at /var/lib/jenkins/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/lib/kobject.c:240 kobject_add_internal+0x290/0x30c
kobject_add_internal failed for i2cfsi00 with -EEXIST, don't try to register things with the same name in the same directory.
CPU: 0 PID: 1200 Comm: pdbg Tainted: G        W       4.7.10-517d997328ab0d56b1fccb9e58cd5cd71090a8f3 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cc0>] (__warn+0xc4/0xf0)
[<80110cc0>] (__warn) from [<80110d30>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d30>] (warn_slowpath_fmt) from [<8026f568>] (kobject_add_internal+0x290/0x30c)
[<8026f568>] (kobject_add_internal) from [<8026f660>] (kobject_add+0x7c/0xb4)
[<8026f660>] (kobject_add) from [<802b2a50>] (device_add+0x118/0x53c)
[<802b2a50>] (device_add) from [<802b2f88>] (device_create_groups_vargs+0x7c/0xc8)
[<802b2f88>] (device_create_groups_vargs) from [<802b2ff4>] (device_create_vargs+0x20/0x28)
[<802b2ff4>] (device_create_vargs) from [<802b3024>] (device_create+0x28/0x48)
[<802b3024>] (device_create) from [<8030af58>] (iic_create_bus+0x124/0x18c)
[<8030af58>] (iic_create_bus) from [<8030841c>] (iic_add_ports+0x98/0x15c)
[<8030841c>] (iic_add_ports) from [<80308988>] (iic_fsi_probe+0xd8/0x170)
[<80308988>] (iic_fsi_probe) from [<802b5324>] (driver_probe_device+0x1b0/0x3d8)
[<802b5324>] (driver_probe_device) from [<802b3bd4>] (bus_for_each_drv+0x90/0xb8)
[<802b3bd4>] (bus_for_each_drv) from [<802b4fd0>] (__device_attach+0x7c/0x110)
[<802b4fd0>] (__device_attach) from [<802b3da0>] (bus_probe_device+0x28/0x80)
[<802b3da0>] (bus_probe_device) from [<802b2c04>] (device_add+0x2cc/0x53c)
[<802b2c04>] (device_add) from [<80306d24>] (fsi_master_register+0x4b0/0x5c4)
[<80306d24>] (fsi_master_register) from [<803072b0>] (store_scan+0x7c/0x90)
[<803072b0>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 19897464222f8556 ]---
bus[00000000] created
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1200 at /var/lib/jenkins/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
sysfs: cannot create duplicate filename '/devices/virtual/iic-fsi/i2cfsi01'
[... 665 lines cut ...]
[<80105558>] (show_stack) from [<80110cc0>] (__warn+0xc4/0xf0)
[<80110cc0>] (__warn) from [<80110d30>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d30>] (warn_slowpath_fmt) from [<8026f568>] (kobject_add_internal+0x290/0x30c)
[<8026f568>] (kobject_add_internal) from [<8026f660>] (kobject_add+0x7c/0xb4)
[<8026f660>] (kobject_add) from [<802b2a50>] (device_add+0x118/0x53c)
[<802b2a50>] (device_add) from [<802b2f88>] (device_create_groups_vargs+0x7c/0xc8)
[<802b2f88>] (device_create_groups_vargs) from [<802b2ff4>] (device_create_vargs+0x20/0x28)
[<802b2ff4>] (device_create_vargs) from [<802b3024>] (device_create+0x28/0x48)
[<802b3024>] (device_create) from [<8030af58>] (iic_create_bus+0x124/0x18c)
[<8030af58>] (iic_create_bus) from [<8030841c>] (iic_add_ports+0x98/0x15c)
[<8030841c>] (iic_add_ports) from [<80308988>] (iic_fsi_probe+0xd8/0x170)
[<80308988>] (iic_fsi_probe) from [<802b5324>] (driver_probe_device+0x1b0/0x3d8)
[<802b5324>] (driver_probe_device) from [<802b3bd4>] (bus_for_each_drv+0x90/0xb8)
[<802b3bd4>] (bus_for_each_drv) from [<802b4fd0>] (__device_attach+0x7c/0x110)
[<802b4fd0>] (__device_attach) from [<802b3da0>] (bus_probe_device+0x28/0x80)
[<802b3da0>] (bus_probe_device) from [<802b2c04>] (device_add+0x2cc/0x53c)
[<802b2c04>] (device_add) from [<80306d24>] (fsi_master_register+0x4b0/0x5c4)
[<80306d24>] (fsi_master_register) from [<803072b0>] (store_scan+0x7c/0x90)
[<803072b0>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace 19897464222f856c ]---
```

shenki, 2017-02-13:
With Eddie's patch applied the issue still appears:

```
root@zaius5:~# echo 1 > /sys/devices/platform/fsi-master/scan
 slave@00:00: engine[2]: type 4, version 6, addr c00 size 400
 slave@00:00: engine[3]: type 5, version 6, addr 1000 size 400
 slave@00:00: engine[4]: type 6, version 1, addr 1400 size 400
 slave@00:00: engine[5]: type 7, version 2, addr 1800 size 400
PROBE    eng[00f5112c]
Adding ports[0x0000000000007fff] to eng[0x00f5112c]
------------[ cut here ]------------
WARNING: CPU: 0 PID: 923 at /var/lib/jenkins/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
sysfs: cannot create duplicate filename '/devices/virtual/iic-fsi/i2cfsi00'
CPU: 0 PID: 923 Comm: sh Tainted: G        W       4.7.10-a5491ea9d3e962daab2587fc33616b9aa5c3257a #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cc0>] (__warn+0xc4/0xf0)
[<80110cc0>] (__warn) from [<80110d30>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d30>] (warn_slowpath_fmt) from [<801f88dc>] (sysfs_warn_dup+0x50/0x70)
[<801f88dc>] (sysfs_warn_dup) from [<801f8970>] (sysfs_create_dir_ns+0x74/0x88)
[<801f8970>] (sysfs_create_dir_ns) from [<8026f3e0>] (kobject_add_internal+0x108/0x30c)
[<8026f3e0>] (kobject_add_internal) from [<8026f660>] (kobject_add+0x7c/0xb4)
[<8026f660>] (kobject_add) from [<802b2a50>] (device_add+0x118/0x53c)
[<802b2a50>] (device_add) from [<802b2f88>] (device_create_groups_vargs+0x7c/0xc8)
[<802b2f88>] (device_create_groups_vargs) from [<802b2ff4>] (device_create_vargs+0x20/0x28)
[<802b2ff4>] (device_create_vargs) from [<802b3024>] (device_create+0x28/0x48)
[<802b3024>] (device_create) from [<8030ad4c>] (iic_create_bus+0xc0/0x128)
[<8030ad4c>] (iic_create_bus) from [<80308394>] (iic_add_ports+0x98/0x15c)
[<80308394>] (iic_add_ports) from [<80308858>] (iic_fsi_probe+0xd0/0x14c)
[<80308858>] (iic_fsi_probe) from [<802b5324>] (driver_probe_device+0x1b0/0x3d8)
[<802b5324>] (driver_probe_device) from [<802b3bd4>] (bus_for_each_drv+0x90/0xb8)
[<802b3bd4>] (bus_for_each_drv) from [<802b4fd0>] (__device_attach+0x7c/0x110)
[<802b4fd0>] (__device_attach) from [<802b3da0>] (bus_probe_device+0x28/0x80)
[<802b3da0>] (bus_probe_device) from [<802b2c04>] (device_add+0x2cc/0x53c)
[<802b2c04>] (device_add) from [<80306d24>] (fsi_master_register+0x4b0/0x5c4)
[<80306d24>] (fsi_master_register) from [<803072b0>] (store_scan+0x7c/0x90)
[<803072b0>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
---[ end trace e06f654f8814e83c ]---
------------[ cut here ]------------
WARNING: CPU: 0 PID: 923 at /var/lib/jenkins/workspace/openbmc-build-gerrit-trigger-multi/distro/ubuntu/target/zaius/openbmc/build/tmp/work-shared/zaius/kernel-source/lib/kobject.c:240 kobject_add_internal+0x290/0x30c
kobject_add_internal failed for i2cfsi00 with -EEXIST, don't try to register things with the same name in the same directory.
CPU: 0 PID: 923 Comm: sh Tainted: G        W       4.7.10-a5491ea9d3e962daab2587fc33616b9aa5c3257a #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<80110cc0>] (__warn+0xc4/0xf0)
[<80110cc0>] (__warn) from [<80110d30>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d30>] (warn_slowpath_fmt) from [<8026f568>] (kobject_add_internal+0x290/0x30c)
[<8026f568>] (kobject_add_internal) from [<8026f660>] (kobject_add+0x7c/0xb4)
[<8026f660>] (kobject_add) from [<802b2a50>] (device_add+0x118/0x53c)
[<802b2a50>] (device_add) from [<802b2f88>] (device_create_groups_vargs+0x7c/0xc8)
[<802b2f88>] (device_create_groups_vargs) from [<802b2ff4>] (device_create_vargs+0x20/0x28)
[<802b2ff4>] (device_create_vargs) from [<802b3024>] (device_create+0x28/0x48)
[<802b3024>] (device_create) from [<8030ad4c>] (iic_create_bus+0xc0/0x128)
[<8030ad4c>] (iic_create_bus) from [<80308394>] (iic_add_ports+0x98/0x15c)
[<80308394>] (iic_add_ports) from [<80308858>] (iic_fsi_probe+0xd0/0x14c)
[<80308858>] (iic_fsi_probe) from [<802b5324>] (driver_probe_device+0x1b0/0x3d8)
[<802b5324>] (driver_probe_device) from [<802b3bd4>] (bus_for_each_drv+0x90/0xb8)
[<802b3bd4>] (bus_for_each_drv) from [<802b4fd0>] (__device_attach+0x7c/0x110)
[<802b4fd0>] (__device_attach) from [<802b3da0>] (bus_probe_device+0x28/0x80)
[<802b3da0>] (bus_probe_device) from [<802b2c04>] (device_add+0x2cc/0x53c)
[<802b2c04>] (device_add) from [<80306d24>] (fsi_master_register+0x4b0/0x5c4)
[<80306d24>] (fsi_master_register) from [<803072b0>] (store_scan+0x7c/0x90)
[<803072b0>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[... 878 lines cut ...]
[<803072b0>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[... cut ...]
 slave@00:00: engine[14]: type 1b, version 1, addr 40000 size 8000
 slave@00:00: engine[15]: type 1b, version 1, addr 48000 size 8000
 slave@00:00: engine[16]: type 1b, version 1, addr 50000 size 8000
 slave@00:00: engine[17]: type 1b, version 1, addr 58000 size 8000
 slave@00:00: engine[18]: type 1b, version 1, addr 60000 size 8000
 slave@00:00: engine[19]: type 1b, version 1, addr 68000 size 8000
 slave@00:00: engine[20]: type 1b, version 1, addr 70000 size 8000
 slave@00:00: engine[21]: type 1b, version 1, addr 78000 size 8000
```

eddiejames, 2017-02-13:
Just put some patches on the mailing list for this issue.

rfrandse, 2017-03-27:
Eddie said this is done

closed 2017-03-27 by rfrandse