openbmc/openbmc #1084: [Witherspoon - v1.99.1-96-g2a46570] Dr510 - BMC hang with WATCHDOG=1 error

sivassrr, 2017-02-06:
Had run loop test with the latest driver  [510](https://openpower.xyz/job/openbmc-build/510/)

Driver Details:
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.1-96"
VERSION_ID="v1.99.1-96-g2a46570-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.1-96"
BUILD_ID="v1.99.1"
root@witherspoon:# obmcutil state
 = HOST_POWERED_ON

```
First few iterations, I was able to do SSH etc and was able to use serial connection as well.
After 10th iteration, Found that SSH was not responding and serial port freezes with following text

Message from Serial prompt
[  OK  ] Stopped Phosphor LED Management Daemon.
[FAILED] Failed to start Phosphor LED Management Daemon.
See 'systemctl status org.openbmc.control.led.service' for details.
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4
witherspoon login: random: nonblocking pool is initialized
ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
systemd-journald[439]: Failed to send WATCHDOG=1 notification message: Connection refused
systemd-journald[439]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
```

Bunch of this error on serial output. 
[WitherspoonCombinedReport10Dr510.pdf](https://github.com/openbmc/openbmc/files/753708/WitherspoonCombinedReport10Dr510.pdf)
[dr510Journal.txt](https://github.com/openbmc/openbmc/files/753709/dr510Journal.txt)

sivassrr, 2017-02-06:
systemd-journald[439]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected - Generally associate with BMC hang due to some kind of file system corruption or application trying to acccess the file system after it is unmounted.

vishwabmc, 2017-02-13:
Related : https://github.com/systemd/systemd/issues/2275 but with the level of systemd that we use, this should have been fixed ?

vishwabmc, 2017-02-13:
https://bbs.archlinux.org/viewtopic.php?id=205856

@amboar Could you help here ?

williamspatrick, 2017-05-30:
We upgraded systemd a while back.

closed 2017-05-30 by williamspatrick