openbmc/openbmc #1085: FSI driver breaks after read to bad address

shenki, 2017-02-06:
I'm using  Zaius system to bring up a OpenFSI pdbg backend. I'm using `/sys/devices/platform/fsi-master/slave@00:00/raw`.

When I do a read from `0xc24`, it succeeds. I can leave the system idle for some time and come back, and it still works.

As soon as I do a read from an invalid address, such as `0x80c24`, the read will fail and every subsequent read will fail until I re-scan the bus with `echo 1 > scan`.

```
root@zaius5v:/sys/devices/platform/fsi-master/slave@00:00# dd if=raw bs=4 count=1 skip=$((0xc09 / 4)) | hexdump -C -v
00000000  00 00 00 00                                       |....|
00000004
1+0 records in
1+0 records out
root@zaius5v:/sys/devices/platform/fsi-master/slave@00:00# dd if=raw bs=4 count=1 skip=$((0xc24 / 4)) | hexdump -C -v
1+0 records in
1+0 records out
00000000  49 10 0d 12                                       |I...|
00000004
root@zaius5v:/sys/devices/platform/fsi-master/slave@00:00# dd if=raw bs=4 count=1 skip=$((0x80c24 / 4)) | hexdump -C -v
dd: raw: Input/output error
root@zaius5v:/sys/devices/platform/fsi-master/slave@00:00# dd if=raw bs=4 count=1 skip=$((0xc24 / 4)) | hexdump -C -v
dd: raw: Input/output error
```

shenki, 2017-02-06:
This patch appears to help: http://patchwork.ozlabs.org/patch/723897/

However, I think we should be rejecting invalid reads/writes where we can in order to avoid resetting the link before every read/write.

cbostic, 2017-02-08:
Fix for issue 1086 will correct this problem as well.

closed 2019-04-27 by stale[bot]