openbmc/openbmc #1088: [Witherspoon v1.99.1-96-g2a46570]  Power Policy Restore Last state not honored

gkeishin, 2017-02-06:
**1. Power Host**
```
  "data": {
    "boot_flags": "default", 
    "boot_policy": "ONETIME", 
    "network_config": "ipaddress=,prefix=,gateway=,mac=,addr_type=", 
    "power_cap": 1000, 
    "power_policy": "RESTORE_LAST_STATE", 
    "restricted_mode": 0, 
    "system_state": "HOST_POWERED_ON", 
    "time_mode": "NTP", 
    "time_owner": "BMC", 
    "use_dhcp_ntp": "yes"
  }, 
```
  
**Policy : RESTORE_LAST_STATE
System state : HOST_POWERED_ON**

**2.  rebooted BMC**

BMC comes online
```  
Jan 31 18:56:30 witherspoon discover_system_state.py[874]: Last System State:1Power Policy:RESTORE_LAST_STATE
Jan 31 18:56:31 witherspoon systemd[1]: Started Restore Last State.

root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~# uptime
 18:57:29 up 2 min,  load average: 2.72, 1.57, 0.61
root@witherspoon:~#
```

3. BMC is at Ready State and not Powering on the Host when the REST system state show powering ON

```
root@witherspoon:~# uptime
 19:02:09 up 6 min,  load average: 0.02, 0.62, 0.45
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~# journalctl --no-pager > journald_v1.99.1-96-g2a46570_RESTORE_LAST_STATE.txt
root@witherspoon:~#

  "data": {
    "boot_flags": "default", 
    "boot_policy": "ONETIME", 
    "network_config": "ipaddress=,prefix=,gateway=,mac=,addr_type=", 
    "power_cap": 1000, 
    "power_policy": "RESTORE_LAST_STATE", 
    "restricted_mode": 0, 
    "system_state": "HOST_POWERED_ON", 
    "time_mode": "NTP", 
    "time_owner": "BMC", 
    "use_dhcp_ntp": "yes"
  }
```
- [x] Journal log  
[journald_v1.99.1-96-g2a46570_RESTORE_LAST_STATE.txt](https://github.com/openbmc/openbmc/files/754887/journald_v1.99.1-96-g2a46570_RESTORE_LAST_STATE.txt)

vishwabmc, 2017-02-13:
Jan 31 18:56:30 witherspoon discover_system_state.py[874]: Last System State:1Power Policy:RESTORE_LAST_STATE

Jan 31 18:56:31 witherspoon systemd[1]: Started Restore Last State.

SO this says that we did do the right thing of powering it back on
==============================================================

@JoshDKing Can you please help take a look into this ?.

Is 'obmcutil state` now using the new BMC state manager ? If so, I see it is not started
```
Jan 31 18:56:29 witherspoon systemd[1]: Failed to start Phosphor BMC State Manager.
Jan 31 18:56:29 witherspoon phosphor-bmc-state-manager[885]: Assertion '!bus->current_slot' failed at ../git/src/libsystemd/sd-bus/sd-bus.c:2760, function bus_process_internal(). Aborting.
Jan 31 18:56:29 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Unit entered failed state.
Jan 31 18:56:29 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Failed with result 'core-dump'.
Jan 31 18:56:30 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Service hold-off time over, scheduling restart.
Jan 31 18:56:30 witherspoon systemd[1]: Stopped Phosphor BMC State Manager.
Jan 31 18:56:30 witherspoon systemd[1]: xyz.openbmc_project.State.BMC.service: Start request repeated too quickly.```

vishwabmc, 2017-02-13:
@JoshDKing I looked at latest skeleton/obmcutil and I see

    'state': {
        'bus_name': 'org.openbmc.managers.System',
        'object_name': '/org/openbmc/managers/System',
        'interface_name': 'org.openbmc.managers.System',
        'method': 'getSystemState',
},

Is this the right thing to have in there ?. Do we not need this updated to say BMCState ?

vishwabmc, 2017-02-13:
@geissonator Can the defect screeners get permission to assign the issues to ? Its easier that way.

gkeishin, 2017-03-31:
Worked good with power policy on
```
```
- [x] SYSTEM     : W50 ( Witherspoon - 2 Sockets - Pass 2 / 16 DIMMs)
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/673/
- [x] VERSION_ID : v1.99.3-189-gb33f84c
- [x] Kernel     : Linux witherspoon 4.7.10-6786cab8778ba96d84d89c7aec34374b5f4f1968 #1 Thu Mar 30 21:21:56 UTC 2017 armv6l GNU/Linux

- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.14
        op-build-v1.15-75-g986c911-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-6a3a09c
        hostboot-a58f695
        linux-4.10.6-openpower1-pef4dc3d
        petitboot-v1.4.2-pe78e1ef
        witherspoon-xml-b208e4f
        occ-4141b5f
        hostboot-binaries-82b82cd
        capp-ucode-1bb7503
```

closed 2017-03-31 by gkeishin