openbmc/openbmc #1089: [Barreleye v1.99.1-current] Need IPMI for retrieving sensors data

yadron, 2017-02-06:
Hi

I would like remote IPMI sensors reading (for example, I need this program reading sensors from Barreleye:
https://github.com/open-power/amester ).

I know that IPMI works only locally now, but it doesn't work as expected even locally. I think that at least it should read sensors like for old dell 1850 (example attached 
[try_read_dell1850_sensors_linux_ipmi.txt](https://github.com/openbmc/openbmc/files/755182/try_read_dell1850_sensors_linux_ipmi.txt)
 ).

The only sensors cmd that works now is:

[root@localhost ~]# ipmitool raw 0x04 0x2d 0x09
 02 00 00 00

- this one returns number of boots (that is /org/openbmc/sensors/host/BootCount )

How to enable other sensors cmds?
Very likely that if nobody does it already, I'll continue developing IPMI commands myself, so if you know more or already do this, please tell here or share original similar tasks/issues/bugs.

If you think no need to develop it now, please tell it here too.

Thank you.

Just some additional info (I've written it as some notes for myself):

First, install openipmi to host if not installed yet (e.g. "yum install openipmi openipmi-tools").
IPMI available only internally (see [](https://github.com/openbmc/ipmitool/issues/3) and [](https://libraries.io/github/openbmc/ipmitool).)
Howeve,r you may run IPMI commands (ipmitool) from petitboot or from host or you can monitor IPMI commands from BMC console using this command:

`root@barreleye:/# ipmid -d 0xff`

; and running ipmitool from host:

```
[root@obmc ~]# yum install OpenIPMI OpenIPMI-tools
[root@obmc ~]# modprobe ipmi_msghandler
[root@obmc ~]# modprobe ipmi_watchdog
[root@obmc ~]# modprobe ipmi_poweroff
[root@obmc ~]# modprobe ipmi_devintf
[root@obmc ~]# modprobe ipmi_powernv

```

and check that host to BMC device available then:

```
[root@obmc ~]# ls -l /dev/ipmi0 
crw-------. 1 root root 244, 0 Jan 27 15:06 /dev/ipmi0

[root@localhost ~]# ipmitool sdr type list
Sensor Types:
 Temperature (0x01) Voltage (0x02)
 Current (0x03) Fan (0x04)

```
... (see https://www.ibm.com/developerworks/community/wikis/home?lang=en_us#!/wiki/W51a7ffcf4dfd_4b40_9d82_446ebc23c550/page/Retrieving%20values%20for%20specific%20PowerKVM%20hardware%20sensor%20using%20IPMI 
and
https://openpowerfoundation.org/wp-content/uploads/2016/02/5_Stewart-Smith-for-Shilspari-Bhat.IBM_.pdf 
and https://github.com/open-power/occ/issues/6 )

The commands like this one
`[root@host ~]# ipmitool sensor list -v`
doesn't work.

However, if IPMI command doesn't work, then you may debug it using raw request (e.g. "ipmitool raw 0 1" and both watch output of "ipmid -d 0xff" at BMC), and these docs describes raw IPMI protocol:
[](https://github.com/open-power/docs/blob/master/occ/OCC_ipmitool_sensors.pdf)
[](https://github.com/open-power/docs/blob/master/occ/OCC_OpenPwr_FW_Interfaces.pdf)

If you need IPMItool at BMC also, you may use ipmi_debug.py program as is (take build@servername.com:~/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-sensor/1.0-r1/git/pyipmitest/ipmi_debug.py and comment one line "#    obj.unmask_signals()" and run at BMC; see example for command "0x04 0x2d 0x09" with answser "0x09" that you can run at BMC itself:

```
root@barreleye:/usr/bin# python ~/ipmi_debug.py 
Enter IPMI packet as hex values. First three bytes will be usedas netfn and cmd.
lun will be zero.
**0x04 0x2d 0x09**
IPMI packet from host:
seq: 0x01
netfn: 0x04
lun: 0x00
cmd: 0x2d
data: [**0x09**]
IPMI packet sent to host:
seq: 0x01
netfn: 0x05
lun: 0x00
cmd: 0x2d
cc: 0xc1
data: []
```

yadron, 2017-02-07:
Waiting for some comments from you.

(regarding what I found myself, apart from above GET_SENSOR "ipmitool raw 0x04 0x2d 0x09", also similar SET_SENSOR command works for this sensor num=9: "[root@localhost ~]# ipmitool raw 0x04 0x30 9 0x01 0x00 0x3 0x00 0x00 0x00 0x00 0x00 0x00" so it means SET SENSOR ID=9 to VALUE=3, see fixed issue of SET_SENSOR here https://github.com/openbmc/openbmc/issues/400 );

And more links found related to IPMI or BarrelEye/IPMI:

https://openpowerfoundation.org/wp-content/uploads/2016/02/5_Stewart-Smith-for-Shilspari-Bhat.IBM_.pdf
this describes ipmitool, ipmi, bmc, occ, hardware sensors chain.

https://lists.ozlabs.org/pipermail/openbmc/2016-November/005387.html
this one regarding IP adr change.

https://lists.ozlabs.org/pipermail/openbmc/2016-November/005489.html
https://lists.ozlabs.org/pipermail/openbmc/2016-October/004750.html
for supporting LAN (need to implement "privilege").

https://github.com/openbmc/openbmc/issues/860
issue with "Add Partial eSEL" IPMI command.

https://github.com/openbmc/openbmc/commit/cc1502f3d6c544116ed570b2a02da2cdcdd550c1
kernel conf updated (and CONFIG_ASPEED_BT_IMPI_HOST changed to CONFIG_ASPEED_BT_IPMI_BMC).

yadron, 2017-02-07:
Also I've seen
https://github.com/openbmc/phosphor-net-ipmid
that probably allows NET IPMI, but why is it off in default BarrelyEye buid, so only host-ipmid available? Is it for security reasons only so and will it work if I just add it to build?

kachini, 2017-02-20:
folks, do i read this thread correctly that SDR/sensor IPMI feature will be implemented in March'17 ?

yadron, 2017-02-28:
you've read https://github.com/openbmc/openbmc-test-automation/issues/337 and it says "_Most likely we can automate it in March_" (for me it looks like auto-tests will be developed first or other kind of "automate" in March-April and the development of IPMI SDR/sensors itself can take much longer maybe). I'm telling this here, because I'm interested too when SDR/sensor IPMI will appear? Or when command "ipmitool sensor list -v" will start working?
Currently (BarrelEye v1.99.3-dirty) this command makes BMC saying:
```
No registered handlers for NetFn:[0x4], Cmd:[0x20] trying Wilcard implementation 
IPMI S/E Wildcard Netfn:[0x4], Cmd:[0x20]
ERROR:[0xC1] handling NetFn:[0x4], Cmd:[0x20]
```

All that I know right now is that it could take more time, than month+ of development, because IPMI is complex mix of things (in OpenBMC this SDR/sensor part of IPMI looks like mix of ipmid CPP code see host-ipmid, phosphor-ipmi-host and hwMon Python code, and of course DBUS and hardware configuration can make difference too). Looking to https://github.com/openbmc/openbmc/issues/441 (SDR/sensors topic) and https://github.com/openbmc/openbmc/issues/360 (overall IPMI epic to be done 2.1) I think that SDR/sensors will be in 2.0 (let's say in summer 2017?) and other IPMI in 2.1 (2018?).

yadron, 2017-03-10:
Posted new IPMI command ("Get Device SDR Info") implementation (patches) to https://github.com/openbmc/openbmc/issues/441

gtmills, 2018-05-23:
@yadron Still a problem or can this be closed?

closed 2018-07-18 by gtmills