openbmc/openbmc #1096: obmc-console-server write error on Zaius

shenki, 2017-02-07:
```"v1.99.1-65-g07df806-dirty``` (07df806) on Zaius

When booting the host on Zaius we see this in the journal:

```
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
Feb 06 16:18:58 zaius5v sh[789]: obmc-console-server: Write error: Resource temporarily unavailable
```

Zaius has a physical UART that the host output is mirrored to:

```
root@zaius5v:~# cat /etc/obmc-console.conf 
lpc-address = 0x3f8
sirq = 4
local-tty = ttyS0
```

GHF, 2017-02-07:
See also #992 where there's no output from UART unless obmc-console-client is listening.

jk-ozlabs, 2017-02-07:
This is a consequence of writes to `/dev/ttyS0` returning `EAGAIN` when we're mirroring console data to the local tty device. In order to properly handle it, we'll need to add bufferring to the local tty handler.

In the meantime, if you're not using it, you can disable the `local-tty`option in `obmc-console.conf` to suppress the errors.

jk-ozlabs, 2017-02-09:
This should be fixed by https://gerrit.openbmc-project.xyz/#/c/2320/ (with prior dependent changes). If I give you a obmc-console-server build, would you be able to drop that onto your BMC filesystem and test with that?

If not, what would be easiest to test with?

GHF, 2017-02-09:
I can pull it into a local openbmc/console branch for testing. Thanks @jk-ozlabs

GHF, 2017-02-10:
I tried those changes but it made the console through `obmc-console-client` significantly slower for Zaius. rlippert reports:

> at the current speed there isn't even time to pick an OS before it starts booting :(

referring to the petitboot autoboot timeout.

gkeishin, 2017-05-04:
Seems like a common problem.. Seen in witherspoon too.. 
```
Apr 20 17:37:23 witherspoon obmc-console-server[806]: obmc-console-server: Write error: Resource temporarily unavailable
Apr 20 17:37:23 witherspoon obmc-console-server[806]: obmc-console-server: Write error: Resource temporarily unavailable
Apr 20 17:37:23 witherspoon obmc-console-server[806]: obmc-console-server: Write error: Resource temporarily unavailable
Apr 20 17:37:23 witherspoon obmc-console-server[806]: obmc-console-server: Write error: Resource temporarily unavailable
```

`Driver:  v1.99.5-1-g8c7336f`

mine260309, 2017-05-16:
The same for Romulus.
But we do have fix in gerrit, with https://gerrit.openbmc-project.xyz/#/c/2316 ~ https://gerrit.openbmc-project.xyz/#/c/2322, the issue can be fixed.

But now it has merge conflicts, so hopefully @jk-ozlabs could rebase the changes and let's submit the changes.

gtmills, 2018-08-21:
@shenki @mine260309 @GHF This is an old build and the fixes mentioned above have been merged. Can we close?

closed 2018-08-21 by shenki