openbmc/openbmc #1097: [Witherspoon w38 v1.99.1-97-gd836be9_20170207020541 ] Reboot Fails

gkeishin, 2017-02-07:
**1. BMC at standby**
```
root@witherspoon:~# obmcutil state
 = HOST_POWERED_OFF
root@witherspoon:~# uptime
 11:34:55 up 55 min,  load average: 0.00, 0.01, 0.04
root@witherspoon:~#
```

**2.  Bunch of Journal message started showing up**
```
root@witherspoon:~# journalctl --no-pager | grep systemd-journald
Feb 07 10:39:14 witherspoon systemd-journald[445]: Runtime journal (/run/log/journal/) is 2.6M, max 21.1M, 18.5M free.
Feb 07 10:39:14 witherspoon systemd-journald[445]: Journal started
Feb 07 10:39:19 witherspoon kernel[745]: systemd-journald[445]: Received request to flush runtime journal from PID 1
Feb 07 11:00:14 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Connection refused
Feb 07 11:02:08 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:03:42 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:05:08 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:06:42 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:08:08 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:09:38 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:11:08 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:12:42 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:14:08 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:15:43 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:17:08 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:18:43 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:20:08 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:21:43 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:23:08 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:24:38 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:25:44 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:27:14 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:28:44 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:30:14 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:31:44 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:33:14 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:34:44 witherspoon kernel[745]: systemd-journald[445]: Failed to send WATCHDOG=1 notification message: Transport endpoint is not connected
Feb 07 11:36:14 wit
```

**3. Try to reboot but failed**

```
root@witherspoon:~# reboot
Failed to start reboot.target: Connection timed out
See system logs and 'systemctl status reboot.target' for details.
Failed to open /dev/initctl: No such device or address
Failed to talk to init daemon.
root@witherspoon:~#
```
```
root@witherspoon:~# systemctl status reboot.target | cat
Failed to get properties: Failed to activate service 'org.freedesktop.systemd1': timed out
root@witherspoon:~#
```
**4. Collected the journal but unable to SSH /SCP to the BMC. The BMC can be connected via Serial console to access**

```
rango@ubuntu:~/Desktop$ scp root@xx.xx.xx.xx:/home/root/journald_v1.99.1-97-gd836be9_20170207020541_notification.txt .
ssh: connect to host 9.3.185.174 port 22: Connection refused
rango@ubuntu:~/Desktop$ ping xx.xx.xx.xx
PING xx.xx.xx.xx (xx.xx.xx.xx) 56(84) bytes of data.
64 bytes from 9.3.185.174: icmp_seq=1 ttl=128 time=940 ms
64 bytes from 9.3.185.174: icmp_seq=2 ttl=128 time=282 ms
64 bytes from 9.3.185.174: icmp_seq=3 ttl=128 time=284 ms
```

gkeishin, 2017-02-07:
PDU cycled it and collected the log
- [x] Journal log  
[journald_v1.99.1-97-gd836be9_20170207020541_notification.txt](https://github.com/openbmc/openbmc/files/757492/journald_v1.99.1-97-gd836be9_20170207020541_notification.txt)

shenki, 2017-02-07:
This suggests systemd (or DBUS?) has hung https://github.com/systemd/systemd/issues/3282

`systemctl poweroff -ff` should force a reboot in this case.

This bug should track why systeemctl is unable to communicate with systemd.

gkeishin, 2017-02-07:
Thanks @shenki for the update..

sivassrr, 2017-02-07:
@shenki what signature matches between systemd/systemd#3282 and this issue.

sivassrr, 2017-02-07:
This can help us to debug and avoid opening similar issues in future.

closed 2017-03-31 by gkeishin