openbmc/openbmc #1102: Soft link needed for BMC flash image

trajeswaran, 2017-02-09:
BMC flash image file name is of the form witherspoon-YYYYMMDDHHMMSS.all.tar. As the file name includes the build date & time, the code update tool has to look for a pattern match. If we can create a soft link called "witherspoon.all.tar to witherspoon-YYYYMMDDHHMMSS.all.tar, then it will simplify the things.

williamspatrick, 2017-06-02:
This is a very low priority work item.

sbtkd85, 2017-07-11:
I'm going to give this a go and see if I can't figure out how to do it. I'm new to both OpenBMC and relatively new to Poky/Yocto, but I think I should be able to do this. I'm also on IRC if anyone would like to discuss this.

sbtkd85, 2017-07-31:
I haven't forgotten about this, I have a fix, but I can't test due to some internal IT issues at my organization. As soon as I get those resolved and I validate my fix, I'll submit it for review.

closed 2019-10-27 by stale[bot]