openbmc/openbmc #1103: Zaius pinctrl error: request() failed for pin 47

GHF, 2017-02-09:
Looks like org.openbmc.buttons.Power can't export GPIOF7 (`CHECKSTOP` or `PV_CP_MD_JTAG_ATTENTION_N` on schematic).

```
Feb 09 20:51:41 zaius kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request() failed for pin 47
Feb 09 20:51:41 zaius kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-47 (1e780000.gpio:327) status -19
Feb 09 20:51:40 zaius systemd[1]: xyz.openbmc_project.State.BMC.service: Failed with result 'core-dump'.
Feb 09 20:51:41 zaius power_control.exe[1046]: GPIO Lookup:  BMC_UCD_LATCH_LE = 292,out
Feb 09 20:51:41 zaius kernel[923]: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request() failed for pin 47
Feb 09 20:51:41 zaius host_checkstop.exe[1047]: GPIO Lookup:  CHECKSTOP = 327,falling
```

Not sure yet what is else occupying that pin.

shenki, 2017-02-10:
This is GPIOF7.
```
root@zaius:~# cat /sys/kernel/debug/pinctrl/1e6e2000.syscon\:pinctrl\@1e6e2000/pinmux-pins | grep 'pin 47'
pin 47 (H18): GPIO 1e780000.gpio:367
```
It's claimed as a GPIO. 

[pinmux-pins.txt](https://github.com/openbmc/openbmc/files/765392/pinmux-pins.txt)

amboar, 2017-02-10:
It may help to build/boot a kernel with `CONFIG_DEBUG_GPIO=y` and `CONFIG_DEBUG_PINCTRL=y` and grab the output. it probably won't give us what is grabbing the pin, but it should tell us when.

nkskjames, 2017-02-10:
F7 is driven low from CPU when a checkstop occurs.   the host_checkstop process is what is exporting that as a gpio to monitor it.  it is set to falling interrupt mode.  you can see that in the trace:

`09 20:51:41 zaius host_checkstop.exe[1047]: GPIO Lookup:  CHECKSTOP = 327,falling`

GHF, 2017-02-10:
My apologies, I did mean that it's **`org.openbmc.control.Checkstop`** that can't export GPIOF7.

I disabled that service and rebooted (`rm /etc/systemd/system/obmc-standby.target.wants/org.openbmc.control.Checkstop\@0.service`) to find that nothing else grabs that pin:

```
root@zaius:~# grep 'pin 47' /sys/kernel/debug/pinctrl/1e6e2000.syscon\:pinctrl\@1e6e2000/pinmux-pins
pin 47 (H18): UNCLAIMED
```

Then if I manually start that service (`systemctl start org.openbmc.control.Checkstop@0.service`), I get the pinctrl failure to export:

```
Feb 10 21:16:09 zaius systemd[1]: Created slice system-org.openbmc.control.Checkstop.slice.
Feb 10 21:16:09 zaius systemd[1]: Starting Phosphor Host0 Checkstop Monitor...
Feb 10 21:16:10 zaius host_checkstop.exe[944]: GPIO Lookup:  CHECKSTOP = 327,falling
Feb 10 21:16:09 zaius systemd[1]: Started Phosphor Host0 Checkstop Monitor.
Feb 10 21:16:10 zaius kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 47 (H18) for 1e780000.gpio:327
Feb 10 21:16:10 zaius kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request() failed for pin 47
Feb 10 21:16:10 zaius kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-47 (1e780000.gpio:327) status -19
Feb 10 21:16:10 zaius kernel: gpio-327 (?): gpiod_request: status -19
Feb 10 21:16:10 zaius kernel: export_store: status -19
Feb 10 21:16:10 zaius kernel[787]: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 47 (H18) for 1e780000.gpio:327
Feb 10 21:16:10 zaius kernel[787]: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request() failed for pin 47
Feb 10 21:16:10 zaius kernel[787]: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-47 (1e780000.gpio:327) status -19
Feb 10 21:16:10 zaius kernel[787]: gpio-327 (?): gpiod_request: status -19
Feb 10 21:16:10 zaius kernel[787]: export_store: status -19
```

And indeed that pin is successfully exported as GPIO and available in sysfs:

```
root@zaius:~# ls -A /sys/class/gpio/gpio347 
active_low  device      direction   edge        subsystem   uevent      value
```

It's my strong suspicion that the host_checkstop daemon is exporting 347 twice and failing the second second time.

nkskjames, 2017-02-13:
The gpio gets exported when you see the message:
`GPIO Lookup:  CHECKSTOP = 327,falling`
how many times do you see that?

amboar, 2017-02-13:
@nkskjames as it turns out, only once

@GHF:
```
Feb 10 21:16:10 zaius kernel: gpio-327 (?): gpiod_request: status -19
Feb 10 21:16:10 zaius kernel: export_store: status -19
```
```
root@zaius:~# ls -A /sys/class/gpio/gpio347 
```

Looks like you have a typo: 327 vs 347.

amboar, 2017-08-10:
@GHF has this been resolved?

GHF, 2017-08-10:
Yep.

closed 2017-08-10 by GHF