openbmc/openbmc #1104: SDK Fails to build - Transaction() undefined reference when building inside phosphor-state-manager

JoshDKing, 2017-02-09:
I recently built a new sdk:

BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "palmetto"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"

Then used the sdk within phosphor-state-manager and when compiling I would discover these undefined references sdbusplus::server::transaction::details::id.

`bash-4.1$ make
make  all-am
make[1]: Entering directory /gsa/ausgsa-h3/18/jdking/git/phosphor-state-manager
  CXX      host_state_manager.o
  CXX      xyz.openbmc_project.State.Host.o
  CXX      host_state_manager_main.o
  CXXLD    phosphor-host-state-manager
host_state_manager.o: In function sdbusplus::server::transaction::details::Transaction::Transaction():
/gsa/ausgsa-h3/18/jdking/workspace/arm_feb09_2/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/server/transaction.hpp:20: undefined reference to sdbusplus::server::transaction::details::id
/gsa/ausgsa-h3/18/jdking/workspace/arm_feb09_2/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/server/transaction.hpp:20: undefined reference to sdbusplus::server::transaction::details::id
/gsa/ausgsa-h3/18/jdking/workspace/arm_feb09_2/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/server/transaction.hpp:20: undefined reference to sdbusplus::server::transaction::details::id
/gsa/ausgsa-h3/18/jdking/workspace/arm_feb09_2/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/server/transaction.hpp:20: undefined reference to sdbusplus::server::transaction::details::id
/gsa/ausgsa-h3/18/jdking/workspace/arm_feb09_2/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/server/transaction.hpp:20: undefined reference to sdbusplus::server::transaction::details::id
host_state_manager.o:/gsa/ausgsa-h3/18/jdking/workspace/arm_feb09_2/sysroots/armv5e-openbmc-linux-gnueabi/usr/include/sdbusplus/server/transaction.hpp:20: more undefined references to sdbusplus::server::transaction::details::id follow
collect2: error: ld returned 1 exit status
make[1]: *** [phosphor-host-state-manager] Error 1
make[1]: Leaving directory /gsa/ausgsa-h3/18/jdking/git/phosphor-state-manager
make: *** [all] Error 2`

I am able to use an old sdk and everything builds fine.  I attempted to clean things out and build the new sdk a few times to make sure and I continued getting the same results.

williamspatrick, 2017-02-10:
This should have been resolved by this commit:

https://gerrit.openbmc-project.xyz/#/c/2145/

Have you rebased recently?

JoshDKing, 2017-02-14:
Yes I did an entire update of my openbmc and thats when I got this problem.  I'll give it another shot though try it again and hopefully it goes away.

JoshDKing, 2017-02-16:
Just to give an update, but yes this seems to be working fine and I'm no longer having issues.

closed 2017-02-16 by williamspatrick