openbmc/openbmc #1105: SDK for x86-64 fails to build - cannot convert 'size_t*' to 'socklen_t*'

mine260309, 2017-02-10:
I met a build failure when trying to build x86-64 SDK.

The error seems to be 64-bit sepcific, which could be an issue in `slpd-lite`:
```
| ../git/sock_channel.cpp:49:49: error: cannot convert 'size_t* {aka long unsigned int*}' to 'socklen_t* {aka unsigned int*}' for argument '6' to 'ssize_t recvfrom(int, void*, size_t, int, sockaddr*, socklen_t*)'
```

Build Configuration:
```
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "x86_64-openbmc-linux"
MACHINE           = "qemux86-64"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
...
meta-phosphor     = "master:4458d4939a33672bf3fc5113523c6d2453478371"
```

Error log:
```
ERROR: slpd-lite-1.0-r1 do_compile: oe_runmake failed
ERROR: slpd-lite-1.0-r1 do_compile: Function failed: do_compile (log file is located at /home/leiyu/openbmc/build/tmp/work/core2-64-openbmc-linux/slpd-lite/1.0-r1/temp/log.do_compile.15975)
ERROR: Logfile of failure stored in: /home/leiyu/openbmc/build/tmp/work/core2-64-openbmc-linux/slpd-lite/1.0-r1/temp/log.do_compile.15975
Log data follows:
| DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4
| make  all-am
| make[1]: Entering directory `/home/leiyu/openbmc/build/tmp/work/core2-64-openbmc-linux/slpd-lite/1.0-r1/build'
| x86_64-openbmc-linux-g++  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse --sysroot=/home/leiyu/openbmc/build/tmp/sysroots/qemux86-64 -std=c++14 -DHAVE_CONFIG_H -I. -I../git     -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/work/core2-64-openbmc-linux/slpd-lite/1.0-r1=/usr/src/debug/slpd-lite/1.0-r1 -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/sysroots/qemux86-64=  -fvisibility-inlines-hidden -c -o main.o ../git/main.cpp
| x86_64-openbmc-linux-g++  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse --sysroot=/home/leiyu/openbmc/build/tmp/sysroots/qemux86-64 -std=c++14 -DHAVE_CONFIG_H -I. -I../git     -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/work/core2-64-openbmc-linux/slpd-lite/1.0-r1=/usr/src/debug/slpd-lite/1.0-r1 -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/sysroots/qemux86-64=  -fvisibility-inlines-hidden -c -o sock_channel.o ../git/sock_channel.cpp
| x86_64-openbmc-linux-g++  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse --sysroot=/home/leiyu/openbmc/build/tmp/sysroots/qemux86-64 -std=c++14 -DHAVE_CONFIG_H -I. -I../git     -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/work/core2-64-openbmc-linux/slpd-lite/1.0-r1=/usr/src/debug/slpd-lite/1.0-r1 -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/sysroots/qemux86-64=  -fvisibility-inlines-hidden -c -o slp_server.o ../git/slp_server.cpp
| x86_64-openbmc-linux-g++  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse --sysroot=/home/leiyu/openbmc/build/tmp/sysroots/qemux86-64 -std=c++14 -DHAVE_CONFIG_H -I. -I../git     -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/work/core2-64-openbmc-linux/slpd-lite/1.0-r1=/usr/src/debug/slpd-lite/1.0-r1 -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/leiyu/openbmc/build/tmp/sysroots/qemux86-64=  -fvisibility-inlines-hidden -c -o slp_parser.o ../git/slp_parser.cpp
| ../git/sock_channel.cpp: In member function 'std::tuple<int, std::vector<unsigned char, std::allocator<unsigned char> > > udpsocket::Channel::read()':
| ../git/sock_channel.cpp:49:49: error: cannot convert 'size_t* {aka long unsigned int*}' to 'socklen_t* {aka unsigned int*}' for argument '6' to 'ssize_t recvfrom(int, void*, size_t, int, sockaddr*, socklen_t*)'
|                                 &address.addrSize);   // Address Length
```

williamspatrick, 2017-02-10:
Fixed by 38c1d044c06c0e46742979159b6155c8afa4eca6

williamspatrick, 2017-02-10:
38c1d04 fixed the same problem in ipmid.  This compile failure is for slp.

closed 2017-02-11 by williamspatrick