openbmc/openbmc #1106: net-ipmid/libapphandler.so.0: undefined symbol: _ZN9sdbusplus6server11transaction7details2idE

geissonator, 2017-02-10:
Flashed an image off of https://github.com/bradbishop/openbmc/commits/deepak for some host IPMI bringup on witherspoon.  The netipmid app is hitting this on startup:

```
netipmid 
Registering handler: /usr/lib/net-ipmid/libapphandler.so.0
Error opening /usr/lib/net-ipmid/libapphandler.so.0/usr/lib/net-ipmid/libapphandler.so.0: undefined symbol: _ZN9sdbusplus6server11transaction7details2idE
```

geissonator, 2017-02-10:
Probably has to do with pulling this batch of code in @anoo1? - https://github.com/bradbishop/openbmc/commit/81bebc0fc4d0551a5849ce2cc9d8deb2c83f7041

williamspatrick, 2017-02-10:
The phosphor-logging repo might be missing an RDEPEND="libsdbusplus".  I think we have a number of those problems with the changes going in.

williamspatrick, 2017-02-10:
@bradbishop - FYI.

dkodihal, 2017-02-10:
I think phoshphor-host-ipmi also needs to link with libsdbusplus.

williamspatrick, 2017-02-10:
That causes a serious issue... phosphor-host-ipmid doesn't itself even use any sdbusplus header files.  How is that repository (both at an autotools level and at a bitbake level) suppose to know that it should link against libsdbusplus?

williamspatrick, 2017-02-10:
Since there is no `libphosphor-logging.so`, there isn't any direct way to get `libipmi.so -> libphosphor-logging.so -> libsdbusplus.so`.  I suspect the "solution" is to have phosphor-logging provide a package-config entry where the LDFLAGS and CFLAGS point, indirectly, at libsdbusplus.so.

williamspatrick, 2017-02-10:
It does seem like pkgconfig 'Requires' statements are there to deal with this.  See "Metadata File Syntax" at https://linux.die.net/man/1/pkg-config.

closed 2017-02-13 by williamspatrick