openbmc/openbmc #1107: ipmid fails to start

dkodihal, 2017-02-10:
It seems the ipmid fails to start on a witherspoon (w37).

root@witherspoon:/usr/sbin# Registering handler:[/usr/lib/host-ipmid/libsysintfcmds.so.0]
Registering NetFn:[0x6], Cmd:[0x35]
Registering NetFn:[0x6], Cmd:[0x2E]
Registering NetFn:[0x6], Cmd:[0x31]
Registering handler:[/usr/lib/host-ipmid/libstrgfnhandler.so.1]

[1]+  Segmentation fault      (core dumped) ./ipmid

williamspatrick, 2017-02-10:
Can we get a core file and/or backtrace?

dkodihal, 2017-02-10:
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-12"
VERSION_ID="v1.99.2-12-g3d4dc92-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-12"
BUILD_ID="v1.99.2"

dkodihal, 2017-02-10:
I'll get a backtrace and upload here.

dkodihal, 2017-02-11:
This will be fixed via https://gerrit.openbmc-project.xyz/#/c/2367/

closed 2017-02-24 by geissonator