openbmc/openbmc #1109: [Witherspoon w38 v1.99.2-14-g38c1d04 - 20170210024342 ] IPMI Inband Failure

gkeishin, 2017-02-10:
**Inband IPMI failed to start on boot**

```
root@witherspoon:~# systemctl status org.openbmc.HostServices.service | cat
● org.openbmc.HostServices.service - Phosphor Inband IPMI
   Loaded: loaded (/lib/systemd/system/org.openbmc.HostServices.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: core-dump) since Tue 2017-01-31 18:55:58 UTC; 4min 48s ago
  Process: 826 ExecStart=/usr/sbin/ipmid (code=dumped, signal=SEGV)
 Main PID: 826 (code=dumped, signal=SEGV)

Jan 31 18:55:57 witherspoon systemd[1]: Starting Phosphor Inband IPMI...
Jan 31 18:55:57 witherspoon systemd[1]: org.openbmc.HostServices.service: Main process exited, code=dumped, status=11/SEGV
Jan 31 18:55:57 witherspoon systemd[1]: Failed to start Phosphor Inband IPMI.
Jan 31 18:55:57 witherspoon systemd[1]: org.openbmc.HostServices.service: Unit entered failed state.
Jan 31 18:55:57 witherspoon systemd[1]: org.openbmc.HostServices.service: Failed with result 'core-dump'.
Jan 31 18:55:58 witherspoon systemd[1]: org.openbmc.HostServices.service: Service hold-off time over, scheduling restart.
Jan 31 18:55:58 witherspoon systemd[1]: Stopped Phosphor Inband IPMI.
Jan 31 18:55:58 witherspoon systemd[1]: org.openbmc.HostServices.service: Start request repeated too quickly.
Jan 31 18:55:58 witherspoon systemd[1]: Failed to start Phosphor Inband IPMI.
root@witherspoon:~#
root@witherspoon:~# uptime
 19:03:00 up 7 min,  load average: 0.01, 0.53, 0.42
root@witherspoon:~#
```

- [x] Journal Log  
[journald_v1.99.2-14-g38c1d04_20170210024342_IPMI_Inband.txt](https://github.com/openbmc/openbmc/files/766043/journald_v1.99.2-14-g38c1d04_20170210024342_IPMI_Inband.txt)

gkeishin, 2017-02-10:
We are seeing in newer build this one... I think this is same with https://github.com/openbmc/openbmc/issues/1107

geissonator, 2017-02-10:
Yep, duplicate

closed 2017-02-10 by geissonator