openbmc/openbmc #1110: [Barreleye ] Network IPMI is not working with latest build

rahulmah, 2017-02-10:
Network IPMI is not working with latest build. It fails with "Unspecified error" error.

````
[rahulmah@gfwa124:~/ipmitool_binary]$ ipmitool -I lanplus -C 1 -P PASSW0RD -H xx.xx.xx.xx sel time get
Get SEL Time command failed: Unspecified error
[rahulmah@gfwa124:~/ipmitool_binary]$
```

Journal Logs: 
[Journal logs for Network IPMI Failure.txt](https://github.com/openbmc/openbmc/files/766788/Journal.logs.for.Network.IPMI.Failure.txt)

BMC Fw: 
```
root@barreleye:/tmp# cat /etc/os-release /etc/version
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-13"
VERSION_ID="v1.99.2-13-gf598488-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-13"
BUILD_ID="v1.99.2"
20170210013109
```

geissonator, 2017-02-10:
Duplicate of https://github.com/openbmc/openbmc/issues/1107 ?

tomjoseph83, 2017-02-10:
The mapper_get_service is failing to get the busname with a connection time out error. This happens when the BMC comes up and the daemon is started the first time by the service. When the net-ipmid service is stopped and started again, this issue is not seen. The same issue is not seen with host-ipmid with which the common provider library is shared. @bradbishop suggestions?
```
Feb 02 20:46:37 barreleye netipmid[689]: Failed to get /org/openbmc/NetworkManager/Interface bus name: Connection timed out
```

tomjoseph83, 2017-02-10:
@geissonator This is not a duplicate for #1107

bradbishop, 2017-02-10:
Adding a:

After=obmc-mapper.target

to the net-ipmi service unit might help.

tomjoseph83, 2017-02-13:
Adding After=obmc-mapper.target , did not fix the problem. Even host-ipmid doesn’t have the "After=obmc-mapper.target' in the unit file.

tomjoseph83, 2017-02-15:
Instead of connecting to the system bus (sd_bus_open_system()) at startup of the daemon, it was delayed till the first command got executed. This fixed the problem. 

So this looks like a case we are getting a bad reference for system bus.

So what would be the preferable dependency to be added in the service file? 

@bradbishop

sivassrr, 2017-02-27:
As discussed in the design meeting, Our focus will be more on external ipmitool way of testing. 
For that we need this. Thank you in advance for your help

bradbishop, 2017-02-27:
I noticed that host-ipmid was missing the mapper dependency the other day.  Added here:

https://gerrit.openbmc-project.xyz/#/c/2688/

rahulmah, 2017-03-02:
This can be closed. I see network IPMI is now working after Brad's fix.

```[rahulmah@gfwa124:~]$ ipmitool -I lanplus -C 3 -P PASSW0RD -H xx.xx.xx.xx sel info
SEL Information
Version          : 1.5 (v1.5, v2 compliant)
Entries          : 0
Free Space       : 65535 bytes or more
Percent Used     : unknown
Last Add Time    : Not Available
Last Del Time    : Not Available
Overflow         : false
Supported Cmds   : 'Partial Add' 'Reserve'
[rahulmah@gfwa124:~]$```

rahulmah, 2017-03-07:
This issue is not fixed. Found that "sel info" command is working but other commands like "sel time get" are failing still with "Unspecified error". So reopening it.

```
bash-4.1$ ipmitool -I lanplus -C 1 -P 0penBmc -H xx.xx.xx.xx sel time get                                                                                   Get SEL Time command failed: Unspecified error
bash-4.1$
```
```
bash-4.1$ipmitool -I lanplus -C 1 -P 0penBmc -H xx.xx.xx.xx sel info                                                                                       SEL Information
Version          : 1.5 (v1.5, v2 compliant)
Entries          : 0
Free Space       : 65535 bytes or more
Percent Used     : unknown
Last Add Time    : Not Available
Last Del Time    : Not Available
Overflow         : false
Supported Cmds   : 'Partial Add' 'Reserve'
```

tomjoseph83, 2017-03-09:
@bradbishop I added the mapper dependency specified in the https://gerrit.openbmc-project.xyz/#/c/2688/ ,this does'nt seem to fix the problem. 

net-ipmid depends on the UUID property in  /org/openbmc/control/chassis0. What would be the preferable dependency to be in place?

geissonator, 2017-03-16:
Potential fix up here - https://gerrit.openbmc-project.xyz/#/c/3038/

williamspatrick, 2017-03-16:
@rahulmah : I would prefer if you do not reopen issues once we have a code fix of some sort in, especially for a different symptom.  Your original issue was for net-ipmid not working at all and now you are describing a specific set of commands that do not work.  Those are two different issues.

rahulmah, 2017-03-17:
@williamspatrick : Sure we will take care of this. I actually reopen this issue as the command which I mentioned at first is still failing. But opening a new issue make more sense.

sivassrr, 2017-03-20:
Discussed in today's squad scribe with Tom,  Right now, he will fix the code using this issue and close it.

closed 2017-03-20 by williamspatrick