openbmc/openbmc #1111: ipmid terminates while trying to process esel

dkodihal, 2017-02-10:
On a patched Witherspoon (w37. multiple patches - all related to inventory), I observed this post successful inventory creation : the ipmid terminated with these messages -

Failed to call the method: send_esel_to_dbus Connection reset by peer
Failed to send esel to dbus
Failed to call the method: Transport endpoint is not connected
Dest: :1.1, Path: /org/openbmc/HostIpmi/1

Maybe a communication to a non-existent object? Could this be fixed by your current work related to host errors @anoo1 ?

dkodihal, 2017-02-11:
Related information in https://gerrit.openbmc-project.xyz/#/c/2289/1/storageaddsel.cpp

anoo1, 2017-02-11:
@tomjoseph83 Can you check this out? Seems the same as https://github.com/openbmc/openbmc/issues/996

dkodihal, 2017-02-15:
Seeing this on w37.

geissonator, 2017-02-15:
@anoo1 what was your quick fix for this one?

anoo1, 2017-02-15:
Comment out the failing code. https://gerrit.openbmc-project.xyz/#/c/2464/

williamspatrick, 2017-02-16:
Fixed by openbmc/phosphor-host-ipmid@513d68e

closed 2017-02-16 by williamspatrick