openbmc/openbmc #1113: Zaius: OpenFSI master scan not working

cbostic, 2017-02-10:
Running echo 1 > scan  on the fsi-master on Zaius does not kick off a proper FSI device scan.   Based on the lack of output from that command this would be typical of a potential GPIO pin assignment problem.

Will need to identify the hardware source documentation and compare to the newly integrated Zaius dts gpio tree bindings.

geissonator, 2017-02-10:
@amboar @spinler ?

nkskjames, 2017-02-10:
I don't see any fsi things in zaius dts.    here is the assignment:
`		clock-gpios = <&gpio ASPEED_GPIO(C, 3) GPIO_ACTIVE_HIGH>;
		data-gpios = <&gpio ASPEED_GPIO(C, 2) GPIO_ACTIVE_HIGH>;
		mux-gpios = <&gpio ASPEED_GPIO(A, 6) GPIO_ACTIVE_HIGH>;
		enable-gpios = <&gpio ASPEED_GPIO(D, 0) GPIO_ACTIVE_HIGH>;
		trans-gpios = <&gpio ASPEED_GPIO(O, 6) GPIO_ACTIVE_HIGH>;`

spinler, 2017-02-10:
I saw on the mailing list that @shenki was working on this.

shenki, 2017-02-11:
There is a patch on the mailing list. I have an ack from @GHF so that will go in on Monday.

closed 2017-06-01 by cbostic