openbmc/openbmc #1114: [Kernel Panic]  BMC to recover post panic

gkeishin, 2017-02-10:
Any simple hack to trigger the BMC kernel to panic and which we could use for testing.. we wanted to check if it reboot automatically,

As per discussion with @shenki  for now they do not reboot when the kernel crashes. 

There is similar changes going on for Ziaus https://gerrit.openbmc-project.xyz/#/c/2332/

Would this be available for witherspoon too ?

williamspatrick, 2017-02-10:
@shenki - I think we should add this to all machines.  Is this a defconfig update or is this something we handle on the bitbake side of things?

@GHF - FYI.

shenki, 2017-02-11:
Yeah lets change it to be a defconfig change for all hardware. The 1 second timeout is fine; if you've got a console attached it will always be in the scroll back, and if you don't there's no benefit to waiting longer before rebooting.

@GHF would you mind changing your zaius commit to instead enable this for both the ast2500 and ast2400 configs?

@gkeishin you can kill PID 1 to force a panic:

```
root@romulus:~# kill -ABRT 1
root@romulus:~# Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000008b

CPU: 0 PID: 1 Comm: systemd Not tainted 4.7.10-e62c7f11175525c028d9a6abaf6a2a5c275664d1 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<8016cec4>] (panic+0xb8/0x24c)
[<8016cec4>] (panic) from [<80112a10>] (do_exit+0x3b0/0x798)
[<80112a10>] (do_exit) from [<80112e6c>] (do_group_exit+0x48/0xa4)
[<80112e6c>] (do_group_exit) from [<8011b0c4>] (get_signal+0x3d4/0x428)
[<8011b0c4>] (get_signal) from [<80104cf8>] (do_signal+0x98/0x3f8)
[<80104cf8>] (do_signal) from [<801051b8>] (do_work_pending+0x54/0xac)
[<801051b8>] (do_work_pending) from [<80102214>] (slow_work_pending+0xc/0x20)
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000008b
```

gkeishin, 2017-02-13:
@shenki  Sounds good to me.. Let me know once it's in so that I can verify. .Thanks

shenki, 2017-02-13:
I updated @GHF's commit to be for all systems

closed 2017-02-13 by williamspatrick