openbmc/openbmc #1115: [Witherspoon w30 - v1.99.2-24-g17064cb] hctosys: unable to read the hardware clock

gkeishin, 2017-02-13:
```
root@witherspoon:~# cat /etc/version
20170211042525
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-24"
VERSION_ID="v1.99.2-24-g17064cb-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-24"
BUILD_ID="v1.99.2"
root@witherspoon:~# uptime
 19:04:43 up 9 min,  load average: 0.00, 0.35, 0.35
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~#
```
- [x] Journald  
[journald_v1.99.2-24-g17064cb_20170211042525_HWCLOCK.txt](https://github.com/openbmc/openbmc/files/770104/journald_v1.99.2-24-g17064cb_20170211042525_HWCLOCK.txt)

```
Jan 31 18:55:31 witherspoon kernel: rtc-rv8803 11-0032: Voltage low, temperature compensation stopped.
Jan 31 18:55:31 witherspoon kernel: rtc-rv8803 11-0032: Voltage low, data loss detected.
Jan 31 18:55:31 witherspoon kernel: rtc-rv8803 11-0032: Voltage low, data is invalid.
Jan 31 18:55:31 witherspoon kernel: rtc-rv8803 11-0032: rtc core: registered rx8900 as rtc0
Jan 31 18:55:31 witherspoon kernel: rtc-rv8803 11-0032: Voltage low, data is invalid.
Jan 31 18:55:31 witherspoon kernel: rtc-rv8803 11-0032: hctosys: unable to read the hardware clock
Jan 31 18:55:41 witherspoon kernel[766]: rtc-rv8803 11-0032: Voltage low, temperature compensation stopped.
Jan 31 18:55:41 witherspoon kernel[766]: rtc-rv8803 11-0032: Voltage low, data loss detected.
Jan 31 18:55:41 witherspoon kernel[766]: rtc-rv8803 11-0032: Voltage low, data is invalid.
Jan 31 18:55:41 witherspoon kernel[766]: rtc-rv8803 11-0032: rtc core: registered rx8900 as rtc0
Jan 31 18:55:41 witherspoon kernel[766]: rtc-rv8803 11-0032: Voltage low, data is invalid.
Jan 31 18:55:41 witherspoon kernel[766]: rtc-rv8803 11-0032: hctosys: unable to read the hardware clock
```

vishwabmc, 2017-02-13:
@gkeishin Please help put command and the result that you see in terminal along with dmesg ?

http://forum.buffalo.nas-central.org/viewtopic.php?t=11496

http://glomation.net/smf/index.php?topic=32.0

That said, I believe @mine260309 had similar issues and that was resoled by setting the time initially.

gkeishin, 2017-02-13:
@vishwabmc  The journal log is attached in the above note which has this trace in it.. Yeah i can check on the topic listed there..

vishwabmc, 2017-02-13:
journal may not have all as its a systemd specific thing.. Please help provide the data

gkeishin, 2017-02-13:
Yup.. i tried the resetting the clock stuff and it worked ..
```
root@witherspoon:~# hwclock --utc --systohc
root@witherspoon:~# dmesg | grep rtc
rtc-rv8803 11-0032: Voltage low, temperature compensation stopped.
rtc-rv8803 11-0032: rtc core: registered rx8900 as rtc0
rtc-rv8803 11-0032: setting system clock to 2016-02-29 09:18:59 UTC (1456737539)
root@witherspoon:~#
root@witherspoon:~# hwclock --show
Tue Jan 31 19:12:39 2017  0.000000 seconds
root@witherspoon:~#
**rebooted and checked again.. looks good now**
root@witherspoon:~# hwclock --show
Tue Jan 31 19:18:10 2017  0.000000 seconds
root@witherspoon:~#
Jan 31 19:14:38 witherspoon kernel: clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
Jan 31 19:14:38 witherspoon kernel: sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Jan 31 19:14:38 witherspoon kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Jan 31 19:14:38 witherspoon kernel: clocksource: Switched to clocksource moxart_timer
Jan 31 19:14:39 witherspoon kernel: rtc-rv8803 11-0032: setting system clock to 2017-01-31 19:14:32 UTC (1485890072)
Jan 31 19:14:49 witherspoon kernel[753]: clocksource: moxart_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
Jan 31 19:14:49 witherspoon kernel[753]: sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
Jan 31 19:14:49 witherspoon kernel[753]: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Jan 31 19:14:49 witherspoon kernel[753]: clocksource: Switched to clocksource moxart_timer
Jan 31 19:14:49 witherspoon kernel[753]: rtc-rv8803 11-0032: setting system clock to 2017-01-31 19:14:32 UTC (1485890072)
```

gkeishin, 2017-02-13:
Closing the issue as it is not seeing post initial setting..

closed 2017-02-13 by gkeishin