openbmc/openbmc #1116: [Witherspoon w38 v1.99.2-24-g17064cb ] org.freedesktop.DBus.Error.ServiceUnknown

gkeishin, 2017-02-13:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-24"
VERSION_ID="v1.99.2-24-g17064cb-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-24"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170211042525
root@witherspoon:~#
```

```
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:     self._named_service = conn.activate_name_owner(bus_name)
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:     self.start_service_by_name(bus_name)
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:     'su', (bus_name, flags)))
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:     message, timeout)
Jan 31 18:56:40 witherspoon discover_system_state.py[845]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.control.Power was not provided by any .service files
```

```
root@witherspoon:~# uptime
 19:33:39 up 38 min,  load average: 0.00, 0.00, 0.02
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~# journalctl --no-pager > journald_v1.99.2-24-g17064cb_20170211042525_DBUS_ERROR.txt
root@witherspoon:~#
```
- [x] Journal log 
[journald_v1.99.2-24-g17064cb_20170211042525_DBUS_ERROR.txt](https://github.com/openbmc/openbmc/files/770188/journald_v1.99.2-24-g17064cb_20170211042525_DBUS_ERROR.txt)

vishwabmc, 2017-02-13:
@gkeishin It says that there is no such service file called 'org.openbmc.control.Power". I wonder if this is one of those `move to xyz/opebmc_project/` instances ?

* This is posted by the application that tries to see 'what should the BMC be doing post BMC-Ready'*

Jan 31 18:56:40 witherspoon discover_system_state.py[845]:     self._named_service = conn.activate_name_owner(bus_name)
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:     self.start_service_by_name(bus_name)
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:     'su', (bus_name, flags)))
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Jan 31 18:56:40 witherspoon discover_system_state.py[845]:     message, timeout)
Jan 31 18:56:40 witherspoon discover_system_state.py[845]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.control.Power was not provided by any .service files

However, I can see that timemanager is able to receive the PGOOD changed signal from the SAME daemon.

@gkeishin: can you try executing `discover_system_state.py` by hand and see if that would be successful ?.

gkeishin, 2017-02-13:
@vishwabmc Sure.. will do that..

gkeishin, 2017-02-16:
Yeah seems to be intermittent.. Will re-open if i see it again..

closed 2017-02-16 by gkeishin