openbmc/openbmc #1117: [Zaius] Incorrect MAC calculated if sequential addresses starting from the 0xFF

johnhcwang, 2017-02-13:
Hi @GHF , 

Base on Zaius FRU EEPROM specification, MAC address for LOM stored in the FRU and BMC calculates the sequential addresses starting from this address for NCSI and PHY.
For example:
LOM MAC is 00:1f:e2:fd:eb:ff that stored in FRU
NCSI MAC should be 00:1f:e2:fd:**ec:00**
PHY MAC should be 00:1f:e2:fd:**ec:01**

In BMc v1.99.0-376-gccf524f6-dirty, we got:
NCSI MAC - 00:1f:e2:fd:**eb:ff** => should be 00:1f:e2:fd:**ec:00**
PHY MAC - 00:1f:e2:fd:**eb:00** => should be 00:1f:e2:fd:**ec:01**
```
=> print
EXPORT  table = befdf034, htab.size = 83, htab.filled = 14, size = 0
Unsorted: n=14
          0: beb82398 ==> stdin      => serial
          1: beb823c0 ==> fdtcontroladdr => beb7c4d0
          2: beb8244c ==> verify     => yes
          3: beb82460 ==> eth2addr   => 00:1f:e2:fd:eb:01
          4: beb8249c ==> ethact     => FTGMAC100#0
          5: beb824b0 ==> baudrate   => 115200
          6: beb825a0 ==> spi_dma    => yes
          7: beb825dc ==> stderr     => serial
          8: beb826e0 ==> stdout     => serial
          9: beb82708 ==> bootargs   => console=ttyS4,115200n8 root=/dev/ram rw
         10: beb82744 ==> bootcmd    => bootm 20080000 20300000
         11: beb82898 ==> ethaddr    => 00:1f:e2:fd:eb:ff
         12: beb8299c ==> bootdelay  => 2
         13: beb829c4 ==> eth1addr   => 00:1f:e2:fd:eb:00
baudrate=115200
bootargs=console=ttyS4,115200n8 root=/dev/ram rw
bootcmd=bootm 20080000 20300000
bootdelay=2
eth1addr=00:1f:e2:fd:eb:00
eth2addr=00:1f:e2:fd:eb:01
ethact=FTGMAC100#0
ethaddr=00:1f:e2:fd:eb:ff
fdtcontroladdr=beb7c4d0
spi_dma=yes
stderr=serial
stdin=serial
stdout=serial
verify=yes
```
```
root@zaius:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 00:1F:E2:FD:EB:FF
          inet addr:192.168.2.122  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::21f:e2ff:fefd:ebff/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:49 errors:0 dropped:0 overruns:0 frame:0
          TX packets:121 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3967 (3.8 KiB)  TX bytes:14960 (14.6 KiB)

eth1      Link encap:Ethernet  HWaddr 00:1F:E2:FD:EB:00
          inet addr:192.168.2.124  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::21f:e2ff:fefd:eb00/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12 errors:0 dropped:0 overruns:0 frame:0
          TX packets:30 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2306 (2.2 KiB)  TX bytes:8566 (8.3 KiB)
```

GHF, 2017-02-13:
If I understand correctly:

1) we're computing incorrect offsets for the MAC address relative to what's stored on the FRU EEPROM
2) the offset computation does not carry correctly to the higher order bytes of the MAC addresses

maksymko, 2017-02-14:
@johnhcwang Which U-Boot branch are you using?

johnhcwang, 2017-02-15:
@maksymko , the image (v1.99.2-26-g613376e8-dirty) was provided by @GHF , I don't know which branch he used and these changes didn't seem to merge to openbmc/u-boot.

The u-boot version is:

```
U-Boot 2016.07 (Feb 09 2017 - 12:46:28 -0800)
```

williamspatrick, 2017-02-15:
@GHF : We should probably consider some kind of versioning scheme to make it obvious that images are not actually based on the real upstream openbmc tree.  This git revision isn't publicly available so it isn't something the open source community in general can debug because we don't really know the contents.

GHF, 2017-02-15:
@williamspatrick: What do you suggest? In this case, the git revision in `/etc/os-release` is for a minor change local to my machine on top of openbmc/master (to override `UBOOT_MACHINE`), which is my usual workflow. On the other hand, the U-Boot branch I built from is maintained by @maksymko and is substantially different from the OpenBMC tree. Both are equally opaque as to their contents.

gtmills, 2018-05-23:
@johnhcwang @GHF Still an issue? Can this be closed if not?

closed 2019-04-27 by stale[bot]