openbmc/openbmc #1118: [Witherspoon w38 v1.99.2-24-g17064cb ]  OS doesn't boot on Poweron

gkeishin, 2017-02-13:
Using this PNOR, I was able to boot to OS on other system w30 but not on this w38.

**PNOR Version - Using the same advised by Lab** 
```
open-power-witherspoon-v1.14-dirty
        buildroot-6b9fd80
        skiboot-5.4.3-opdirty
        hostboot-edea301-opdirty
        linux-4.10-rc4-rc4-0c63710
        petitboot-v1.4.0-2284300
        witherspoon-xml-site_local-pragupta-671f883
        occ-4141b5f
        hostboot-binaries-674c635
        capp-ucode-1bb7503
```
On BMC SOL I don't see the ISTEP's being triggered. 

**From Journald** 
```
Jan 31 20:33:34 witherspoon systemd[1]: Starting Start Watchdog0...
Jan 31 20:33:35 witherspoon host_watchdog.exe[759]: Setting watchdog poll interval to: 30000
Jan 31 20:33:35 witherspoon host_watchdog.exe[759]: Starting watchdog with poll interval: 30000
Jan 31 20:33:35 witherspoon systemd[1]: Started Start Watchdog0.
Jan 31 20:33:35 witherspoon systemd[1]: Reached target Chassis0 (Start).
Jan 31 20:33:35 witherspoon phosphor-host-state-manager[881]: Recieved signal that host is running
Jan 31 20:33:35 witherspoon phosphor-host-state-manager[881]: Change to Host State
Jan 31 20:34:05 witherspoon host_watchdog.exe[759]: Polling watchdog: 0

```
Driver details 
```
root@witherspoon:~# uptime;obmcutil state
 20:38:04 up  1:42,  load average: 0.07, 0.23, 0.21
 = HOST_POWERED_ON
root@witherspoon:~# journalctl --no-pager > journald_v1.99.2-24-g17064cb_20170211042525_OS.txt
root@witherspoon:~# cat /etc/version
20170211042525
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-24"
VERSION_ID="v1.99.2-24-g17064cb-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-24"
BUILD_ID="v1.99.2"
```

- [x] Journal log 
[journald_v1.99.2-24-g17064cb_20170211042525_OS.txt](https://github.com/openbmc/openbmc/files/770545/journald_v1.99.2-24-g17064cb_20170211042525_OS.txt)

vishwabmc, 2017-02-15:
@geissonator I talked to @gkeishin about this couple days ago and he said to me that 'state-manager showing the state as Host-Running was through some hack and it does not mean that host is really on.`

Wonder what hack is that and why would state manager tell something is running when its not ?

That said I watched obmc-console of a system that boots vs this and saw that ipmi messages were not flowing on this box.

With the new state manager, how is the initial 'bit banging' done ?. I suppose we did not even kick start the SBE / load HB ?

geissonator, 2017-02-15:
Host-Running happens when the target file which starts all required services (including the one's to do the FSI bit banging) complete.  That's not full proof though because if the services fail, we still say the host is running.  My story this sprint, and this chain of commits, https://gerrit.openbmc-project.xyz/#/c/2283/, will fix that part.

gkeishin, 2017-03-29:
Closing this as it is fixed...

closed 2017-03-29 by gkeishin