openbmc/openbmc #1121: [Zaius v1.99.2-26-g613376e8-dirty] Fail to PING the host from BMC

johnhcwang, 2017-02-14:
Configurations:
1. Host IP: 192.168.2.199
2. BMC eth0: 192.168.2.101
3. BMC eth1: 192.168.2.124

Steps:
1. Success to ping eth0 and eth1 from host
2. Fail to ping host from eth0: ping 192.168.2.199 -I eth0, but there is echo_reply from host actually
3. Success to ping host from eth1: ping 192.168.2.199 -I eth1
4. Success to ping host from eth0 once I remove the cable from eth1 port

Failure information:
```
root@zaius:~# ping 192.168.2.199 -I eth0
PING 192.168.2.199 (192.168.2.199): 56 data bytes
^C
--- 192.168.2.199 ping statistics ---
11 packets transmitted, 0 packets received, 100% packet loss
root@zaius:~# ping 192.168.2.199 -I eth1
PING 192.168.2.199 (192.168.2.199): 56 data bytes
64 bytes from 192.168.2.199: seq=0 ttl=128 time=1.284 ms
64 bytes from 192.168.2.199: seq=1 ttl=128 time=0.936 ms
64 bytes from 192.168.2.199: seq=2 ttl=128 time=0.841 ms
64 bytes from 192.168.2.199: seq=3 ttl=128 time=0.963 ms
^C
--- 192.168.2.199 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 0.841/1.006/1.284 ms
```
```
No.     Time            Source          Destination     Protocol   Length   Info
30	10.195191	192.168.2.101	192.168.2.199	ICMP	   98	    Echo (ping) request  id=0xbb03, seq=9/2304, ttl=64 (reply in 31)
31	10.195257	192.168.2.199	192.168.2.101	ICMP	   98	    Echo (ping) reply    id=0xbb03, seq=9/2304, ttl=128 (request in 30)
34	11.195854	192.168.2.101	192.168.2.199	ICMP	   98	    Echo (ping) request  id=0xbb03, seq=10/2560, ttl=64 (reply in 35)
35	11.196005	192.168.2.199	192.168.2.101	ICMP	   98	    Echo (ping) reply    id=0xbb03, seq=10/2560, ttl=128 (request in 34)
53	20.468266	192.168.2.124	192.168.2.199	ICMP	   98	    Echo (ping) request  id=0xbc03, seq=0/0, ttl=64 (reply in 56)
56	20.468984	192.168.2.199	192.168.2.124	ICMP	   98	    Echo (ping) reply    id=0xbc03, seq=0/0, ttl=128 (request in 53)
57	21.469502	192.168.2.124	192.168.2.199	ICMP	   98	    Echo (ping) request  id=0xbc03, seq=1/256, ttl=64 (reply in 58)
58	21.469653	192.168.2.199	192.168.2.124	ICMP	   98	    Echo (ping) reply    id=0xbc03, seq=1/256, ttl=128 (request in 57)
```

vishwabmc, 2017-02-14:
@spinler Have you seen anything like this ?.

@jk-ozlabs I believe petitboot / host asks BMC for the saved in IP address and consumes that somehow ?. Is there something that you could help add here ?

johnhcwang, 2017-02-14:
My test is running between Zaius BMC and remote desktop/laptop, not the host on Zaius. I corrected the title and the description. Sorry about that if any confusion.

vishwabmc, 2017-02-14:
@johnhcwang Thanks. Its a much needed clarity. Can you get me access to the system ?

I need to look at the .network file

johnhcwang, 2017-02-14:
@vishwabmc , it's difficult because the machine doesn't connect to WAN. I can describe or upload the file you need for debugging.

vishwabmc, 2017-02-14:
I need the .network files in 

/etc/systemd/network/*.network

and also the output of 'ifconfig'

johnhcwang, 2017-02-15:
@vishwabmc , there is no any file under /etc/systemd/network/, but I found out some network files from other folders.
```
root@zaius:~# find / -type f -name "*.network"
/lib/systemd/network/80-container-host0.network
/lib/systemd/network/80-container-ve.network
/run/initramfs/ro/lib/systemd/network/80-container-host0.network
/run/initramfs/ro/lib/systemd/network/80-container-ve.network
/run/initramfs/ro/usr/lib/systemd/network/default.network
/usr/lib/systemd/network/default.network

======

root@zaius:~# cat /lib/systemd/network/80-container-*
#  This file is part of systemd.
#
#  systemd is free software; you can redistribute it and/or modify it
#  under the terms of the GNU Lesser General Public License as published by
#  the Free Software Foundation; either version 2.1 of the License, or
#  (at your option) any later version.

[Match]
Virtualization=container
Name=host0

[Network]
DHCP=yes
LinkLocalAddressing=yes

[DHCP]
UseTimezone=yes
#  This file is part of systemd.
#
#  systemd is free software; you can redistribute it and/or modify it
#  under the terms of the GNU Lesser General Public License as published by
#  the Free Software Foundation; either version 2.1 of the License, or
#  (at your option) any later version.

[Match]
Name=ve-*
Driver=veth

[Network]
# Default to using a /28 prefix, giving up to 13 addresses per container.
Address=0.0.0.0/28
LinkLocalAddressing=yes
DHCPServer=yes
IPMasquerade=yes

======

root@zaius:~# cat /usr/lib/systemd/network/default.network
[Match]
Name=eth*
[Network]
DHCP=ipv4

======

root@zaius:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 00:1F:E2:FD:EB:FF
          inet addr:192.168.2.101  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::21f:e2ff:fefd:ebff/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:675 errors:0 dropped:0 overruns:0 frame:0
          TX packets:735 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:36552 (35.6 KiB)  TX bytes:54738 (53.4 KiB)

eth1      Link encap:Ethernet  HWaddr 00:1F:E2:FD:EB:00
          inet addr:192.168.2.124  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::21f:e2ff:fefd:eb00/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:57 errors:0 dropped:0 overruns:0 frame:0
          TX packets:99 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7936 (7.7 KiB)  TX bytes:15296 (14.9 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:400 errors:0 dropped:0 overruns:0 frame:0
          TX packets:400 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:30400 (29.6 KiB)  TX bytes:30400 (29.6 KiB)
```

williamspatrick, 2017-02-15:
How do we have two network adapters?  Is there both an on-board and NCSI being used?  I don't believe anyone (at least from IBM) has actively tested dual NICs on the BMC.  @GHF : Is this a mode you intend to support and test?

williamspatrick, 2017-02-15:
@johnhcwang - What image are you using?  That might be important for debugging as well.

johnhcwang, 2017-02-15:
@williamspatrick , Zaius has BCM5719 (NCSI) on MAC1 and BCM54612 (PHY) on MAC2.
My original test is running with the image which provided by @GHF because upstream v1.99.2-dirty doesn't include this change (https://github.com/GHF/linux/commit/40d6f3ac5433949951bdcb357a6852fbf7c4cd15). There is a mux in front of BCM5719 on Zaius.

By the way, I try to switch the mux from u-boot and then re-test this issue with upstream v1.99.2-dirty image and got the same situation.

jk-ozlabs, 2017-02-16:
@vishwabmc : petitboot does not consume IP address details from the BMC for any network config - we only use that to display information in the UI.

GHF, 2017-02-16:
Thanks for including me; I've only just seen this. John is indeed testing with my dts change to select NC-SI muxing to the BCM5719 LOM.

I've tested both network adapters (NC-SI to LOM and gigabit PHY) enabled at the same time, but not tried to pass traffic through both at the same time, especially on the same network.

gtmills, 2018-05-23:
@johnhcwang @GHF Still a problem? Can this be closed?

gtmills, 2018-10-12:
@wak-google @pstrinkle Can we close?

wak-google, 2018-10-12:
Yeah, this is probably no longer relevant to us and networking is working fine for all of my machines.