openbmc/openbmc #1123: Huge numbers of packets loss when sent multiple packets per second

prkatti1, 2017-02-14:
We tried sending packets to tcp ports 20, 21, 80 etc. In each case we are seeing lots of packets loss.
We used a tool named nping to send packets.
When we send 3 packets/second, 22-25% packets dropped
When we send 4 packets/second, 50% packets dropped

FW details
------------------------
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-32"
VERSION_ID="v1.99.2-32-g71526c2-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-32"
BUILD_ID="v1.99.2"

Output details
******************************
Max rtt: 293.501ms | Min rtt: 18.442ms | Avg rtt: 145.727ms
Raw packets sent: 100 (4.000KB) | Rcvd: 78 (3.588KB) | Lost: 22 (22.00%)

gkeishin, 2017-02-14:
@prkatti1 Is this a FW or network related issue we are stating here.. any test to isolate those ?... Probably it could be the network.

vishwabmc, 2017-02-15:
@prkatti1 Please help clarify ?.  I am not too sure how to use nping. Also we would need some kind of debug data for these bugs. 

but what does `ping -n < count in integer>` offer you ?

prkatti1, 2017-02-15:
@vishwabmc ping -c <count in integer>, we can send these many packets to target. If -c 1000 then it will send 1000 packets. -i offers interval between packets if -i 0.2 then it will send 5 packets per second.
But ping will only send icmp packets.

amboar, 2017-02-16:
@gwshan and @ozbenh

gwshan, 2017-02-17:
on what type of boxes, we observed this issue? I plan to reproduce on
Palmetto. It's only BMC accessible to me currently.

sivassrr, 2017-02-17:
We observed this in Witherspoon, Barreleye and Palmetto as well. @prkatti could you share system wise test report.

prkatti1, 2017-02-17:
@sivassrr result is consistent on all systems.
Here is a report of packets sent & received, last 2 lines show percentage of packets lost.

Starting Nping 0.6.47 ( http://nmap.org/nping ) at 2017-02-16 17:35 IST
SENT (0.0241s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (0.2244s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (0.4142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=4836 iplen=40  seq=1644239411 win=0
SENT (0.4246s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (0.6251s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (0.8142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=4838 iplen=40  seq=606144155 win=0
SENT (0.8255s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (1.0258s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (1.2142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=4840 iplen=40  seq=1008572801 win=0
SENT (1.2258s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (1.4259s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (1.6142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=4842 iplen=40  seq=1895150067 win=0
SENT (1.6260s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (1.8260s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (2.0142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5074 iplen=40  seq=552612693 win=0
SENT (2.0264s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (2.2268s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (2.4142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5076 iplen=40  seq=1070826873 win=0
SENT (2.4269s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (2.6273s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (2.8141s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5078 iplen=40  seq=1122841689 win=0
SENT (2.8276s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (3.0277s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (3.2142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5080 iplen=40  seq=449658863 win=0
SENT (3.2277s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (3.4278s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (3.6141s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5082 iplen=40  seq=1498947955 win=0
SENT (3.6280s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (3.8280s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (4.0141s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5084 iplen=40  seq=1521222537 win=0
SENT (4.0281s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (4.2285s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (4.4141s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5086 iplen=40  seq=487055479 win=0
SENT (4.4288s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (4.6293s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (4.8142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5123 iplen=40  seq=313410401 win=0
SENT (4.8296s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (5.0296s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (5.2142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5125 iplen=40  seq=1493605167 win=0
SENT (5.2297s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (5.4298s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (5.6141s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5127 iplen=40  seq=1643854969 win=0
SENT (5.6298s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (5.8299s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (6.0141s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5129 iplen=40  seq=1535555873 win=0
SENT (6.0299s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (6.2300s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (6.4141s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=5131 iplen=40  seq=421069005 win=0
SENT (6.4301s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
[... cut ...]
RCVD (19.6142s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=6053 iplen=40  seq=1910669399 win=0
SENT (19.6405s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
SENT (19.8408s) TCP Source_IP_Addr:5095 > BMC_IP_Addr:80 S ttl=64 id=33286 iplen=40  seq=3163697784 win=1480
RCVD (20.0141s) TCP BMC_IP_Addr:80 > Source_IP_Addr:5095 RA ttl=49 id=6055 iplen=40  seq=319957261 win=0

Max rtt: 189.764ms | Min rtt: 173.236ms | Avg rtt: 180.964ms
Raw packets sent: 100 (4.000KB) | Rcvd: 50 (2.300KB) | Lost: 50 (50.00%)

ozbenh, 2017-02-17:
So I have a work-in-progress major rework of the aspeed driver. I've attached my current patch against 4.7, it's not finished but I've already more than tripled the TCP throughput. Let me know if that helps. Note: I only tested that on the AST2500 eval board with a real PHY so far. I haven't tested on 2400 nor with NC-SI.

http://ozlabs.org/~benh/0001-WIP-rework-ftgmac100.patch

gwshan, 2017-02-20:
I tried it on Palmetto, Without the patch from @ozbenh

gwshan@gwshan:~$ sudo nping --rate 5 --data-length 1460 --tcp -p 80 10.61.253.42
Max rtt: 178.421ms | Min rtt: 178.014ms | Avg rtt: 178.150ms
Raw packets sent: 5 (7.500KB) | Rcvd: 5 (230B) | Lost: 0 (0.00%)
Nping done: 1 IP address pinged in 1.05 seconds

gwshan@gwshan:~$ sudo nping --rate 5 --data-length 1461 --tcp -p 80 10.61.253.42
Max rtt: N/A | Min rtt: N/A | Avg rtt: N/A
Raw packets sent: 5 (7.505KB) | Rcvd: 0 (0B) | Lost: 5 (100.00%)
Nping done: 1 IP address pinged in 1.87 seconds

With the patch applied, the network interface isn't reachable:

gwshan, 2017-02-20:
With the patch applied:

# ifconfig
eth0      Link encap:Ethernet  HWaddr F2:26:16:AC:67:8A  
          inet addr:10.61.253.42  Bcast:10.61.255.255  Mask:255.255.0.0
          inet6 addr: fe80::f026:16ff:feac:678a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:270 errors:0 dropped:0 overruns:0 frame:0
          TX packets:120 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:18166 (17.7 KiB)  TX bytes:7592 (7.4 KiB)

gwshan@gwshan:~/sandbox/l/net/ncsi$ ping 10.61.253.42
PING 10.61.253.42 (10.61.253.42) 56(84) bytes of data.
From 10.61.2.199 icmp_seq=1 Destination Host Unreachable
From 10.61.2.199 icmp_seq=2 Destination Host Unreachable
From 10.61.2.199 icmp_seq=3 Destination Host Unreachable
^C
--- 10.61.253.42 ping statistics ---
4 packets transmitted, 0 received, +3 errors, 100% packet loss, time 2999ms
pipe 4

ozbenh, 2017-02-21:
On Mon, 2017-02-20 at 15:44 -0800, Gavin Shan wrote:
> I tried it on Palmetto, Without the patch from @ozbenh
> gwshan@gwshan:~$ sudo nping --rate 5 --data-length 1460 --tcp -p 80
> 10.61.253.42
> Max rtt: 178.421ms | Min rtt: 178.014ms | Avg rtt: 178.150ms [...]

My patch isn't testing on NC-SI machines. I need to work on that a bit
more.

I've started rewriting the whole driver btw. I got it to go from
270Mbit/s to about 640Mbit/sec on the 2500 eval board.

Cheers,
Ben.

gwshan, 2017-02-21:
Thanks, Ben. The improved speed is inspiring. In the mean while, I will spend some time to locate the bottleneck. Before that, I need to understand how "nping" works. I would guess the bottleneck might be not the driver (MAC layer).

gwshan, 2017-02-21:
First of all, I doubt the result report from nping is reliable enough. According to comment#4 posted by @prkatti1, he was sending short packets obviously because of "iplen=40" shown in the log. I had one similar experiment: Adding code in BMC's kernel to output message when any TCP packet is received.

1. nping reports loss of packets on my laptop:

gwshan@gwshan:~/sandbox/l/net/ncsi$ sudo nping -c 4 --rate 100 --data-length 1460 --tcp -p 80 10.61.253.42
WARNING: Payload exceeds maximum recommended payload (1400)

Starting Nping 0.7.01 ( https://nmap.org/nping ) at 2017-02-21 12:40 AEDT
SENT (0.0400s) TCP 10.61.2.199:57261 > 10.61.253.42:80 S ttl=64 id=10092 iplen=1500  seq=2349726183 win=1480 
SENT (0.0502s) TCP 10.61.2.199:57261 > 10.61.253.42:80 S ttl=64 id=10092 iplen=1500  seq=2349726183 win=1480 
SENT (0.0602s) TCP 10.61.2.199:57261 > 10.61.253.42:80 S ttl=64 id=10092 iplen=1500  seq=2349726183 win=1480 
SENT (0.0702s) TCP 10.61.2.199:57261 > 10.61.253.42:80 S ttl=64 id=10092 iplen=1500  seq=2349726183 win=1480 
RCVD (0.2144s) TCP 10.61.253.42:80 > 10.61.2.199:57261 RA ttl=64 id=61267 iplen=40  seq=0 win=0 
 
Max rtt: 144.162ms | Min rtt: 144.162ms | Avg rtt: 144.162ms
Raw packets sent: 4 (6.000KB) | Rcvd: 1 (46B) | Lost: 3 (75.00%)
Nping done: 1 IP address pinged in 0.24 seconds

2. There are 8 TCP packets captured by tcpdump from my latop. The filtered packet is attached.

[small_pkt_high_speed.pkt.tar.gz](https://github.com/openbmc/openbmc/files/788866/small_pkt_high_speed.pkt.tar.gz)

3. 4 TCP packets (source IP and destination port matched) received on BMC:

TCP: [73] TCP: src_ip=0xc7023d0a, tcp_dst_port=0x5000
TCP: [74] TCP: src_ip=0xc7023d0a, tcp_dst_port=0x5000
TCP: [75] TCP: src_ip=0xc7023d0a, tcp_dst_port=0x5000
TCP: [76] TCP: src_ip=0xc7023d0a, tcp_dst_port=0x5000

gwshan, 2017-02-21:
Also, nping supports 1460 as its payload length. Error reported when specifying the argument "data_length" to 1461. So this seems a nping problem generally?

gwshan@gwshan:~$ sudo nping -c 1 --rate 1 --data-length 1461 --tcp -p 80 10.61.253.42
WARNING: Payload exceeds maximum recommended payload (1400)

Starting Nping 0.7.01 ( https://nmap.org/nping ) at 2017-02-21 13:25 AEDT
sendto in send_ip_packet_sd: sendto(4, packet, 1501, 0, 10.61.253.42, 16) => Message too long

gwshan, 2017-03-09:
According to the last two update, it should be nping's problem. It seems nping cannot receive responses and count them in high-speed scenario.

prkatti1, 2017-04-18:
Same issues is happening with https port 443, intermittently getting connection error from port 443 although BMC IP address is reachable.

williamspatrick, 2017-04-18:
This is an underlying problem with the network driver on the BMC.  @ozbenh did significant updates to it that @shenki is pulling into 4.10.  Once we move onto 4.10 you can retest these scenarios.

sivassrr, 2017-04-19:
@williamspatrick Any story / issue associate with what @ozbenh working on.

sivassrr, 2017-04-19:
May be https://github.com/openbmc/openbmc/issues/1432, I guess.

williamspatrick, 2017-04-19:
@ozbenh does not work on stories.  There have been a number of issues about network timeout errors, many of which were assigned to @gwshan.  We will probably close any of these older issues once we get the new network driver in.

williamspatrick, 2017-05-30:
4.10 has been integrated into master with the newest network driver.  Please retest.

closed 2017-05-30 by williamspatrick