openbmc/openbmc #1124: [Code update] Phanton reboot post Update

gkeishin, 2017-02-14:
```
There is an extra reboot post code update is completed, It's not from code update script for sure.. Here is a console log
```

**Driver details**

```
root@witherspoon:~# cat /etc/version
20170214120856
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-35"
VERSION_ID="v1.99.2-35-ga0491c2-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-35"
BUILD_ID="v1.99.2"
root@witherspoon:~#

```

**Code update completed**

```
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 witherspoon ttyS4

witherspoon login: root
Password:
root@witherspoon:~# uptime
 18:56:32 up 1 min,  load average: 7.40, 2.22, 0.77
root@witherspoon:~# cat /etc/version
ca20170214120856
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-35"
VERSION_ID="v1.99.2-35-ga0491c2-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-35"
BUILD_ID="v1.99.2"
root@witherspoon:~# uptime
 18:56:53 up 1 min,  load average: 7.62, 2.60, 0.92
random: nonblocking pool is initialized
root@witherspoon:~# journalctl -f
-- Logs begin at Tue 2017-01-31 18:55:30 UTC. --
```

**Journal log started and rebooted after few seconds latter as can be seen in bellow ** 
```
Jan 31 18:57:22 witherspoon systemd[1]: Created slice system-dropbear.slice.
Jan 31 18:57:22 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:53438).
Jan 31 18:57:22 witherspoon dropbear[910]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 31 18:57:22 witherspoon dropbear[910]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 31 18:57:22 witherspoon dropbear[910]: Child connection from ::ffff:9.41.165.103:53438
Jan 31 18:57:22 witherspoon dropbear[910]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:53438
Jan 31 18:57:26 witherspoon hwmon.py[827]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon0/
Jan 31 18:57:26 witherspoon hwmon.py[827]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon1/
Jan 31 18:57:26 witherspoon hwmon.py[827]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon2/
Jan 31 18:57:26 witherspoon hwmon.py[827]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon3/
Jan 31 18:57:26 witherspoon hwmon.py[827]: WARNING - hwmon: Unhandled hwmon: /sys/class/hwmon/hwmon4/
Jan 31 18:57:37 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:53440).
Jan 31 18:57:37 witherspoon dropbear[914]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 31 18:57:37 witherspoon dropbear[914]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 31 18:57:37 witherspoon dropbear[914]: Child connection from ::ffff:9.41.165.103:53440
Jan 31 18:57:38 witherspoon dropbear[914]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:53440
Jan 31 18:57:39 witherspoon systemd[1]: Started SSH Per-Connection Server (9.41.165.103:53442).
Jan 31 18:57:39 witherspoon dropbear[917]: Failed loading /etc/dropbear/dropbear_dss_host_key
Jan 31 18:57:39 witherspoon dropbear[917]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 31 18:57:39 witherspoon dropbear[917]: Child connection from ::ffff:9.41.165.103:53442
Jan 31 18:57:40 witherspoon dropbear[917]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:53442
[  OK  ] Stopped target Timers.
         Stopping Phosphor Webserver...
[... 187 lines cut ...]
systemd-shutdown[1]: Returning to initrd...
watchdog: watchdog0: watchdog did not stop!
shutdown: reboot --log-level 6 --log-target kmsg --log-color
+ awk /oldroot|mnt/ { print $2 }
+ sort -r
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
+ umount /mnt
+ set +x
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=215708k,nr_inodes=53927,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
reboot: Restarting system

```

vishwabmc, 2017-02-15:
@gkeishin and I had a detailed discussion about this and my suggestion was that he `clear journal` and then `start` the code update and then we could look at journal and see the number of instances of reboot and possibly the source and while all this is happening, watch the console.

@gkeishin Is this recreate post following that procedure ?

Do you have journal ?

Also, Is there a way you can disconnect the test soon after the second reset takes place as part of code update ?. I understand you have said its not a script that is doing all these but need to confirm. :)

gkeishin, 2017-02-15:
@vishwabmc   This is also reported from DVT folks as well. 

- [ ]   Clear journal and start one we have not yet added the code for it. We may plan to add in next sprint
- [x]   It's just a normal code update procedure 
- [x] Yes we can do that as a sandbox to exit soon after the update is completed.

vishwabmc, 2017-02-15:
@gkeishin We would need this test tried with the config I suggested and then we can take a look at it and until then, this defect will be kept on hold.

Also, Can you try this code update test manually once  ?. that may be another data point to see if its the robot script that is doing it or something inside the code.

gkeishin, 2017-05-24:
This we see there is a bug in automation too. fixed it.. and we seems to be good here.. With new code update coming in the process will also change.. so closing it..

closed 2017-05-24 by gkeishin