openbmc/openbmc #1126: [Witherspoon w37 v1.99.2-35-ga0491c2] Poweron stuck at Inventory Upload

gkeishin, 2017-02-14:
**Driver info** 
```
Build : https://openpower.xyz/job/openbmc-build/537/
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-35"
VERSION_ID="v1.99.2-35-ga0491c2-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-35"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170214120856
root@witherspoon:~#
```

- [x]  Journal log  
[journald_v1.99.2-35-ga0491c2_20170214120856_Inventory_upload.txt](https://github.com/openbmc/openbmc/files/774171/journald_v1.99.2-35-ga0491c2_20170214120856_Inventory_upload.txt)

```
Jan 31 18:57:20 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jan 31 18:57:20 witherspoon kernel[743]: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jan 31 18:57:20 witherspoon systemd[1]: Started Disable UCD VCS rails.
Jan 31 18:57:20 witherspoon systemd[1]: Starting Perform AVS bus workaround on VRMs...
Jan 31 18:57:20 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/state/chassis0.
Jan 31 18:57:21 witherspoon systemd[1]: Started Wait for /org/openbmc/watchdog/host0.
Jan 31 18:57:21 witherspoon systemd[1]: Started Perform AVS bus workaround on VRMs.
Jan 31 18:57:21 witherspoon systemd[1]: Starting Disable the AVS bus on the VRMs...
Jan 31 18:57:21 witherspoon systemd[1]: Started Disable the AVS bus on the VRMs.
Jan 31 18:57:22 witherspoon systemd[1]: Starting Apply voltage overrides to VRMs...
Jan 31 18:57:22 witherspoon vrm-control.sh[943]: rail        set        read        current
Jan 31 18:57:22 witherspoon vrm-control.sh[943]: -------        ------- ------- -------
Jan 31 18:57:22 witherspoon vrm-control.sh[943]: vdna        0.898V        0.004V        0.000A
Jan 31 18:57:23 witherspoon vrm-control.sh[943]: vddra        1.352V        0.012V        0.000A
Jan 31 18:57:23 witherspoon vrm-control.sh[943]: vdnb        0.898V        0.004V        0.000A
Jan 31 18:57:24 witherspoon vrm-control.sh[943]: vddrb        1.352V        0.012V        0.000A
Jan 31 18:57:24 witherspoon systemd[1]: Started Apply voltage overrides to VRMs.
Jan 31 18:57:24 witherspoon systemd[1]: Starting Enable the AVS bus on VRMs...
Jan 31 18:57:24 witherspoon systemd[1]: Started Enable the AVS bus on VRMs.
Jan 31 18:57:24 witherspoon systemd[1]: Starting Disable VCS before Witherspoon power on...
Jan 31 18:57:25 witherspoon systemd[1]: Started Disable VCS before Witherspoon power on.
Jan 31 18:57:25 witherspoon systemd[1]: Reached target Power0 On (Pre).
Jan 31 18:57:25 witherspoon systemd[1]: Started Start Power0.
Jan 31 18:57:25 witherspoon system_manager.py[821]: Running System State: HOST_POWERING_ON
Jan 31 18:57:25 witherspoon timemanager[874]:  PGOOD has changed..
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: ERROR (name_properties): "name" property in /bmc/inventory/misc is incorrect ("0000000000000000" instead of base node name)
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: ERROR (name_properties): "name" property in /bmc/inventory/cpu0 is incorrect ("PROCESSOR MODULE" instead of base node name)
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: ERROR: Input tree has errors, aborting (use -f to force output)
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: Uploading inventory to PNOR in dtb format...
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: About to erase 0x02744000..0x0274d000 !
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: Erasing...
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: [570B blob data]
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: About to program "/tmp/tmpxg5YTg" at 0x02744000..0x0274d000 !
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: Programming & Verifying...
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: [56B blob data]
Jan 31 18:57:28 witherspoon inventory_upload.py[899]: Updating actual size in partition header...
Jan 31 18:57:28 witherspoon systemd[1]: Started Inventory Upload.
```

geissonator, 2017-02-14:
@gkeishin Did the build work on other systems?  Did you try a power cycle?  Was this the first boot?

gkeishin, 2017-02-14:
@geissonator The is like 2nd time poweron when I try to boot the OS...  I rebooted the system and then try to poweron...

gkeishin, 2017-02-14:
Now doing power cycle test and will see if that hits this one as well..

spinler, 2017-02-14:
If this happens again, can you do a:
```pdbg -a -b fsi -d p9w getcfam 0x1007```
```pdbg -a -b fsi -d p9w getcfam 0x2809```
that could give us a clue about hostboot status.  the traces show we successfully kicked off hostboot, they just never got anywhere.  Uma saw this yesterday on w38.

gkeishin, 2017-02-14:
@spinler Sure.. I next time i hit this , I ll collect those info..

spinler, 2017-02-14:
Oh, here is the problem:
```Jan 31 18:55:48 witherspoon systemd[1]: Failed to start Phosphor Inband IPMI.```

Do you have a fix for this @dkodihal?

vishwabmc, 2017-02-15:
`Failed to start Phosphor Inband IPMI.` is our classic host-ipmid. Why is this failing ?. I know we have had some issues around starting IPMI*

dkodihal, 2017-02-15:
@spinler This wasn't starting due to a segfault, which @ratagupt fixed. The fix has been merged to ipmi-fru-parser yesterday.

spinler, 2017-02-15:
@dkodihal So this issue can be closed?

dkodihal, 2017-02-15:
Yep.

gkeishin, 2017-02-15:
Closing as requested and in today run I' don't see this failure.

closed 2017-02-15 by gkeishin