openbmc/openbmc #1128: phosphor-bmc-state-manager : Use a call to systemd to resolve 'obmc_standby.target'

JoshDKing, 2017-02-14:
In phosphor-state-manager/bmc_state_manager.cpp we currently have:
constexpr auto SYSTEMD_TGT_PATH      = "/org/freedesktop/systemd1/unit/"                                          "obmc_2dstandby_2etarget";

It would be great to somehow get rid of "obmc_2dstandby_2etarget" and instead use a systemd call (possibly GetUnit) to resolve that into a path.

https://gerrit.openbmc-project.xyz/#/c/2227/2/bmc_state_manager.cpp@33

vishwabmc, 2017-02-15:
@rfrandse hey Reed, I have a request on the color that is used to label the "Phase". This 'Red' color makes me think its a 'Bug' :).  May be some other color ?.. Thank you !

geissonator, 2018-07-12:
This was resolved, code uses obmc-standby.target

closed 2018-07-12 by geissonator