openbmc/openbmc #1129: [Witherspoon w33 - v1.99.2-46-g04f01f7 ] PGOOD poll timedout

gkeishin, 2017-02-15:
This could be system specific issue, the same driver works good on other
**Driver Info:**
```
root@witherspoon:~# cat /etc/version
20170215035323
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-46"
VERSION_ID="v1.99.2-46-g04f01f7-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-46"
BUILD_ID="v1.99.2"
root@witherspoon:~#

PNOR BUILD:
open-power-witherspoon-713df6c-dirty
        buildroot-6b9fd80
        skiboot-8c51fda-opdirty
        hostboot-eb217e5-opdirty
        linux-4.9.4-opdirty-bdade6f
        petitboot-v1.4.0-opdirty-5ad2029
        witherspoon-xml-site_local-wghoffa-3a71790-dirty
        occ-4141b5f-opdirty
        hostboot-binaries-01cbe09-opdirty
        capp-ucode-1bb7503-opdirty

```

```
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: Uploading inventory to PNOR in dtb format...
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: WARNING: Erase region adjusted to 0x027a0000..0x027a2000
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: About to erase 0x027a0000..0x027a2000 !
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: Erasing...
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: [171B blob data]
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: About to program "/tmp/tmpWCNDWl" at 0x027a0000..0x027a1852 !
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: Programming & Verifying...
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: [114B blob data]
Jan 31 18:58:25 witherspoon inventory_upload.py[907]: Updating actual size in partition header...
Jan 31 18:58:26 witherspoon systemd[1]: Started Inventory Upload.
Jan 31 18:58:38 witherspoon power_control.exe[854]: ERROR PowerControl: Pgood poll timeout
Jan 31 18:58:38 witherspoon timemanager[847]:  PGOOD has changed..
```

```
root@witherspoon:~# obmcutil state
 = HOST_POWERING_ON
root@witherspoon:~# uptime
 19:00:58 up 5 min,  load average: 0.23, 0.98, 0.58
root@witherspoon:~# journalctl --no-pager > journald_v1.99.2-46-g04f01f7_20170215035323_Hoffa_PGOOD_W33.txt
root@witherspoon:~# pdbg -a -b fsi -d p9w getcfam 0x1007
getaddr: Error reading register
No valid targets found or specified. Try adding -p/-c/-t options to specify a target.
Alternatively run pdbg -a probe to get a list of all valid targets
root@witherspoon:~# pdbg -a -b fsi -d p9w getcfam 0x2809
getaddr: Error reading register
No valid targets found or specified. Try adding -p/-c/-t options to specify a target.
Alternatively run pdbg -a probe to get a list of all valid targets
root@witherspoon:~# pdbg -a probe
Error opening bus: Bad address
backend_init: Unable to I2C initialise backend
root@witherspoon:~# 
```
- [x]   Journal log 
[journald_v1.99.2-46-g04f01f7_20170215035323_Hoffa_PGOOD_W33.txt](https://github.com/openbmc/openbmc/files/776326/journald_v1.99.2-46-g04f01f7_20170215035323_Hoffa_PGOOD_W33.txt)

vishwabmc, 2017-02-15:
@gkeishin: is this a one off or something you constantly see on this machine ?. 

To me it looks like the pgood hw signal never was raised ( power did not stabilize ?? ) / we did not wait longer / or that we started to look for pgood signal after it was sent ?

@geissonator Is it anything related to the reordering with the new state manager ?

gkeishin, 2017-02-15:
@vishwabmc  Yes , I see this constantly for this PNOR and BMC driver build, but can't comment for others builds since I have not tested on this particular system earlier.

geissonator, 2017-02-15:
This just looks like the pgood pin not coming on.  Does an A/C cycle fix it?

gkeishin, 2017-02-15:
@geissonator AC cycle didnt fix this one ...

spinler, 2017-02-15:
Maybe ask Uma to try cronus scripts and see if those can power it on.  If it can't, we can ask for help from the HW team.

yadlapati, 2017-02-15:
on script worked to power on the system. Even obmcutil poweron worked and host OS is booted.

gkeishin, 2017-02-15:
@yadlapati This is on w33 and I can still see the error there...

yadlapati, 2017-02-15:
w33 is up and running after power cycle

gkeishin, 2017-02-15:
@yadlapati Great. I'll try on new driver and confirm once.. and will close this.. Thanks  for the help..

gkeishin, 2017-02-16:
Tested and found good in new builds

closed 2017-02-16 by gkeishin