openbmc/openbmc #1130: [Witherspoon w33 - v1.99.2-46-g04f01f7 ] Phosphor Hwmon Poller Crashed

gkeishin, 2017-02-15:
**Boot Host with new PNOR Build. The OS boots but saw this crash.**

**Driver info**

```
root@witherspoon:~# cat /etc/version
20170215035323
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-46"
VERSION_ID="v1.99.2-46-g04f01f7-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-46"
BUILD_ID="v1.99.2"
root@witherspoon:~#

PNOR BUILD:
open-power-witherspoon-713df6c-dirty
        buildroot-6b9fd80
        skiboot-8c51fda-opdirty
        hostboot-eb217e5-opdirty
        linux-4.9.4-opdirty-bdade6f
        petitboot-v1.4.0-opdirty-5ad2029
        witherspoon-xml-site_local-wghoffa-3a71790-dirty
        occ-4141b5f-opdirty
        hostboot-binaries-01cbe09-opdirty
        capp-ucode-1bb7503-opdirty
```
**Service status** 

```
root@witherspoon:~# systemctl status xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service | cat
● xyz.openbmc_project.Hwmon@ahb-apb-i2cx401e78a000-i2cx2dbusx40400-ucd90160x4064.service - Phosphor Hwmon Poller
   Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Hwmon@.service; static; vendor preset: enabled)
   Active: inactive (dead)
root@witherspoon:~#
```

- [x] Journal log   
[journald_v1.99.2-46-g04f01f7_20170215035323_w38.txt](https://github.com/openbmc/openbmc/files/776906/journald_v1.99.2-46-g04f01f7_20170215035323_w38.txt)

```
Jan 31 19:06:35 witherspoon phosphor-hwmon-readd[1160]: terminate called after throwing an instance of 'std::ios_base::failure'
Jan 31 19:06:35 witherspoon phosphor-hwmon-readd[1160]:   what():  basic_filebuf::underflow error reading the file
Jan 31 19:06:35 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=dumped, status=6/ABRT
Jan 31 19:06:35 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Jan 31 19:06:35 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'core-dump'.
Jan 31 19:06:35 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Service hold-off time over, scheduling restart.
Jan 31 19:06:35 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jan 31 19:06:35 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jan 31 19:06:35 witherspoon phosphor-hwmon-readd[1175]: terminate called after throwing an instance of 'std::ios_base::failure'
Jan 31 19:06:35 witherspoon phosphor-hwmon-readd[1175]:   what():  basic_filebuf::underflow error reading the file
Jan 31 19:06:35 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=dumped, status=6/ABRT
Jan 31 19:06:35 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Jan 31 19:06:35 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'core-dump'.
Jan 31 19:06:36 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Service hold-off time over, scheduling restart.
Jan 31 19:06:36 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jan 31 19:06:36 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jan 31 19:06:36 witherspoon phosphor-hwmon-readd[1178]: terminate called after throwing an instance of 'std::ios_base::failure'
Jan 31 19:06:36 witherspoon phosphor-hwmon-readd[1178]:   what():  basic_filebuf::underflow error reading the file
Jan 31 19:06:36 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=dumped, status=6/ABRT
Jan 31 19:06:36 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Jan 31 19:06:36 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'core-dump'.
```

gkeishin, 2017-02-16:
**In newer build as well** 

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-60"
VERSION_ID="v1.99.2-60-gcb1cc2b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-60"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170216042029
root@witherspoon:~# uptime
 05:52:27 up 13 min,  load average: 0.12, 0.32, 0.41
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~#
```

```
Feb 16 06:04:29 witherspoon systemd[1]: Reached target Phosphor System - Standby.
Feb 16 06:04:29 witherspoon systemd[1]: Startup finished in 6.655s (kernel) + 1min 33.036s (userspace) = 1min 39.691s.
Feb 16 06:04:29 witherspoon phosphor-bmc-state-manager[847]: BMC_READY
Feb 16 06:04:29 witherspoon phosphor-bmc-state-manager[847]: Setting the BMCState field
Feb 16 06:04:38 witherspoon phosphor-hwmon-readd[755]: terminate called after throwing an instance of 'std::ios_base::failure'
Feb 16 06:04:38 witherspoon phosphor-hwmon-readd[755]:   what():  basic_filebuf::underflow error reading the file
Feb 16 06:04:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=dumped, status=6/ABRT
Feb 16 06:04:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Feb 16 06:04:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'core-dump'.
Feb 16 06:04:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Service hold-off time over, scheduling restart.
Feb 16 06:04:39 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Feb 16 06:04:39 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Feb 16 06:04:39 witherspoon phosphor-hwmon-readd[886]: terminate called after throwing an instance of 'std::ios_base::failure'
Feb 16 06:04:39 witherspoon phosphor-hwmon-readd[886]:   what():  basic_filebuf::underflow error reading the file
Feb 16 06:04:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=dumped, status=6/ABRT
Feb 16 06:04:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Feb 16 06:04:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'core-dump'.
Feb 16 06:04:39 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Service hold-off time over, scheduling restart.
Feb 16 06:04:39 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Feb 16 06:04:39 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
```
**Service status**

```
root@witherspoon:~# systemctl status xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service | cat
● xyz.openbmc_project.Hwmon@ahb-apb-i2cx401e78a000-i2cx2dbusx40400-ucd90160x4064.service - Phosphor Hwmon Poller
   Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Hwmon@.service; static; vendor preset: enabled)
   Active: inactive (dead)
root@witherspoon:~#
```

- [x]   Journal log  
[journald_v1.99.2-60-gcb1cc2b_20170216042029_POLLER_FAILED.txt](https://github.com/openbmc/openbmc/files/779231/journald_v1.99.2-60-gcb1cc2b_20170216042029_POLLER_FAILED.txt)

shenki, 2017-02-17:
@bradbishop I saw this one today. You mentioned in you were interested in hwmon-readd crashes

geissonator, 2017-02-17:
I saw this on the openFSI based boots last night on witherspoon as well.

gkeishin, 2017-02-21:
As requested by @ratagupt adding follow cores seen on one of the system

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-72"
VERSION_ID="v1.99.2-72-gd11ec2e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-72"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170221015852

```

- [x] Core dump data  -
[core_phosphor-hwmon.tar.gz](https://github.com/openbmc/openbmc/files/789339/core_phosphor-hwmon.tar.gz)

bradbishop, 2017-02-23:
found this possibly related bug:

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=53984

bradbishop, 2017-02-23:
I added some additional tracing and this happens while reading the hwmon sysfs entries.  I'm trying to make sense of what the correct way to get errno is when using the c++ io subsystem when these happen.

bradbishop, 2017-02-23:
caught this in strace:

```
open("/sys/class/hwmon/hwmon3/in9_input", O_RDONLY|O_LARGEFILE) = 4
read(4, 0x1c688d8, 8191)                = -1 EREMOTEIO (Remote I/O error)
write(2, "terminate called after throwing "..., 48) = 48```

bradbishop, 2017-02-23:
@shenki I'm guessing this is likely coming from ast_i2c_xfer, which seems to suggest the master is reporting a command error.

If I retry my read system call it ends up working the next time.

shenki, 2017-02-23:
`ast_i2c_xfer` is calling `ast_i2c_do_msgs_xfer`. @amboar was looking at IRQ completion as the cause of #998.

@bradbishop could you reproduce your error with debugging turned on?

```
# echo file drivers/i2c/busses/i2c-aspeed.c +p > /sys/kernel/debug/dynamic_debug/control
# ./testcase
#   echo file drivers/i2c/busses/i2c-aspeed.c -p > /sys/kernel/debug/dynamic_debug/control
```

And attach the output of `dmesg`?

shenki, 2017-02-24:
I reproduced this when attempting to read from the `temp1_input`. hwmon0 in this case the the ucd90160.

This was with a patch added to the kernel that removed all locking from the Aspeed i2c bus driver.

```
root@zaius:~# cat /sys/class/hwmon/hwmon0/name 
ucd90160
root@zaius:~# for f in `ls /sys/class/hwmon/hwmon0/*_input`; do echo -n $f; echo -n " "; cat $f; done
/sys/class/hwmon/hwmon0/in10_input 3
/sys/class/hwmon/hwmon0/in11_input 3
/sys/class/hwmon/hwmon0/in12_input 1
/sys/class/hwmon/hwmon0/in13_input 3
/sys/class/hwmon/hwmon0/in1_input 12076
/sys/class/hwmon/hwmon0/in2_input 8
/sys/class/hwmon/hwmon0/in3_input 3
/sys/class/hwmon/hwmon0/in4_input 1
/sys/class/hwmon/hwmon0/in5_input 1
/sys/class/hwmon/hwmon0/in6_input 3
/sys/class/hwmon/hwmon0/in7_input 4
/sys/class/hwmon/hwmon0/in8_input 3
/sys/class/hwmon/hwmon0/in9_input 6
/sys/class/hwmon/hwmon0/temp1_input cat: read error: Remote I/O error
```

shenki, 2017-02-24:
I booted a kernel with a larger kernel log buffer:
```
console=ttyS4,115200n8 root=/dev/ram rw log_buf_len=1M
```

and then ran this test:
```
root@zaius:~# echo file drivers/i2c/busses/i2c-aspeed.c +p > /sys/kernel/debug/dynamic_debug/control && cat /sys/class/hwmon/hwmon0/temp1_input && echo file drivers/i2c/busses/i2c-aspeed.c -p > /sys/kernel/debug/dynamic_debug/control
cat: read error: Remote I/O error
```

The output is attached in [i2c-log.txt](https://github.com/openbmc/openbmc/files/798382/i2c-log.txt)

shenki, 2017-02-24:
[0001-i2c-aspeed-drop-the-locks.patch.txt](https://github.com/openbmc/openbmc/files/798388/0001-i2c-aspeed-drop-the-locks.patch.txt) 🎤⬇️️

shenki, 2017-02-24:
Note that the dropping of the locks patch is not related to the fix for this issue. I can reproduce the bug both with and without that patch on Zaius (4.7.10-6107661a7855efc07dfbe8f2b8847dbf2326e653)

shenki, 2017-02-24:
@bradbishop what file is your C++ program crashing on?

Can you provide a small test case?

Do we know if temp1 should be providing valid data?

spinler, 2017-02-24:
There's no requirement (on IBM systems at least) to report the temperature for that chip.

shenki, 2017-03-02:
Can we make the exception print the name of the file, to ease future debugging?

If we make that change we can confirm that it is only happening for the input data.

bradbishop, 2017-03-03:
@shenki 

you should be able to

```journalctl -o verbose |grep CALLOUT```

to see the file having trouble.

bradbishop, 2017-03-03:
I haven't had a chance to come up with a testcase Joel..will shoot for something on Monday.

bradbishop, 2017-03-09:
@shenki My application only reads from the foox_input files.

you can hop any any Barreleye or Witherspoon and see this with a command like:

```journalctl -o verbose xyz.openbmc_project.Hwmon* |grep CALLOUT_DEVICE_PATH```

Here is a sample:

```    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002e
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002e
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002e
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002e
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002e
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d
    CALLOUT_DEVICE_PATH=/sys/devices/platform/ahb/ahb:apb/1e78a000.i2c/i2c-6/i2c-6/6-002d

bradbishop, 2017-03-09:
@shenki You mentioned being able to reproduce this with cat - do you still want a testcase?

Are you ready for this to be assigned to you?  Latest userspace is:

1 - not coredumping
2 - logging the device path and errno on failure
3 - exitting gracefully

I'm not sure what more I can do with this.  Any more data you'd like logged?

gkeishin, 2017-03-16:
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/630/
- [x] VERSION_ID : v1.99.3-93-gf0a463a
- [x] Kernel     : Linux witherspoon 4.7.10-c72ced5a19ceae19c7d792ae3e982bee5607dab4 #1 Wed Mar 15 22:22:06 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```

```
root@witherspoon:~# ls -larth /tmp/core_phosphor-hwmon-.3733
-rw-------    1 root     root      476.0K Feb 22 06:47 /tmp/core_phosphor-hwmon-.3733
root@witherspoon:~#

Feb 22 06:47:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit not needed anymore. Stopping.
Feb 22 06:47:07 witherspoon phosphor-hwmon-readd[3733]: terminate called after throwing an instance of 'std::experimental::filesystem::v1::__cxx11::filesystem_error'
Feb 22 06:47:07 witherspoon phosphor-hwmon-readd[3733]:   what():  filesystem error: directory iterator cannot open directory: No such file or directory [/sys/class/hwmon/hwmon4]
Feb 22 06:47:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=dumped, status=6/ABRT
Feb 22 06:47:07 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Feb 22 06:47:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit entered failed state.
Feb 22 06:47:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed with result 'core-dump'.
Feb 22 06:47:07 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit not needed anymore. Stopping.
```
unbind the ucd90160 device as part of the power-on process ... so the sysfs entries for it disappear and crash the associated hwmon poller.
the hwmon poller needs to do something more robust than just crash out in this particular case.

gkeishin, 2017-03-16:
- [x] Journal log  
[journald_v1.99.3-93-gf0a463_hwmon_core.txt](https://github.com/openbmc/openbmc/files/847329/journald_v1.99.3-93-gf0a463_hwmon_core.txt)

- [x] Core dump file  
[core_phosphor-hwmon-.3733.gz](https://github.com/openbmc/openbmc/files/847331/core_phosphor-hwmon-.3733.gz)

williamspatrick, 2017-03-16:
Don't we have the hwmon service 'BindsTo' the device? Shouldn't it be automatically shut down when the device goes away or is there a race condition here we need to take care of?  Maybe on file not found we should exit quietly and on read errors exit with error. 

George, we expect the application to at least exit when the device goes away.

gkeishin, 2017-03-16:
Yes but it also cored out that's not something we want here while exiting..

geissonator, 2017-03-30:
Continue to fill our journals with this one.  Doesn't hurt anything, just messy.

williamspatrick, 2017-03-30:
There are two different issues reported here. 

1. Sometime we get an error on read. This is a kernel issue. 

2. When unbound, the device disappears and hwmon cores. This is an hwmon issue. 

Since this issue was given to @shenki for 1 please open another issue for 2 for @bradbishop.

gkeishin, 2017-05-26:
On new 4.10 kernel, similar crashed with different signature

````
•	BUILD ID   : https://openpower.xyz/job/openbmc-build/766/
•	VERSION_ID : v1.99.6-10-g1d22c22
•	Kernel     : Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Fri May 26 05:22:23 UTC 2017 armv6l GNU/Linux

Apr 20 17:42:26 witherspoon phosphor-hwmon-readd[1046]: terminate called after throwing an instance of 'std::experimental::filesystem::v1::__cxx11::filesystem_error'
Apr 20 17:42:26 witherspoon phosphor-hwmon-readd[1046]:   what():  filesystem error: cannot canonicalize: No such file or directory [/sys/class/hwmon/hwmon4/device] [/]
Apr 20 17:42:26 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Apr 20 17:42:26 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Main process exited, code=dumped, status=6/ABRT
```
- [x]  Journald log 
[journald_hwmon_v1.99.6-10-g1d22c22_crashed.txt](https://github.com/openbmc/openbmc/files/1030982/journald_hwmon_v1.99.6-10-g1d22c22_crashed.txt)

bjwyman, 2017-05-31:
This seems like it might be related to the code I am working on to find the BMP280 and other IIO HWMON driver instances. openbmc/openbmc#1268.

Most witherspoons I have looked at have the IIO HWMON files for BMP280 in hwmon4.

bjwyman, 2017-06-06:
Actually, looks like hwmon4 was for the UCD device in this case.
I am not certain if it is related to the updates from #1265.

geissonator, 2017-07-27:
@gkeishin - Can you see if this is still happening?

gkeishin, 2017-07-27:
Not on the same bus but on different one
```
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Service hold-off time over, scheduling restart.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Service hold-off time over, scheduling restart.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Service hold-off time over, scheduling restart.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 25 19:25:32 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Unit not needed anymore. Stopping.
```

@geissonator

shenki, 2017-09-25:
There doesn't appear to be an outstanding kernel issue here. Removing myself as assignee

gkeishin, 2017-09-25:
I  ll close this one and open new for the newer HWMOn failures seen

closed 2017-09-25 by gkeishin

AloneWarrior11011, 2025-04-23:
> There are two different issues reported here.
>
>     1. Sometime we get an error on read. This is a kernel issue.
>
>     2. When unbound, the device disappears and hwmon cores. This is an hwmon issue. [...]

```
root@raspberrypi3:~# systemctl status   xyz.openbmc_project.Hwmon@-soc-i2c\x407e804000-tmp75\x404d.service   
* xyz.openbmc_project.Hwmon@-soc-i2cx407e804000-tmp75x404d.service - Phosphor Hwmon Poller
     Loaded: loaded (/usr/lib/systemd/system/xyz.openbmc_project.Hwmon@.service; static)
     Active: inactive (dead)

Apr 22 07:25:19 raspberrypi3 systemd[1]: Phosphor Hwmon Poller was skipped because of an unmet co...conf).
Hint: Some lines were ellipsized, use -l to show in full.

```

I got the same kind of issue as of my sensor configs are under /etc/default/obmc/hwmon/soc/i2c@7e804000/tmp75@4d.conf. 
But I found that sensor got exposed on dbus already. Do help me with this issue.

williamspatrick, 2025-04-23:
> > There are two different issues reported here.
> > ```
> > 1. Sometime we get an error on read. This is a kernel issue.
> > 
> > 2. When unbound, the device disappears and hwmon cores. This is an hwmon issue.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Since this issue was given to [@shenki](https://github.com/shenki) for 1 please open another issue for 2 for [@bradbishop](https://github.com/bradbishop).
> 
> ```
> root@raspberrypi3:~# systemctl status   xyz.openbmc_project.Hwmon@-soc-i2c\x407e804000-tmp75\x404d.service   
> * xyz.openbmc_project.Hwmon@-soc-i2cx407e804000-tmp75x404d.service - Phosphor Hwmon Poller
>      Loaded: loaded (/usr/lib/systemd/system/xyz.openbmc_project.Hwmon@.service; static)
>      Active: inactive (dead)
> 
> Apr 22 07:25:19 raspberrypi3 systemd[1]: Phosphor Hwmon Poller was skipped because of an unmet co...conf).
> Hint: Some lines were ellipsized, use -l to show in full.
> ```
> 
> I got the same kind of issue as of my sensor configs are under /etc/default/obmc/hwmon/soc/i2c@7e804000/[tmp75@4d.conf](mailto:tmp75@4d.conf). But I found that sensor got exposed on dbus already. Do help me with this issue.

Responding to other unrelated and closed issues is not likely to get you traction at solving your problems.

AloneWarrior11011, 2025-04-23:
> Responding to other unrelated and closed issues is not likely to get you traction at solving your problems.

sorry for that, do I suppose to create an issue separately, Please do help me to resolve that issue.