openbmc/openbmc #1132: [Witherspoon w38 ]  Kernel panic - not syncing: stack-protector: Kernel stack is corrupted

gkeishin, 2017-02-15:
Something strange happened when booting OS.

**Driver Info**
````
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-46"
VERSION_ID="v1.99.2-46-g04f01f7-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-46"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170215035323
root@witherspoon:~#
````

**Saw this on Journald but as soon power on boom the kernel panic**

```
root@witherspoon:~# journalctl --no-pager | grep host_checkstop.exe
Jan 31 18:56:49 witherspoon host_checkstop.exe[834]: GPIO Lookup:  CHECKSTOP = 354,falling

root@witherspoon:~# obmcutil poweron
root@witherspoon:~# Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: 802e62f8

CPU: 0 PID: 887 Comm: ucd_disable_vcs Not tainted 4.7.10-5f605673ef9354d7653ab0c8eef80007a7fbb8b1 #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<8016d06c>] (panic+0xb8/0x248)
[<8016d06c>] (panic) from [<80110df4>] (__stack_chk_fail+0x10/0x14)
[<80110df4>] (__stack_chk_fail) from [<802e62f8>] (i2c_smbus_xfer+0x508/0x54c)
[<802e62f8>] (i2c_smbus_xfer) from [<ffffffff>] (0xffffffff)
Rebooting in 1 seconds..

U-Boot 2016.07 (Jan 31 2017 - 20:42:15 +0000)

DRAM:  496 MiB
Flash: 32 MiB
```

- [x] /proc  data post recovery  
[proc.tar.gz](https://github.com/openbmc/openbmc/files/777424/proc.tar.gz)

- [x] Serial console log   
[Kernel_panic_serial_console.txt](https://github.com/openbmc/openbmc/files/777430/Kernel_panic_serial_console.txt)

- [x] Journal log  
[journald_v1.99.2-46-g04f01f7_20170215035323_w38_checkstop.txt](https://github.com/openbmc/openbmc/files/777427/journald_v1.99.2-46-g04f01f7_20170215035323_w38_checkstop.txt)

amboar, 2017-02-16:
Duplicate of #998

gkeishin, 2017-02-16:
Sure then @amboar  Closing this one up..

closed 2017-02-16 by gkeishin