openbmc/openbmc #1135: boot_flags does not support the value "BIOS" to stop at petit boot

trajeswaran, 2017-02-15:
Looks like boot_flags should be set to "BIOS" to stop at petit boot when the host is powered on. 

I am getting error when I tried to set "BIOS" to boot_flags". 

As per the yaml file description, the possible values for boot_flags are "Network", "Disk", "Safe", "CDROM", "Setup", "default"

 https://github.com/openbmc/phosphor-settingsd/blob/master/settings.yaml

williamspatrick, 2017-02-15:
The IPMI code itself currently only supports the following:
```
bootOptionTypeMap_t g_bootOptionTypeMap_t[] = {

    {0x01, "Network"},
    {0x02, "Disk"},
    {0x03, "Safe"},
    {0x05, "CDROM"},
    {0x06, "Setup"},
    {0x00, "Default"},
    {0xFF, INVALID_STRING}
};
```

So that is where that list in the YAML comes from.

The IPMI spec does list:
    0110b = Force boot into BIOS Setup

@sammj - Does Petitboot support this flag if we were to add support for it?

williamspatrick, 2017-02-15:
```
bool ipmi_bootdev_is_valid(int x)
{
	switch (x) {
	case IPMI_BOOTDEV_NONE:
	case IPMI_BOOTDEV_NETWORK:
	case IPMI_BOOTDEV_DISK:
	case IPMI_BOOTDEV_SAFE:
	case IPMI_BOOTDEV_CDROM:
	case IPMI_BOOTDEV_SETUP:
		return true;
	}

	return false;
}
```

```
lib/types/types.h:      IPMI_BOOTDEV_SETUP = 0x6,
```

williamspatrick, 2017-02-15:
I just realized that "Setup" is 0x06, which is the IPMI field "Force boot into BIOS Setup".  @trajeswaran - I think "Setup" is what you're looking for.

closed 2017-02-15 by williamspatrick