openbmc/openbmc #1137: Need to support dbus messages in openbmc CI

geissonator, 2017-02-15:
Looks like the docker config needs some tweaks to support some of our gtests which use DBUS.

Here's the hang: https://openpower.xyz/job/openbmc-repository-ci/290/console

```
21:24:31 make[2]: Entering directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/sdbusplus/test'
21:24:31   CXX      message/message_append-append.o
21:24:33   CXXLD    message_append
21:24:33   CXX      message/message_read-read.o
21:24:37   CXXLD    message_read
21:24:37   CXX      message/native_types.o
21:24:38   CXXLD    message_native_types
21:24:38   CXX      message/types.o
21:24:39   CXXLD    message_types
21:24:39   CXX      utility/tuple_to_array.o
21:24:40   CXXLD    utility_tuple_to_array
21:24:40   CXX      utility/type_traits.o
21:24:40   CXXLD    utility_type_traits
21:24:40   CXX      vtable/vtable.o
21:24:40   CC       vtable/vtable_c.o
21:24:40   CXXLD    vtable_vtable
21:24:40 make[2]: Leaving directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/sdbusplus/test'
21:24:40 make  check-TESTS
21:24:40 make[2]: Entering directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/sdbusplus/test'
21:24:40 make[3]: Entering directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/sdbusplus/test'
```

Patrick has some info on this but basically when he wants to run these tests, he has to tell his session where the DBUS daemon is via DBUS_SESSION_BUS_ADDRESS.

So in https://github.com/openbmc/openbmc-build-scripts/blob/master/build-unit-test-docker.sh we may need to make sure we're installing dbus, and then we may need to update the scripts we run to execute the tests to find and set this env variable.  

You can verify all is good by running the CI script successfully against the openbmc/sdbusplus repo.

williamspatrick, 2017-02-15:
There is a directory ~/.dbus/session_bus that has files that can be sourced for the session bus.  On my system, it looks almost like there is a new file created per reboot so you'd need to get the latest one.  You'll also need to figure out how to send this environment variable to the docker instance and be sure that /tmp is mapped into the docker instance.

lgon, 2017-02-17:
I was able to build and run the phosphor-logging unit tests in docker using 
cd /esw/san2/lgonzalez/obmc
export WORKSPACE=/esw/san2/lgonzalez/obmc
export UNIT_TEST_PKG=phosphor-logging
./openbmc-build-scripts/build-unit-test-docker.sh openbmc/ubuntu-unit-test ubuntu:latest
 ./openbmc-build-scripts/run-unit-test-docker.sh 

i never set DBUS_SESSION_BUS_ADDRESS, but I got a passing result, and it did create the sdbusplus and  phosphor-dbus-interfaces directories in WORKSPACE.  

PASS: elog_unittest
============================================================================
Testsuite summary for phosphor-logging 1.0
============================================================================
# TOTAL: 1
# PASS:  1
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================

Am I missing something?

geissonator, 2017-02-18:
I never really detailed where we hit the fail :)

Try running it against the openbmc/sdbusplus repo, that's the one with the test cases that use dbus.

williamspatrick, 2017-02-18:
When you log in, Docker might preserve your environment?  The Jenkins user maybe doesn't have a dbus session server running.

lgon, 2017-03-07:
I was able to get openbmc/openbmc-build-scripts/run-unit-test-docker.sh  to work with sdbusplus by having it start a session dbus with a new system-local.conf file with rules to allow the test messages to be sent.  This required changes to the unit test to use the session instead of the system bus.  I also tested to make sure these changes worked with the phosphor-logging unit tests. I'll put these changes up for code review.

williamspatrick, 2017-03-07:
The current code uses the 'default' bus.  DBus uses the 'system' bus as the default for 'root' and 'session' bus as the default for non-root.  There shouldn't be a reason to change code here.

Does Docker run as root?  (I'm surprised because that sounds a little dangerous)  Is there a mechanism to either give the session bus to the Docker session (sounds even more dangerous) or else to run the Docker instance as a non-root user?

lgon, 2017-03-07:
Docker currently runs as root. I'll look at running the unit test in docker as a non-root user.

lgon, 2017-03-07:
I looked at the source code, and sdbusplus uses the environment variable DBUS_STARTER_BUS_TYPE. So, starting up a session bus with the new config file, etting this to "session" in the docker run command, along with , and setting the environment variables DBUS_STARTER_BUS_TYPE, and DBUS_SESSION_BUS_ADDRESS in the docker run command allow the unit tests to work without changing anything in sdbusplus.  Running the docker container as a non-root user doesn't work, because there is package configuration as part of the build process. 
``
running install_lib
creating /usr/local/lib/python2.7/dist-packages/sdbusplus
error: could not create '/usr/local/lib/python2.7/dist-packages/sdbusplus': Permission denied
Makefile:477: recipe for target 'install-exec-hook' failed
make[3]: *** [install-exec-hook] Error 1
``

lgon, 2017-03-09:
I updated the code review. The dbus session is now launched from the docker container. https://gerrit.openbmc-project.xyz/#/c/2906/

rfrandse, 2017-03-28:
Leonel Gonzalez: fx for dbus docker unit tests have passed review, been merged, and passed Jenkins

closed 2017-03-28 by rfrandse