openbmc/openbmc #1140: [Witherspoon w33 v1.99.2-60-gcb1cc2b ] Failed to start Phosphor Version Software Manager

gkeishin, 2017-02-16:
**Driver Info**
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-60"
VERSION_ID="v1.99.2-60-gcb1cc2b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-60"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170216042029
root@witherspoon:~# uptime
 05:32:57 up 24 min,  load average: 0.16, 0.18, 0.27
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~#
```

**Failed to start Phosphor Version Software Manager**
```
root@witherspoon:~# systemctl status xyz.openbmc_project.Software.Version.service | cat
● xyz.openbmc_project.Software.Version.service - Phosphor Version Software Manager
   Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Software.Version.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: core-dump) since Thu 2017-02-16 05:09:28 UTC; 22min ago
  Process: 809 ExecStart=/usr/sbin/phosphor-version-software-manager (code=dumped, signal=ABRT)
 Main PID: 809 (code=dumped, signal=ABRT)

Feb 16 05:09:28 witherspoon systemd[1]: Stopped Phosphor Version Software Manager.
Feb 16 05:09:28 witherspoon systemd[1]: Starting Phosphor Version Software Manager...
Feb 16 05:09:28 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Main process exited, code=dumped, status=6/ABRT
Feb 16 05:09:28 witherspoon systemd[1]: Failed to start Phosphor Version Software Manager.
Feb 16 05:09:28 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Unit entered failed state.
Feb 16 05:09:28 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Failed with result 'core-dump'.
Feb 16 05:09:28 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Service hold-off time over, scheduling restart.
Feb 16 05:09:28 witherspoon systemd[1]: Stopped Phosphor Version Software Manager.
Feb 16 05:09:28 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Start request repeated too quickly.
Feb 16 05:09:28 witherspoon systemd[1]: Failed to start Phosphor Version Software Manager.
```

- [x] Journal lod  
[journald_v1.99.2-60-gcb1cc2b_20170216042029_SW_version.txt](https://github.com/openbmc/openbmc/files/779156/journald_v1.99.2-60-gcb1cc2b_20170216042029_SW_version.txt)

vishwabmc, 2017-02-16:
@gkeishin please help collect the coredump .

@ratagupt has collected a coredump sometimes and he gave these instructions.. 

Execute `echo '/tmp/core_%e.%p' | tee /proc/sys/kernel/core_pattern`

And then start the process and you will see a coredump in $PWD.

@ratagupt can help here for mre.

gkeishin, 2017-02-16:
@vishwabmc  There you go the core dump attached 
[core_dumps.tar.gz](https://github.com/openbmc/openbmc/files/779261/core_dumps.tar.gz)

vishwabmc, 2017-02-16:
@geissonator  Please assign this to @gtmills ?

rahulmah, 2017-02-16:
Also with Barreleye latest build v1.99.2-60-gcb1cc2b/20170216043607

geissonator, 2017-02-16:
Yeah, Gunnar's app is hitting the same thing that Josh is hitting with phosphor-bmc-state-manager.  This is a weird one.

williamspatrick, 2017-02-16:
Duplicate of #1072; will be fixed in sdbusplus.

gkeishin, 2017-02-17:
Closing this..

closed 2017-02-17 by gkeishin