openbmc/openbmc #1141: [Config Setting ] Couldn't write '16' to 'kernel/sysrq'

gkeishin, 2017-02-16:
Had a discussion with @amboar   and this shouldn't be ideally show up here.

**This is not breaking any flow for now but seems like there are `clearly some missing config settings from the kernel config vs what userspace is expecting.`**

```
Jan 31 18:55:31 witherspoon systemd-sysctl[510]: Couldn't write '16' to 'kernel/sysrq', ignoring: No such file or directory
Jan 31 18:55:31 witherspoon systemd-sysctl[510]: Couldn't write 'fq_codel' to 'net/core/default_qdisc', ignoring: No such file or directory
```

We don't see this often but pop up in the log now and then. So here is one journal which have this 

- [x]  Journald log 
[journald_v1.99.2-46-g04f01f7_20170215035323_w38_sysctl.txt](https://github.com/openbmc/openbmc/files/779178/journald_v1.99.2-46-g04f01f7_20170215035323_w38_sysctl.txt)

vishwabmc, 2017-02-16:
@gkeishin journal would not help here IMO but yeah I understand that's our standard debug data.

@shenki / @amboar Something you could help here ?

shenki, 2017-02-16:
systemd is making an error in assuming these files will be present. We could take it up with upstream systemd, suggesting they should not print warnings when these features are not configured into the kernel.

vishwabmc, 2017-02-16:
@shenki While using MCP kernel, I was doing a 'echo "c" > /proc/sysrq_trigger` to do panics and I saw that it was not available in OE kernel. I found that useful for test folks to trigger some resets.

shenki, 2017-02-16:
I don't think we should enable that testing feature in our images as we follow the general principle of not enabling code that could expose bugs.

If you feel strongly about enabling debug features then we can have that discussion.

vishwabmc, 2017-02-16:
Thanks.. I don't have a strong opinion on that now..

vishwabmc, 2017-02-16:
@williamspatrick So we will open an issue for systemd upstream -or- something we want to patch here ?

gtmills, 2018-08-20:
@gkeishin @shenki This is an old kernel, 4.7.10, and an old build. Do we still see? Can this be closed?

gkeishin, 2018-08-20:
I m pretty sure I still see those in latest and greatest builds out there. 

```
root@witherspoon:~# journalctl --no-pager | grep "Couldn't"
Aug 20 16:34:01 witherspoon systemd-sysctl[710]: Couldn't write '16' to 'kernel/sysrq', ignoring: No such file or directory
Aug 20 16:34:01 witherspoon systemd-sysctl[710]: Couldn't write 'fq_codel' to 'net/core/default_qdisc', ignoring: No such file or directory
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v2.3-159"
VERSION_ID="v2.3-159-g7a81a2d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v2.3-159"
BUILD_ID="v2.3"
root@witherspoon:~#
```

```
root@witherspoon:~# uname -a
Linux witherspoon 4.17.14-4b44547f456a87e8456fe3f4ed2b62ab696249b7 #1 Wed Aug 15 02:38:57 UTC 2018 armv6l GNU/Linux
root@witherspoon:~#
```

amboar, 2019-03-27:
> The comment I made still stands; we can take it up with upstream systemd if we want to change the warnings.

No one seems motivated to do anything about this so I'm closing it. Re-open if someone bothers to talk to upstream about it.

closed 2019-03-27 by amboar