openbmc/openbmc #1142: [Witherspoon] Deivce tree BMC Name, Clocks Name, Compatible property not set/defined

sivassrr, 2017-02-16:
With the latest version, 
```
root@witherspoon:/proc/device-tree# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-56"
VERSION_ID="v1.99.2-56-g02ac806-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-56"
BUILD_ID="v1.99.2"
root@witherspoon:/proc/device-tree# cat /etc/version
```

BMC Name property is not set on Witherspoon, Palmetto, Barreleye. 
```
root@witherspoon:/sys/firmware/devicetree/base# ls
#address-cells    chosen            interrupt-parent  name
#size-cells       compatible        leds              reserved-memory
ahb               cpus              memory
aliases           fsi-master        model
root@witherspoon:/sys/firmware/devicetree/base# cat name
root@witherspoon:/sys/firmware/devicetree/base#
```

Checked under /proc/device-tree as well
```
root@witherspoon:/proc/device-tree# ls
#address-cells    chosen            interrupt-parent  name
#size-cells       compatible        leds              reserved-memory
ahb               cpus              memory
aliases           fsi-master        model
root@witherspoon:/proc/device-tree# cat name
root@witherspoon:/proc/device-tree#
```
Similarly for Witherspoon clocks name and compatibility property is missing as below
```
root@witherspoon:/sys/firmware/devicetree/base# ls
#address-cells    chosen            interrupt-parent  name
#size-cells       compatible        leds              reserved-memory
ahb               cpus              memory
aliases           fsi-master        model
```

Test Results as follows:
==============================================================================
Test Bmc Devicetree :: This suite is for Verifying BMC device tree.
==============================================================================
Check BMC Model Property Is Set :: Verify if the BMC Model is popu... | PASS |
------------------------------------------------------------------------------
Check BMC Name Property Is Set :: Verify if the BMC name property ... | FAIL |
'1 > 1' should be true.
------------------------------------------------------------------------------
Check BMC Compatible Property Is Set :: Verify if the BMC compatib... | PASS |
------------------------------------------------------------------------------
Check BMC CPU Name Property Is Set :: Verify if the BMC CPU name p... | PASS |
------------------------------------------------------------------------------
Check BMC CPU Compatible Property Is Set :: Verify if the BMC CPU ... | PASS |
------------------------------------------------------------------------------
Check BMC Memory Name Property Is Set :: Verify if the BMC Memory ... | PASS |
------------------------------------------------------------------------------
Check BMC Memory Device Type Property Is Set :: Verify if the BMC ... | PASS |
------------------------------------------------------------------------------
Check BMC FSI Name Property Is Set :: Verify if the BMC FSI name p... | PASS |
------------------------------------------------------------------------------
Check BMC FSI Compatible Property Is Set :: Verify if the BMC FSI ... | PASS |
------------------------------------------------------------------------------
Check BMC LED Name Property Is Set :: Verify if the BMC LED name p... | PASS |
------------------------------------------------------------------------------
Check BMC LED Compatible Property Is Set :: Verify if the BMC LED ... | PASS |
------------------------------------------------------------------------------
Check BMC Clocks Name Property Is Set :: Verify if the BMC clocks ... | FAIL |
'cat: can't open '/sys/firmware/devicetree/base/clocks/name': No such file or directory' should be empty.
------------------------------------------------------------------------------
Check BMC Clocks Compatible Property Is Set :: Verify if the BMC c... | FAIL |
'cat: can't open '/sys/firmware/devicetree/base/clocks/clk_clkin/compatible': No such file or directory' should be empty.
------------------------------------------------------------------------------
Test Bmc Devicetree :: This suite is for Verifying BMC device tree.   | PASS |
0 critical tests, 0 passed, 0 failed
13 tests total, 10 passed, 3 failed
==============================================================================

vishwabmc, 2017-02-16:
@sivassrr looks like your latest update needs to be formatted.

sivassrr, 2017-02-16:
Done.

mdmillerii, 2017-02-20:
FYI `/proc/device-tree` is a symlink to `/sys/firmware/devicetree/base` so these will always match.

gkeishin, 2017-05-26:
On New BMC kernel 4.10
```
•	BUILD ID   : https://openpower.xyz/job/openbmc-build/766/
•	VERSION_ID : v1.99.6-10-g1d22c22
•	BMC Kernel : Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Fri May 26 05:22:23 UTC 2017 armv6l GNU/Linux

```

```
Check_BMC_FSI_Compatible_Property_Is_Set

'cat: can't open '/sys/firmware/devicetree/base/fsi-master/compatible': No such file or directory' should be empty
```

```
Check_BMC_FSI_Name_Property_Is_Set
'cat: can't open '/sys/firmware/devicetree/base/fsi-master/name': No such file or directory' should be empty.
```

williamspatrick, 2017-05-29:
I suspect these properties changed for 4.10.

cbostic, 2017-05-31:
The name changed for 4.10:

```/sys/firmware/devicetree/base/gpio-fsi/sys/firmware/devicetree/base/gpio-fsi/compatible```

```/sys/firmware/devicetree/base/gpio-fsi/sys/firmware/devicetree/base/gpio-fsi/name```

sivassrr, 2017-06-23:
@cbostic this new changes applicable only to witherspoon, right. Not to Barreleye / Palmetto.

sivassrr, 2017-06-23:
So we could add 2 more TCs associated with GPIO-FSI for witherspoon. 
But witherspoon basic BMC name, compatible and clock properties are not populated.

cbostic, 2017-10-16:
@rfrandse can you please mark as Phase 7?

cbostic, 2017-10-20:
Verified the name property is still missing with latest kernel.

amboar, 2017-10-22:
Why do we care about the name property on the root node? What are the requirements that might be met by it? What cannot be fulfilled by the compatible and model attributes?

mdmillerii, 2017-10-25:
Thinking about this, the `name` property is a standard property that is the name of the node minus the unit address, so it should always be empty for the root node.

amboar, 2017-10-25:
@mdmillerii right, I could have been less cryptic about it :)

gtmills, 2018-06-29:
@sivassrr @amboar Can this be closed or still an issue?

amboar, 2018-07-09:
Yeah I don't see a way forward with this. Expecting name to be populated on the root node is a misunderstanding of what's going on, not something that needs to be fixed.

closed 2018-07-09 by amboar